esphome/issues #2693: WLAN connection not possible

minimag450, 2021-11-09:
### The problem

Since the last update, the WLAN connection from a programmed ESP8266 nodemcu (I also tried different D1, ESP8266 and ESP32) does not work anymore. I have a very simple sketch to try. Once I entered only the FRITZBox 7490, once only the FRITZ!Box 3390 in the sketch, no connection to the WLAN there either. In the log you can see that the 7490 is found, the 3390 is sometimes not found. And then the 7490 is gone. If I press the reset button on the D1, then it finds the 7490, tries to log in 2 times and then no longer finds it.
There are also no failed login attempts in the 7490.
Translated with www.DeepL.com/Translator (free version)

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.0

### What platform are you using?

ESP8266

### Board

nodemcu, D1 mini

### Component causing the issue

none

### Example YAML snippet

```yaml
esphome:
  name: test1a
  platform: ESP8266
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "73857115a105b5a4f92eb74b7431d8fd"

wifi:
  ssid: "FRITZ!Box 7490"
  password: "39428586847057207169"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Test1A Fallback Hotspot"
    password: "QWekGR2hqcGx"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/test1a.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing test1a (board: nodemcuv2; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
[... 322 lines cut ...]
[10:17:58][D][wifi:436]: - [redacted] [redacted]▂▄▆█
[10:17:58][D][wifi:436]: - [redacted] [redacted]▂▄▆█
[10:17:58][I][wifi:247]: WiFi Connecting to 'FRITZ!Box 7490'...
[10:18:00][W][wifi_esp8266:483]: Event: Disconnected ssid='FRITZ!Box 7490' bssid=[redacted] reason='Association Expired'
[10:18:00][W][wifi:535]: Error while connecting to network.
[10:18:00][W][wifi:570]: Restarting WiFi adapter...
[10:18:00][I][wifi:247]: WiFi Connecting to 'FRITZ!Box 7490'...
[10:18:06][W][wifi_esp8266:479]: Event: Disconnected ssid='FRITZ!Box 7490' reason='Probe Request Unsuccessful'
[10:18:06][W][wifi:535]: Error while connecting to network.
[10:18:06][W][wifi:570]: Restarting WiFi adapter...
[10:18:11][D][wifi:372]: Starting scan...
[10:18:17][D][wifi:387]: Found networks:
[10:18:17][D][wifi:436]: - [redacted] [redacted]▂▄▆█
[10:18:17][W][wifi:441]: No matching network found!
[10:18:22][D][wifi:372]: Starting scan...
[10:18:28][D][wifi:387]: Found networks:
[10:18:28][D][wifi:436]: - [redacted] [redacted]▂▄▆█
[10:18:28][W][wifi:441]: No matching network found!
[10:18:33][D][wifi:372]: Starting scan...
```

### Additional information

[image]
[image]

LordBoos, 2021-11-09:
I have similar problem. After updating my Sonoff Basic to latest ESPHome, it no longer connects to WiFi reliably. it's behaviour is really strange thought because when powered from UART-to-serial adapter it will connect to WiFi on like 50% of boots on the rest of the boots it will just scan for networks, sometimes find my network and then fail to connect with **reason='Probe Request Unsuccessful'**. When powered directly from outlet it never connects to wifi (I've tried many reboots).
I had no problems with versions older than 2021.10.X.

minimag450, 2021-11-10:
Maybe it's interesting for the developers, today I noticed that I have a network "TestA1 Fallback Hotspot".
Regards, Andreas

LordBoos, 2021-11-10:
I also have an update. I flashed the same code to another device (Sonoff MINI R2) and it works fine. Looks like the only problem is, at least for me, with Sonoff Basic R1.

itanczos, 2021-11-13:
**For me, removing the status_led component solved the problem.**

This wifi problem has already occurred in versions 2021.9.x, but only if the web_server component was also enabled.
Then the web_server and status_led components did not work at the same time. Then the web_server component was caught with a large memory requirement.
I just don't understand why it only occurs at sonoff basic r1? Also not for esp8266 based wemos d1?

LordBoos, 2021-11-13:
I do use web_server or status_led components.
```
esphome:
  name: kitchen
  platform: ESP8266
  board: esp8285

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

api:

i2c:
  scan: false
  sda: GPIO1
  scl: GPIO3

sensor:
  - platform: bme280
    temperature:
      name: "Teplota v kuchyni"
      oversampling: 2x
    pressure:
      name: "Tlak v kuchyni"
      oversampling: 2x
    humidity:
      name: "Vhkost v kuchyni"
      oversampling: 2x
    address: 0x76
    update_interval: 30s

output:
  - platform: gpio
    pin: 12
    id: light_main

binary_sensor:
  - platform: gpio
    id: button_sensor
    pin:
      number: GPIO0
      mode: INPUT_PULLUP
      inverted: True
    on_press:
      then:
        - light.toggle:
            id: light_light
  - platform: gpio
    id: switch_sensor
    pin:
      number: GPIO14
      mode: INPUT_PULLUP
    on_press:
      then:
        - light.toggle:
            id: light_light
    on_release:
      then:
        - light.toggle:
            id: light_light
  - platform: status
    name: "Kuchyně vypínač status"

light:
  - platform: binary
    id: light_light
    name: Kuchyně
    output: light_main

switch:
  - platform: restart
    name: "Kuchyně vypínač Restart"

logger:

ota:
```

minimag450, 2021-11-14:
I don't understand exactly what your posts have to do with my Problem. They seem to work.

minimag450, 2021-11-14:
Maybe the log from ESPHome will help.

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Add-on: ESPHome
 ESPHome add-on for intelligently managing all your ESP8266/ESP32 devices
-----------------------------------------------------------
 Add-on version: 2021.10.3
 You are running the latest version of this add-on.
 System: Home Assistant OS 6.6  (aarch64 / raspberrypi4-64)
 Home Assistant Core: 2021.11.3
 Home Assistant Supervisor: 2021.10.8
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] 10-requirements.sh: executing... 
[cont-init.d] 10-requirements.sh: exited 0.
[cont-init.d] 20-nginx.sh: executing... 
[cont-init.d] 20-nginx.sh: exited 0.
[cont-init.d] 30-dirs.sh: executing... 
[cont-init.d] 30-dirs.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[21:15:58] INFO: Waiting for dashboard to come up...
[21:15:58] INFO: Starting ESPHome dashboard...
2021-11-14 21:16:00,040 INFO Starting dashboard web server on unix socket /var/run/esphome.sock and configuration dir /config/esphome...
[21:16:00] INFO: Starting NGINX...
2021-11-14 21:16:15,863 INFO 304 GET / (0.0.0.0) 669.97ms
2021-11-14 21:16:16,351 INFO 200 GET /static/fonts/material-icons/material-icons.css?hash=ead4c276 (0.0.0.0) 349.57ms
2021-11-14 21:16:16,390 INFO 200 GET /static/css/vendor/materialize/materialize.min.css?hash=ec1df3ba (0.0.0.0) 33.08ms
2021-11-14 21:16:16,398 INFO 200 GET /static/css/esphome-2.css?hash=95c08d5a (0.0.0.0) 35.96ms
2021-11-14 21:16:16,404 INFO 200 GET /static/js/vendor/jquery/jquery.min.js?hash=12108007 (0.0.0.0) 40.18ms
2021-11-14 21:16:16,417 INFO 200 GET /static/js/vendor/jquery-ui/jquery-ui.min.js?hash=cab3392d (0.0.0.0) 50.68ms
2021-11-14 21:16:16,428 INFO 200 GET /static/js/vendor/jquery-validate/jquery.validate.min.js?hash=aad785d4 (0.0.0.0) 7.21ms
2021-11-14 21:16:16,477 INFO 200 GET /static/js/vendor/materialize/materialize.min.js?hash=4be20daf (0.0.0.0) 5.44ms
2021-11-14 21:16:16,563 INFO 200 GET /static/js/vendor/ace/ace.js?hash=65322f49 (0.0.0.0) 22.18ms
2021-11-14 21:16:16,612 INFO 200 GET /static/js/esphome/index-e0440f6f.js (0.0.0.0) 42.85ms
2021-11-14 21:16:16,937 INFO 200 GET /static/fonts/material-icons/MaterialIcons-Regular.woff2 (0.0.0.0) 184.45ms
2021-11-14 21:16:16,995 INFO 200 GET /static/js/vendor/ace/theme-dreamweaver.js (0.0.0.0) 54.43ms
2021-11-14 21:16:17,223 INFO 200 GET /static/js/vendor/ace/mode-yaml.js (0.0.0.0) 280.52ms
2021-11-14 21:16:17,568 INFO 304 GET /devices (0.0.0.0) 340.20ms
2021-11-14 21:16:17,573 INFO 101 GET /ace (0.0.0.0) 340.89ms
2021-11-14 21:16:17,582 INFO Running command 'esphome --dashboard -q vscode --ace /config/esphome'
2021-11-14 21:16:22,279 INFO 304 GET /devices (0.0.0.0) 11.93ms
2021-11-14 21:16:25,390 INFO 200 GET /static/js/esphome/c.86557d2d.js (0.0.0.0) 19.05ms
2021-11-14 21:16:25,432 INFO 200 GET /static/js/esphome/c.42357ee2.js (0.0.0.0) 3.34ms
2021-11-14 21:16:25,529 INFO 200 GET /static/js/esphome/c.9fbc2162.js (0.0.0.0) 76.59ms
2021-11-14 21:16:25,611 INFO 200 GET /static/js/esphome/c.849528ae.js (0.0.0.0) 78.18ms
2021-11-14 21:16:25,662 INFO 200 GET /static/js/esphome/c.82ae9110.js (0.0.0.0) 126.16ms
2021-11-14 21:16:25,722 INFO 200 GET /static/js/esphome/c.989b6db0.js (0.0.0.0) 184.00ms
2021-11-14 21:16:27,008 INFO 304 GET /devices (0.0.0.0) 9.61ms
2021-11-14 21:16:28,790 INFO 200 GET /static/js/esphome/c.177725c5.js (0.0.0.0) 3.96ms
2021-11-14 21:16:28,839 INFO 200 GET /static/js/esphome/c.2cdbdbed.js (0.0.0.0) 3.67ms
2021-11-14 21:16:28,977 INFO 101 GET /upload (0.0.0.0) 1.79ms
2021-11-14 21:16:28,995 INFO Running command 'esphome --dashboard run /config/esphome/heizung-kinderzimmer.yaml --device /dev/ttyUSB0'
2021-11-14 21:16:32,020 INFO 304 GET /devices (0.0.0.0) 6.86ms
2021-11-14 21:16:37,017 INFO 304 GET /devices (0.0.0.0) 13.83ms
2021-11-14 21:16:42,047 INFO 304 GET /devices (0.0.0.0) 16.28ms
2021-11-14 21:16:47,012 INFO 304 GET /devices (0.0.0.0) 6.88ms
2021-11-14 21:16:52,017 INFO 304 GET /devices (0.0.0.0) 10.62ms
2021-11-14 21:16:57,019 INFO 304 GET /devices (0.0.0.0) 5.92ms
2021-11-14 21:17:02,020 INFO 304 GET /devices (0.0.0.0) 9.25ms
2021-11-14 21:17:07,028 INFO 304 GET /devices (0.0.0.0) 8.07ms
2021-11-14 21:17:12,027 INFO 304 GET /devices (0.0.0.0) 6.39ms
2021-11-14 21:17:17,036 INFO 304 GET /devices (0.0.0.0) 7.47ms
2021-11-14 21:17:22,033 INFO 304 GET /devices (0.0.0.0) 7.80ms
2021-11-14 21:17:27,027 INFO 304 GET /devices (0.0.0.0) 6.06ms
2021-11-14 21:17:32,031 INFO 304 GET /devices (0.0.0.0) 5.37ms
2021-11-14 21:17:37,026 INFO 304 GET /devices (0.0.0.0) 6.47ms
2021-11-14 21:17:42,042 INFO 304 GET /devices (0.0.0.0) 6.41ms
2021-11-14 21:17:47,035 INFO 304 GET /devices (0.0.0.0) 6.31ms
2021-11-14 21:17:52,045 INFO 304 GET /devices (0.0.0.0) 6.98ms

minimag450, 2021-11-17:
Thanks!  I didn't know until now that it would work that way.  But then there is an error message, as can be seen in the picture.
[image]

closed 2022-04-05 by github-actions[bot]