esphome/issues #2694: Modbus controller ignoring write_lambda

Jopand, 2021-11-09:
### The problem

Following the exact same lambda from the example (https://esphome.io/components/number/modbus_controller.html), nothing seems to happen. Looking at the generated main.cpp, it seems like it is totally ignored:

  // number.modbus_controller:
  //   platform: modbus_controller
  //   modbus_controller_id: nilan_modbus_controller
  //   name: User time set (minutes)
  //   id: nilan_user_time_set
  //   value_type: U_WORD
  //   min_value: 1.0
  //   max_value: 480.0
  //   write_lambda: !lambda |-
  //     ESP_LOGD("main","Modbus Number incoming value = %f",x);
  //     uint16_t b_capacity = x ;
  //     payload.push_back(b_capacity);
  //     return x * 1.0 ;
  //   step: 1.0
  //   address: 602
  //   force_new_range: true
  //   disabled_by_default: false
  //   offset: 0
  //   bitmask: 0xFFFFFFFF
  //   register_count: 0
  //   skip_updates: 0
  //   multiply: 1.0
  //   update_interval: 60s
  nilan_user_time_set = new modbus_controller::ModbusNumber(602, 0, 0xFFFFFFFF, modbus_controller::SensorValueType::U_WORD, 1, 0, true);
  nilan_user_time_set->set_update_interval(60000);
  nilan_user_time_set->set_component_source("modbus_controller.number");
  App.register_component(nilan_user_time_set);
  App.register_number(nilan_user_time_set);
  nilan_user_time_set->set_name("User time set (minutes)");
  nilan_user_time_set->set_disabled_by_default(false);
  nilan_user_time_set->traits.set_min_value(1.0f);
  nilan_user_time_set->traits.set_max_value(480.0f);
  nilan_user_time_set->traits.set_step(1.0f);
  nilan_user_time_set->set_write_multiply(1.0f);
  nilan_user_time_set->set_parent(nilan_modbus_controller);
  nilan_modbus_controller->add_sensor_item(nilan_user_time_set);

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Component causing the issue

modbus_controller

### Example YAML snippet

```yaml
uart:
  rx_pin: GPIO16
  tx_pin: GPIO17
  parity: EVEN
  baud_rate: 19200
  id: uart_modbus
  stop_bits: 1
  
modbus:
  id: modbus_id
  uart_id: uart_modbus
  
modbus_controller:
  id: nilan_modbus_controller
  address: 30
  modbus_id: modbus_id

number:
  - platform: modbus_controller
    modbus_controller_id: nilan_modbus_controller
    name: "User time set (minutes)"
    id: nilan_user_time_set
    value_type: U_WORD
    min_value: 1.0
    max_value: 480.0
    write_lambda: |-
      ESP_LOGD("main","Modbus Number incoming value = %f",x);
      uint16_t b_capacity = x ;
      payload.push_back(b_capacity);
      return x * 1.0 ;
    step: 1.0
    address: 602
    force_new_range: true
```

martgras, 2021-11-10:
Strange just tested again and I can't repro

```
number:
  - platform: modbus_controller
    modbus_controller_id: epever
    id: battery_capacity_number
    name: "Battery Cap Number"
    address: 0x9001
    value_type: U_WORD
    lambda: "return  x * 1.0; "
    write_lambda: |-
      ESP_LOGD("main","Modbus Number incoming value = %f",x);
      uint16_t b_capacity = x ;
      payload.push_back(b_capacity);
      return x * 1.0 ;
    ## multiply is ignored because lamdba is used
    multiply: 1.0
```
in main.cpp I see 
```
  epever->add_sensor_item(battery_capacity_number);
  battery_capacity_number->set_template([=](modbus_controller::ModbusNumber * item, float x, const std::vector<uint8_t> & data) -> optional<float> {
      #line 198 "..\\tracer\\epever.yaml"
      return  x * 1.0; 
  });
  battery_capacity_number->set_write_template([=](modbus_controller::ModbusNumber * item, float x, std::vector<uint16_t> & payload) -> optional<float> {
      #line 200 "..\\tracer\\epever.yaml"
      ESP_LOGD("main","Modbus Number incoming value = %f",x);
      uint16_t b_capacity = x ;
      payload.push_back(b_capacity);
      return x * 1.0 ;
  });
```
and changing the value from HA triggers
```
[14:55:45][D][number:010]: 'Battery Cap Number' - Setting
[14:55:45][D][number:029]:   Value: 116.000000
[14:55:45][D][main:200]: Modbus Number incoming value = 116.000000

[14:55:45][D][modbus.number:065]: Updating register: connected Sensor=Battery Cap Number start address=0x9001 register count=1 new value=116.00 (val=116.00)                                         
 [14:55:45][D][number:036]: 'Battery Cap Number': Sending state 116.000000                                                           
[14:55:47][D][number:010]: 'Battery Cap Number' - Setting
[14:55:47][D][number:029]:   Value: 115.000000
[14:55:47][D][main:200]: Modbus Number incoming value = 115.000000

[14:55:47][D][modbus.number:065]: Updating register: connected Sensor=Battery Cap Number start address=0x9001 register count=1 new value=115.00 (val=115.00)
[14:55:47][D][number:036]: 'Battery Cap Number': Sending state 115.000000
```

Jopand, 2021-11-10:
I think I see the issue:

It is in your python code generation. The CONF_WRITE_LAMBDA check is inside the CONF_LAMBDA check. So if I only use write_lambda, but not lambda, it ignores my write_lambda...

martgras, 2021-11-10:
Good catch - I missed the additional space :)
https://github.com/esphome/esphome/pull/2680 fixes the issue 

You can try 
```
external_components:
  # - source: github://martgras/esphome@modbus-customcmd
  - source: github://pr#2543
    components: ["modbus_controller"]
```

closed 2021-11-14 by jesserockz