esphome/issues #2696: Installation errors none.

marcinwrobel, 2021-11-10:
### The problem

When I try to install the code on the ESP Wemos D1 mini I get errors

Link to the project on Github:

https://github.com/tailsu/miele-brain

### Which version of ESPHome has the issue?

2021.10.3 

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.10.8

### What platform are you using?

ESP8266

### Board

WEMOS D1 mini

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esphome:
  name: miele-novotronic
  platform: ESP8266
  board: d1_mini
  
  includes:
    - MotorolaLedDriverSniffer.h
  platformio_options:
    build_flags: -O2
    board_build.f_cpu: 160000000L

wifi:
  networks:
    - ssid: "xxx"
      password: "XXX"
  reboot_timeout: 1min
  

ota:
  num_attempts: 5

mqtt:
  broker: mqtt.local

# logger:

text_sensor:
- platform: custom
  lambda: |-
    auto sniffer = new MotorolaLedDriverSniffer(miele_time, miele_state);
    App.register_component(sniffer);
    return {sniffer};
  text_sensors:
    id: sniffer

[... 194 lines cut ...]
                                   ^
src/MotorolaLedDriverSniffer.h:87:35: note: candidates are:
In file included from src/esphome/core/hal.h:4:0,
                 from src/esphome/components/esp8266/gpio.h:5,
                 from src/esphome.h:3,
                 from src/main.cpp:3:
src/esphome/core/gpio.h:50:7: note: constexpr esphome::GPIOPin::GPIOPin()
 class GPIOPin {
       ^
src/esphome/core/gpio.h:50:7: note:   candidate expects 0 arguments, 2 provided
src/esphome/core/gpio.h:50:7: note: constexpr esphome::GPIOPin::GPIOPin(const esphome::GPIOPin&)
src/esphome/core/gpio.h:50:7: note:   candidate expects 1 argument, 2 provided
src/esphome/core/gpio.h:50:7: note: constexpr esphome::GPIOPin::GPIOPin(esphome::GPIOPin&&)
src/esphome/core/gpio.h:50:7: note:   candidate expects 1 argument, 2 provided
In file included from src/main.cpp:29:0:
src/MotorolaLedDriverSniffer.h: In member function 'virtual void esphome::MotorolaLedDriverSniffer::setup()':
src/MotorolaLedDriverSniffer.h:112:14: error: 'class esphome::GPIOPin' has no member named 'attach_interrupt'
         _clk.attach_interrupt(&handleClk, this, RISING);
              ^
Compiling /data/miele-novotronic/.pioenvs/miele-novotronic/lib67b/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
Compiling /data/miele-novotronic/.pioenvs/miele-novotronic/lib67b/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
Compiling /data/miele-novotronic/.pioenvs/miele-novotronic/lib67b/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
*** [/data/miele-novotronic/.pioenvs/miele-novotronic/src/main.cpp.o] Error 1
========================= [FAILED] Took 16.81 seconds =========================
```

oxan, 2021-11-10:
This is an issue in the 3rd party component you're using, which is not (yet?) compatible with ESPHome >= 2021.10. I would recommend reporting your issue there, as there's nothing ESPHome itself can do about it.

closed 2021-11-10 by oxan

marcinwrobel, 2021-11-10:
It worked 7 months ago.

oxan, 2021-11-10:
> It worked 7 months ago.

Not with ESPHome 2020.10 ;)

We don't guarantee API stability. From time to time 3rd party components will need changes to be compatible with new ESPHome versions.