esphome/issues #2697: Webserver Not Serving Complete Page When css And js Files Are Local On S31

sjmellen, 2021-11-10:
### The problem

After reconfiguring my yaml file to serve the webserver-v1.min.js and css files locally, the webpage hangs when loading.  Sometimes a page refresh will rectify the problem.  Sometimes I have to refresh it several times.  I've seen the same issue on multiple machines (PCs, Android, etc) using multiple browsers (Chrome, Firefox, etc).

In Chrome, after enabling the developer tools, I've noticed in the "Networking" tab, "0.js" status stays at "pending".  As I've stated above, sometimes refreshing the page will get things unstuck resulting in all of the data being displayed (and apparently 0.js loads).

So far the Sonoff S31 is the only device I've seen this issue occur on.  I've also compiled and download firmware for a Martin Jerry SD-01 dimmer using the same HA and ESPHome versions and webserver doesn't hang on that device.

I thought perhaps this is a "low on memory" situation but when I compile the firmware for the S31 it's using 41016 bytes from 81920 bytes (RAM) and using 491008 bytes from 1023984 bytes (FLASH)... so only about half of each.  The MJ SD-01 uses about the same amount of memory.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.1

### What platform are you using?

ESP8266

### Board

Sonoff S31 With Power Monitoring

### Component causing the issue

webserver

### Example YAML snippet

```yaml
web_server:
  port: 80
  css_include: "/config/esphome/esphome-docs/_static/webserver-v1.min.css"
  css_url: ""
  js_include: "/config/esphome/esphome-docs/_static/webserver-v1.min.js"
  js_url: ""
```

### Anything in the logs that might be useful for us?

```txt
With logging set to verbose I don't see any messages pertaining to the webserver when I attempt to access the server.
```

EugeneKorneev, 2022-02-14:
Same effect

NopeNix, 2022-05-29:
Same here! I'm on v2022.5.1

here is how it shows up in the log:
`
[C][lcd_pcf8574:014]: Setting up PCF8574 LCD Display...
[C][wifi:037]: Setting up WiFi...
[W][wifi_esp8266:081]: Setting WiFi mode failed!
[C][wifi:201]: Setting up AP...
[C][wifi:203]:   AP SSID: 'TestAP'
[C][wifi:204]:   AP Password: '12345678'
[W][wifi_esp8266:081]: Setting WiFi mode failed!
[C][wifi:213]:   IP Address: 0.0.0.0
[C][web_server:096]: Setting up web server...

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: cont
sp: 3ffffc70 end: 3fffffc0 offset: 0000
3ffffc70:  00000004 00000038 3fff8498 402330dc  
3ffffc80:  000000fe 00000000 00000000 00000000  
3ffffc90:  00000000 00000000 00000000 00000001  
3ffffca0:  00007fff 00000000 3ffe8ef9 3fffb21c  
3ffffcb0:  00000000 4021b94a 00000010 40235532  
3ffffcc0:  00000000 4021b94a 00000020 40235551  
3ffffcd0:  00000004 00000000 00000010 40234b9c  
3ffffce0:  00000000 00000000 00000001 4021b94a  
3ffffcf0:  00000000 00000000 3ffffd40 00000000  
3ffffd00:  3fff9c48 00000018 3fff9c44 00000000  
3ffffd10:  3fffb18c 3fffae04 3fffb21c 4021b9c8  
3ffffd20:  3fff0000 00000020 3fff9c44 4021b9e3  
3ffffd30:  3ffffd60 3ffffd50 00000010 40222fb8  
3ffffd40:  3fff9c44 3fffb18c 40239a10 402399f4  
3ffffd50:  3fffae08 3fffa9c4 00000008 3ffffda8  
3ffffd60:  00ff0000 ff000000 00000001 3fffb42c  
3ffffd70:  3fff9c48 00000018 3fff9c44 4021c7e3  
3ffffd80:  00000000 3fffaba4 3fff80b8 3fffaba8  
3ffffd90:  00000000 00000018 3fff80b8 4021c878  
3ffffda0:  3fff992c 00000017 3fff80b8 40239898  
3ffffdb0:  3fffb428 3ffffdd0 3fff80b8 40239950  
3ffffdc0:  3fff992c 00000017 3fff80b8 402227ed  
3ffffdd0:  00000022 00000022 3fffb6dc 4010122e  
3ffffde0:  00000064 3fff80b8 3fffa2dc 402217b2  
3ffffdf0:  3fffaa80 3fffb18c 00000004 3fffa76c  
3ffffe00:  3fff81b4 3fff80b8 000000bc 4022659b  
3ffffe10:  3f800000 3f800000 3f800000 00000000  
3ffffe20:  3f800000 3f800000 feefeffe feefeffe  
3ffffe30:  3fffa2dc 3ffffe3c 00000000 00000000  
3ffffe40:  74732d6b 00747261 3f800000 3ffffe38  
3ffffe50:  0000000f 00000000 40223ffc 000001f4  
3ffffe60:  feefef00 00000000 3f800000 3f800000  
3ffffe70:  3f800000 3f800000 3f800000 3f800000  
3ffffe80:  00000000 3f800000 3f800000 000001f4  
3ffffe90:  3ffffe98 0000000a 74747548 65422065  
3ffffea0:  fe006772 feefeffe feefef00 00000000  
3ffffeb0:  3ffffeb8 0000000f 34333231 38373635  
3ffffec0:  31303139 00323131 00000000 00000000  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 3fff9e14 3fff9ea4  
3ffffef0:  00000000 3fff9fa0 56c42041 c2de6c1f  
3fffff00:  2d0de7cb 29df9247 7b068f85 b2099ada  
3fffff10:  8d0bbe36 bbd2ad8e 56c42041 c2de6c1f  
3fffff20:  2d0de7cb 29df9247 3ffffee8 00000001  
3fffff30:  3ffffe30 00000002 3ffffee8 00000001  
3fffff40:  3ffffe30 00000002 3ffffee8 00000001  
3fffff50:  3ffffe30 00000002 3ffffee8 00000001  
3fffff60:  3ffffe30 00000002 feefeffe feefeffe  
3fffff70:  00000101 00000001 3fff81c0 00000001  
3fffff80:  3f800000 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3fff8660  
3fffffa0:  3fffdad0 00000000 3fff864c 40235070  
3fffffb0:  feefeffe feefeffe 3ffe8618 401004c1  
<<<stack<<<

last failed alloc call: 4021B94A(16)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x4021b94a

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v00093f60
~ld
`

ferbar, 2022-09-06:
@NopeNix: If you are using an esp8266: I think the web_server v2 doesn't work on an esp8266 due to lack of ram. There has been a new web_server (v2) added which is the default. I'd suggest adding version: 1 if you get OOM

```
web_server
  version: 1
```