esphome/issues #2700: HDC1080 fails with latest ESPhome version

grob6000, 2021-11-11:
### The problem

I updated my script to change a temperature offset value. Compiling updated to latest version of ESPHome.
Now, the HDC1080 SPI temperature sensor is no longer working (it was working fine with 2021.8.2.

During log init, error "[E][hdc1080:032]: Communication with HDC1080 failed!" is shown. No temperature sensor updates then occur.

What's changed between these versions that's broken this?

Note that my SPI setup is unconventional (this is on a hacked device; I didn't get to choose which pins to use!):

i2c:
  sda: 2
  scl: 14
  scan: false

### Which version of ESPHome has the issue?

2021.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.3

### What platform are you using?

ESP8266

### Component causing the issue

HDC1080

### Example YAML snippet

```yaml
esphome:
  name: guestroomaircon
  platform: ESP8266
  board: esp_wroom_02

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: **SECRET**

wifi:
  networks:
  - ssid: **SECRET**
    password: **SECRET**
    hidden: true
  domain: ".localdomain"
  fast_connect: true

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  #ap:
  #  ssid: "Livingroomaircon"
  #  password: "RCsEPnkL5xjM"

#captive_portal:

web_server:
  port: 80
  
status_led:
  pin:
    number: 12
    inverted: false
    
i2c:
  sda: 2
  scl: 14
  scan: false
  
sensor:
  - platform: hdc1080
    temperature:
      name: "Guest Room Temperature"
      id: tempsensor
      filters:
        - offset: -3.4
        - sliding_window_moving_average:
            window_size: 5
            send_every: 1
    humidity:
      name: "Guest Room Humidity"
      id: humiditysensor
      filters:
        - lambda: |-
            float rh = x;
            float t1 = id(tempsensor).raw_state;
            float t2 = id(tempsensor).state;
            rh *= exp((17.67 * t1) / (t1 + 243.5));
            rh *= (273.15 + t2);
            rh /= exp((17.67 * t2) / (t2 + 243.5));
            rh /= (273.15 + t1);
            return rh;
        - sliding_window_moving_average:
            window_size: 5
            send_every: 1
    update_interval: 60s

binary_sensor:
  - platform: gpio
    pin:
      number: 13
      inverted: true
      mode: INPUT_PULLUP
    id: resetbutton
    internal: true

remote_transmitter:
  pin: 15
  carrier_duty_percent: 50%

climate:
  - platform: daikin       # adjust to match your AC unit!
    name: "Guest Room Air Conditioner"
    sensor: tempsensor
    receiver_id: receiver1
    
remote_receiver:
  - id: receiver1
    pin:
      number: 4
      inverted: true
      #mode: INPUT_PULLUP
    tolerance: 55%
    #dump: all
  - id: receiver2
    pin:
      number: 5
      inverted: true
      #mode: INPUT_PULLUP
    tolerance: 55%
    #dump: all
```

### Anything in the logs that might be useful for us?

```txt
INFO Successfully connected to guestroomaircon.localdomain
[17:37:02][I][app:099]: ESPHome version 2021.10.3 compiled on Nov 11 2021, 17:36:24
[17:37:02][C][status_led:019]: Status LED:
[17:37:02][C][status_led:020]:   Pin: GPIO12
[17:37:02][C][wifi:490]: WiFi:
[17:37:02][C][wifi:352]:   Local MAC: 2C:F4:32:12:12:6F
[17:37:02][C][wifi:353]:   SSID: [redacted]
[17:37:02][C][wifi:354]:   IP Address: 192.168.1.20
[17:37:02][C][wifi:356]:   BSSID: [redacted]
[17:37:02][C][wifi:357]:   Hostname: 'guestroomaircon'
[17:37:02][C][wifi:359]:   Signal strength: -55 dB ▂▄▆█
[17:37:02][C][wifi:363]:   Channel: 11
[17:37:02][C][wifi:364]:   Subnet: 255.255.255.0
[17:37:02][C][wifi:365]:   Gateway: 192.168.1.1
[17:37:02][C][wifi:366]:   DNS1: 192.168.1.1
[17:37:02][C][wifi:367]:   DNS2: 0.0.0.0
[17:37:02][C][i2c.arduino:032]: I2C Bus:
[17:37:02][C][i2c.arduino:033]:   SDA Pin: GPIO2
[17:37:02][C][i2c.arduino:034]:   SCL Pin: GPIO14
[17:37:02][C][i2c.arduino:035]:   Frequency: 50000 Hz
[17:37:02][C][i2c.arduino:038]:   Recovery: bus successfully recovered
[17:37:02][C][gpio.binary_sensor:015]: GPIO Binary Sensor 'resetbutton'
[17:37:02][C][gpio.binary_sensor:016]:   Pin: GPIO13
[17:37:02][C][logger:233]: Logger:
[17:37:02][C][logger:234]:   Level: DEBUG
[17:37:02][C][logger:235]:   Log Baud Rate: 115200
[17:37:02][C][logger:236]:   Hardware UART: UART0
[17:37:02][C][hdc1080:029]: HDC1080:
[17:37:02][C][hdc1080:030]:   Address: 0x40
[17:37:02][E][hdc1080:032]: Communication with HDC1080 failed!
[17:37:02][C][hdc1080:034]:   Update Interval: 60.0s
[17:37:02][C][hdc1080:035]:   Temperature 'Guest Room Temperature'
[17:37:02][C][hdc1080:035]:     Device Class: 'temperature'
[17:37:02][C][hdc1080:035]:     State Class: 'measurement'
[17:37:02][C][hdc1080:035]:     Unit of Measurement: '°C'
[17:37:02][C][hdc1080:035]:     Accuracy Decimals: 1
[17:37:02][C][hdc1080:036]:   Humidity 'Guest Room Humidity'
[17:37:02][C][hdc1080:036]:     Device Class: 'humidity'
[17:37:02][C][hdc1080:036]:     State Class: 'measurement'
[17:37:02][C][hdc1080:036]:     Unit of Measurement: '%'
[17:37:02][C][hdc1080:036]:     Accuracy Decimals: 0
[17:37:02][C][remote_transmitter:018]: Remote Transmitter...
[17:37:02][C][remote_transmitter:019]:   Carrier Duty: 50%
[17:37:02][C][remote_transmitter:020]:   Pin: GPIO15
[17:37:02][C][climate_ir:072]: IR Climate 'Guest Room Air Conditioner'
[17:37:02][C][climate_ir:073]:   Min. Temperature: 10.0°C
[17:37:02][C][climate_ir:074]:   Max. Temperature: 30.0°C
[17:37:02][C][climate_ir:075]:   Supports HEAT: YES
[17:37:02][C][climate_ir:076]:   Supports COOL: YES
[17:37:02][C][remote_receiver.esp8266:060]: Remote Receiver:
[17:37:02][C][remote_receiver.esp8266:061]:   Pin: GPIO4
[17:37:02][C][remote_receiver.esp8266:066]:   Buffer Size: 1000
[17:37:02][C][remote_receiver.esp8266:067]:   Tolerance: 55%
[17:37:02][C][remote_receiver.esp8266:068]:   Filter out pulses shorter than: 50 us
[17:37:02][C][remote_receiver.esp8266:069]:   Signal is done after 10000 us of no changes
[17:37:02][C][remote_receiver.esp8266:060]: Remote Receiver:
[17:37:02][C][remote_receiver.esp8266:061]:   Pin: GPIO5
[17:37:02][C][remote_receiver.esp8266:066]:   Buffer Size: 1000
[17:37:02][C][remote_receiver.esp8266:067]:   Tolerance: 55%
[17:37:02][C][remote_receiver.esp8266:068]:   Filter out pulses shorter than: 50 us
[17:37:02][C][remote_receiver.esp8266:069]:   Signal is done after 10000 us of no changes
[17:37:02][C][web_server:160]: Web Server:
[17:37:02][C][web_server:161]:   Address: guestroomaircon.localdomain:80
[17:37:02][C][ota:082]: Over-The-Air Updates:
[17:37:02][C][ota:083]:   Address: guestroomaircon.localdomain:8266
[17:37:02][C][ota:086]:   Using Password.
[17:37:02][C][api:134]: API Server:
[17:37:02][C][api:135]:   Address: guestroomaircon.localdomain:6053
[17:37:02][C][api:139]:   Using noise encryption: NO
```

grob6000, 2021-11-11:
I've commented out the following check line from hdc1080.cpp, and success; sensor is working (getting valid temperature output now):

```
void HDC1080Component::setup() {
  ESP_LOGCONFIG(TAG, "Setting up HDC1080...");

  const uint8_t data[2] = {
      0b00000000,  // resolution 14bit for both humidity and temperature
      0b00000000   // reserved
  };

  //if (!this->write_bytes(HDC1080_CMD_CONFIGURATION, data, 2)) {
  //  this->mark_failed();
  //  return;
  //}
}
```

```
[21:06:25][D][sensor:113]: 'Guest Room Temperature': Sending state 22.41268 °C with 1 decimals of accuracy
[21:06:25][D][climate:381]: 'Guest Room Air Conditioner' - Sending state:
[21:06:25][D][climate:384]:   Mode: OFF
[21:06:25][D][climate:389]:   Fan Mode: LOW
[21:06:25][D][climate:401]:   Swing Mode: BOTH
[21:06:25][D][climate:404]:   Current Temperature: 22.41°C
[21:06:25][D][climate:410]:   Target Temperature: 22.00°C
[21:06:25][D][sensor:113]: 'Guest Room Humidity': Sending state 69.82722 % with 0 decimals of accuracy
[21:06:25][D][hdc1080:067]: Got temperature=25.8°C humidity=57.6%
[21:07:25][D][sensor:113]: 'Guest Room Temperature': Sending state 22.40764 °C with 1 decimals of accuracy
[21:07:25][D][climate:381]: 'Guest Room Air Conditioner' - Sending state:
[21:07:25][D][climate:384]:   Mode: OFF
[21:07:25][D][climate:389]:   Fan Mode: LOW
[21:07:25][D][climate:401]:   Swing Mode: BOTH
[21:07:25][D][climate:404]:   Current Temperature: 22.41°C
[21:07:25][D][climate:410]:   Target Temperature: 22.00°C
[21:07:25][D][sensor:113]: 'Guest Room Humidity': Sending state 69.81764 % with 0 decimals of accuracy
[21:07:25][D][hdc1080:067]: Got temperature=25.8°C humidity=57.6%
[21:08:25][D][sensor:113]: 'Guest Room Temperature': Sending state 22.42275 °C with 1 decimals of accuracy
[21:08:25][D][climate:381]: 'Guest Room Air Conditioner' - Sending state:
[21:08:25][D][climate:384]:   Mode: OFF
[21:08:25][D][climate:389]:   Fan Mode: LOW
[21:08:25][D][climate:401]:   Swing Mode: BOTH
[21:08:25][D][climate:404]:   Current Temperature: 22.42°C
[21:08:25][D][climate:410]:   Target Temperature: 22.00°C
[21:08:25][D][sensor:113]: 'Guest Room Humidity': Sending state 69.85987 % with 0 decimals of accuracy
[21:08:25][D][hdc1080:067]: Got temperature=25.9°C humidity=57.6%
```

grob6000, 2021-11-11:
Next time I get the device out of the box I'll do a usb log with VV, see what the i2c error actually is.

schinckel, 2021-11-14:
This fixed my problem on a Sensibo.

grob6000, 2021-12-20:
Added a PR to fix this - change from error to warning if this fails. Works well.
Underlying bug can be tracked down later. :)

closed 2021-12-22 by jesserockz