esphome/issues #2701: ct_clamp sensor not receiving data from adc sensor

jacix, 2021-11-11:
### The problem

I have a split-core current clamp attached to an ESP32 that was working fine with version 2021.9.3, but after compiling the same YAML file with 2021.10, the adc sensor is reading properly, but the ct_clamp isn't getting any data. I also tried updating the YAML to the latest syntax (breaking out the platform, etc), but that didn't help.  I recompiled the config with 2021.9.3 and it works again.  Log snippets and more are below.

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

2021.11.2 (but it's working fine)

### What platform are you using?

ESP32

### Board

AITRIP D1 Mini (https://www.amazon.com/gp/product/B08PTRLLDC)

### Component causing the issue

ct_clamp

### Example YAML snippet

```yaml
sensor:
  - platform: adc
    pin: GPIO35
    id: adc_sensor
    attenuation: 11db
    update_interval: 2s

  - platform: ct_clamp
    sensor: adc_sensor
    name: "AC amps"
    update_interval: 2s
    filters:
      - calibrate_linear:
          - 0.0 -> 0.0
          - 0.47 -> 11.09
```

### Anything in the logs that might be useful for us?

```txt
Log snippet from 2021.11.2 which does not work:
-----------------
>[13:57:02][D][adc:109]: 'adc_sensor': Got voltage=0.00V
[13:57:02][D][sensor:113]: 'adc_sensor': Sending state 0.20000 V with 2 decimals of accuracy
[13:57:02][D][sensor:113]: 'AC amps': Sending state nan A with 2 decimals of accuracy
[13:57:04][D][adc:109]: 'adc_sensor': Got voltage=0.00V
[13:57:04][D][sensor:113]: 'adc_sensor': Sending state 0.20000 V with 2 decimals of accuracy
[13:57:04][D][sensor:113]: 'AC amps': Sending state nan A with 2 decimals of accuracy

Log snippet from 2021.9.3 which works well.  (note the additional line with each reading)
------------------
>[15:04:45][D][adc:104]: 'adc_sensor': Got voltage=0.00V
[15:04:45][D][sensor:121]: 'adc_sensor': Sending state 0.20000 V with 2 decimals of accuracy
[15:04:45][D][ct_clamp:051]: 'AC amps' - Raw Value: 0.00A
[15:04:45][D][sensor:121]: 'AC amps': Sending state 0.00000 A with 2 decimals of accuracy
[15:04:47][D][adc:104]: 'adc_sensor': Got voltage=0.00V
[15:04:47][D][sensor:121]: 'adc_sensor': Sending state 0.20000 V with 2 decimals of accuracy
[15:04:47][D][ct_clamp:051]: 'AC amps' - Raw Value: 0.00A
[15:04:47][D][sensor:121]: 'AC amps': Sending state 0.00000 A with 2 decimals of accuracy
```

### Additional information

I'm running this on MacOS 11.6.1 with Python3.8.  

Also, both versions are throwing the same incompatible pointer type warnings at compile-time, but since 2021.9.3 works fine, they don't appear to be causing the problem.  I'll include it anyway, just in case.  In fact, 2021.9.3 adds one additional complaint about not being able to find the `Hash` package for MacOS. 

2021.9.3
-------
> Library Manager: Installing Hash
Warning! Could not find the package with 'Hash' requirements for your system 'darwin_x86_64'
Dependency Graph
[...]
Compiling .pioenvs/dishwasherpower1/FrameworkArduino/esp32-hal-time.c.o
/Users/jason/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/Users/jason/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/Users/jason/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

2021.11.2
-------
> Compiling .pioenvs/dishwasherpower1/FrameworkArduino/esp32-hal-timer.c.o
/Users/jason/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/Users/jason/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/Users/jason/.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

jesserockz, 2021-11-14:
I am unable to reproduce this with my CT clamp on the latest dev branch.
```
[09:24:10][D][ct_clamp:038]: 'CT Measured Current' - Raw AC Value: 0.054A after 44 different samples (220 SPS)
[09:24:10][D][sensor:113]: 'CT Measured Current': Sending state 2.21924 A with 2 decimals of accuracy
[09:24:11][D][ct_clamp:038]: 'CT Measured Current' - Raw AC Value: 0.055A after 42 different samples (210 SPS)
[09:24:11][D][sensor:113]: 'CT Measured Current': Sending state 2.24959 A with 2 decimals of accuracy
[09:24:12][D][ct_clamp:038]: 'CT Measured Current' - Raw AC Value: 0.037A after 48 different samples (240 SPS)
[09:24:12][D][sensor:113]: 'CT Measured Current': Sending state 1.50856 A with 2 decimals of accuracy
```

yanshay, 2022-05-15:
Happens to me as well on 2022.4.0 .

In general CT clamp work ok but from time to time (every few seconds) it generates a nan value.

```
[17:05:45][D][ct_clamp:038]: 'Current1' - Raw AC Value: nanA after 382 different samples (1910 SPS)
[17:05:45][D][sensor:125]: 'Current1': Sending state nan A with 2 decimals of accuracy
[17:05:45][D][ct_clamp:038]: 'Current2' - Raw AC Value: 0.001A after 440 different samples (2200 SPS)
[17:05:45][D][sensor:125]: 'Current2': Sending state 0.02347 A with 2 decimals of accuracy
[17:05:50][D][ct_clamp:038]: 'Current1' - Raw AC Value: nanA after 405 different samples (2025 SPS)
[17:05:50][D][sensor:125]: 'Current1': Sending state nan A with 2 decimals of accuracy
[17:05:50][D][ct_clamp:038]: 'Current2' - Raw AC Value: nanA after 432 different samples (2160 SPS)
[17:05:50][D][sensor:125]: 'Current2': Sending state nan A with 2 decimals of accuracy
[17:05:55][D][ct_clamp:038]: 'Current1' - Raw AC Value: 0.001A after 408 different samples (2040 SPS)
[17:05:55][D][sensor:125]: 'Current1': Sending state 0.02202 A with 2 decimals of accuracy
[17:05:55][D][ct_clamp:038]: 'Current2' - Raw AC Value: 0.000A after 435 different samples (2175 SPS)
[17:05:55][D][sensor:125]: 'Current2': Sending state 0.00981 A with 2 decimals of accuracy
[17:06:00][D][sensor:125]: 'adc_sensor2': Sending state 1.65800 V with 2 decimals of accuracy
[17:06:00][D][ct_clamp:038]: 'Current1' - Raw AC Value: nanA after 394 different samples (1970 SPS)
[17:06:00][D][sensor:125]: 'Current1': Sending state nan A with 2 decimals of accuracy
[17:06:00][D][ct_clamp:038]: 'Current2' - Raw AC Value: 0.001A after 414 different samples (2070 SPS)
[17:06:00][D][sensor:125]: 'Current2': Sending state 0.02131 A with 2 decimals of accuracy
[17:06:05][D][ct_clamp:038]: 'Current1' - Raw AC Value: nanA after 387 different samples (1935 SPS)
[17:06:05][D][sensor:125]: 'Current1': Sending state nan A with 2 decimals of accuracy
[17:06:05][D][ct_clamp:038]: 'Current2' - Raw AC Value: 0.001A after 436 different samples (2180 SPS)
[17:06:05][D][sensor:125]: 'Current2': Sending state 0.04423 A with 2 decimals of accuracy
[17:06:10][D][ct_clamp:038]: 'Current1' - Raw AC Value: 0.000A after 397 different samples (1985 SPS)
[17:06:10][D][sensor:125]: 'Current1': Sending state 0.00833 A with 2 decimals of accuracy
[17:06:10][D][ct_clamp:038]: 'Current2' - Raw AC Value: 0.001A after 429 different samples (2145 SPS)
[17:06:10][D][sensor:125]: 'Current2': Sending state 0.03889 A with 2 decimals of accuracy
[17:06:10][D][sensor:125]: 'adc_sensor1': Sending state 1.65800 V with 2 decimals of accuracy
[17:06:15][D][ct_clamp:038]: 'Current1' - Raw AC Value: 0.001A after 388 different samples (1940 SPS)
[17:06:15][D][sensor:125]: 'Current1': Sending state 0.02492 A with 2 decimals of accuracy
[17:06:15][D][ct_clamp:038]: 'Current2' - Raw AC Value: 0.001A after 439 different samples (2195 SPS)
[17:06:15][D][sensor:125]: 'Current2': Sending state 0.02928 A with 2 decimals of accuracy
```

Here is my configuration:
```
sensor:
  - platform: ct_clamp
    sensor: adc_sensor1
    name: "Current1"
    update_interval: 5s
    filters:
    - calibrate_linear:
      - 0.0000 -> 0.0
      - 0.0124 -> 0.397
  
  - platform: adc
    id: adc_sensor1
    pin: 36
    attenuation: 11db

#############################

  - platform: ct_clamp
    sensor: adc_sensor2
    name: "Current2"
    update_interval: 5s
    filters:
    - calibrate_linear:
      - 0.000 -> 0.0
      - 0.0124 -> 0.390

  - platform: adc
    id: adc_sensor2
    pin: 39
    attenuation: 11db

```

avanti1, 2022-06-23:
Happens to me, too. Only with small values. When the monitored device is turned off completely, 99% of the returned values are NaN.

gtakarting, 2022-10-20:
Hi Happens to me at low current, I belive it's some to do with the filter part, had anyone sorted this out ?

closed 2023-03-20 by jesserockz