esphome/issues #2703: Fix incorrect GPIO for M5Stack Camera Pixel Data Bit 0.

markusg1234, 2021-11-12:
### The problem

Fix for incorrect GPIO for M5Stack Camera Pixel Data Bit 0.

### Which version of ESPHome has the issue?

V2021.12.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.3

### What platform are you using?

ESP32

### Component causing the issue

esphome-docs

### Additional information

https://github.com/markusg1234/esphome-docs/commit/a9ff86936317a15e46408106a4fb25a88ff737be

oxan, 2021-11-14:
Please open a PR in the [esphome-docs repository](https://github.com/esphome/esphome-docs) with your proposed change.

markusg1234, 2021-11-14:
Thankyou, will try again. My first pr.

markusg1234, 2021-11-15:
Further investigation, this camera in the docs is the non PSRAM. I will put a PR in for the PSRAM camera.

closed 2021-11-15 by markusg1234