esphome/issues #2704: compilation terminated fatal error: Wire.h: No such file or directory

CRZTFR, 2021-11-12:
### The problem

I'm trying to create custom I2C sensor components. Since the update to 10.3 the custom sensors don't compile due to not being able to find Wire.h. Wire is listed in the libdeps and there is an i2c: initialisation in the main yaml. 

### Which version of ESPHome has the issue?

v2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.3

### What platform are you using?

ESP32

### Board

Devkit C

### Component causing the issue

Custom i2c sensor

### Example YAML snippet

```yaml
esphome:
  name: greenhouse_sensor_platform
  platform: ESP32
  board: esp-wrover-kit
  includes:
  - veml6075.h
  # - spectral_triad.h
  libraries:
  - "SparkFun VEML6075 Arduino Library"
  # - "SparkFun Spectral Triad AS7265X"

wifi:
  ssid: !secret network_ssid
  password: !secret password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Greenhouse-Sensor-Platform"
    password: !secret password

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

i2c:
  sda: 21
  scl: 22
  scan: True

sensor:
  - platform: scd30
    co2:
      name: "CO2"
      accuracy_decimals: 1
    temperature:
      name: "External Temperature"
      accuracy_decimals: 2
    humidity:
      name: "External Humidity"
      accuracy_decimals: 1
    address: 0x61
    update_interval: 30s
    automatic_self_calibration: true
    
  - platform: custom
    lambda: |-
      auto VEML6075 = new VEML6075Sensor();
      App.register_component(VEML6075);
      return {VEML6075->uva_sensor, VEML6075->uvb_sensor, VEML6075->uv_index_sensor};
  
    sensors:
    - name: "UVA - 350-375nm"
      unit_of_measurement: W/m2
      accuracy_decimals: 2
    - name: "UVB - 315-340nm"
      unit_of_measurement: W/m2
      accuracy_decimals: 2
    - name: "UV Index"
      unit_of_measurement: " "
      accuracy_decimals: 1

/config/esphome/veml6075.h
#include "esphome.h"
#include <SparkFun_VEML6075_Arduino_Library.h>

class VEML6075Sensor : public PollingComponent {
 public:
  VEML6075 uv;
  Sensor *uva_sensor = new Sensor();
  Sensor *uvb_sensor = new Sensor();
  Sensor *uv_index_sensor = new Sensor();
  
  VEML6075Sensor() : PollingComponent(15000) { }
  
  float get_setup_priority() const override { return esphome::setup_priority::BUS; }
  
  void setup() override {
    // This will be called by App.setup()
    Wire.begin();
    if(uv.begin() == false){
        Serial.println("Unable to communicate with VEML6075");
    }
    uv.setIntegrationTime(VEML6075::IT_200MS);
    uv.setHighDynamic(VEML6075::DYNAMIC_HIGH);
  }
  void update() override {
    // This will be called by App.loop()
    float uva = uv.uva();
    uva_sensor->publish_state(uva);
    
    float uvb = uv.uvb();
    uvb_sensor->publish_state(uvb);
    
    float uv_index = uv.index();
    uv_index_sensor->publish_state(uv_index);
    
  }
};
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/greenhouse-sensor-platform.yaml...
WARNING 'greenhouse_sensor_platform': Using the '_' (underscore) character in the hostname is discouraged as it can cause problems with some DHCP and local name services. For more information, see https://esphome.io/guides/faq.html#why-shouldn-t-i-use-underscores-in-my-device-name
INFO Generating C++ source...
INFO Compiling app...
Processing greenhouse_sensor_platform (board: esp-wrover-kit; framework: arduino; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <SparkFun VEML6075 Arduino Library> 1.1.4
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.0.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <Wire> 1.0.1
|-- <ESPmDNS> 1.0
Compiling /data/greenhouse_sensor_platform/.pioenvs/greenhouse_sensor_platform/src/main.cpp.o
Archiving /data/greenhouse_sensor_platform/.pioenvs/greenhouse_sensor_platform/lib528/libAsyncTCP-esphome.a
Compiling /data/greenhouse_sensor_platform/.pioenvs/greenhouse_sensor_platform/lib63b/SparkFun VEML6075 Arduino Library/SparkFun_VEML6075_Arduino_Library.cpp.o
Compiling /data/greenhouse_sensor_platform/.pioenvs/greenhouse_sensor_platform/lib64d/WiFi/WiFi.cpp.o
Compiling /data/greenhouse_sensor_platform/.pioenvs/greenhouse_sensor_platform/lib64d/WiFi/WiFiAP.cpp.o
In file included from /data/greenhouse_sensor_platform/.piolibdeps/greenhouse_sensor_platform/SparkFun VEML6075 Arduino Library/src/SparkFun_VEML6075_Arduino_Library.cpp:26:0:
/data/greenhouse_sensor_platform/.piolibdeps/greenhouse_sensor_platform/SparkFun VEML6075 Arduino Library/src/SparkFun_VEML6075_Arduino_Library.h:36:18: fatal error: Wire.h: No such file or directory

**************************************************************
* Looking for Wire.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:Wire.h"
* Web  > https://platformio.org/lib/search?query=header:Wire.h
*
**************************************************************

compilation terminated.
*** [/data/greenhouse_sensor_platform/.pioenvs/greenhouse_sensor_platform/lib63b/SparkFun VEML6075 Arduino Library/SparkFun_VEML6075_Arduino_Library.cpp.o] Error 1
========================= [FAILED] Took 15.11 seconds =========================
```

oxan, 2021-11-13:
Try adding `Wire` to the `includes`.

CRZTFR, 2021-11-15:
Solved by adding `Wire` under `libraries` in the esphome yaml. If anyone else comes across this, `Wire` has to be above the other platformio libraries: 

```
esphome:
  name: greenhouse_sensor_platform
  platform: ESP32
  board: esp-wrover-kit
  includes:
  - veml6075.h
  - spectral_triad.h
  libraries:
  - "Wire"
  - "SparkFun VEML6075 Arduino Library"
  - "SparkFun Spectral Triad AS7265X"
```

closed 2021-11-15 by CRZTFR