esphome/issues #2705: esp32_camera_web_server not working with Motioneye unless

markusg1234, 2021-11-13:
### The problem

Previously I have used ESP32 cam's with https://github.com/easytarget/esp32-cam-webserver to use with Motioneye.

I have found that Motioneye times out if trying to find the new esp32_camera_web_server with esphome.
The workaround I have been able to get to work is 

1) Program ESP32 cam web_server as 
   esp32_camera_web_server:
  - port: 8080
    mode: stream
  - port: 8081
    mode: snapshot 
2) Add network camera into motioneye as http://ipaddress:8081/ and it will find as a jpeg camera. No picture is displayed.
3) Turn off motion eye video device for new camera.
4) Reprogram the ESP32 cam web_server to
   esp32_camera_web_server:
  - port: 8081
    mode: stream
  - port: 8080
    mode: snapshot
5) Turn on motioneye video device.

Now the video stream works in Motioneye.

### Which version of ESPHome has the issue?

v2021.12.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.3

### What platform are you using?

ESP32

### Board

ESP32-cam

### Component causing the issue

esp32_camera_web_server 

### Example YAML snippet

```yaml
esphome:
  name: esp32cam
  platform: ESP32
  board: esp32cam

wifi:
  ssid: "*"
  password: "*"

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: "*"

ota:
  password: "*"
  

esp32_camera_web_server:
  - port: 8081
    mode: stream
  - port: 8080
    mode: snapshot 

esp32_camera:
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32
  resolution: XGA
  max_framerate: 10fps
 

  # Image settings
  name: My Camera
  # ...
  
sensor:
 - platform: wifi_signal
   name: esp32cam Wifi Signal 
   
output:
   - platform: gpio
     pin: GPIO4
     id: gpio_4 

switch:
   - platform: gpio
     pin: GPIO14
     name: "Test"
     

light:
  - platform: binary
    output: gpio_4
    name: esp32 cam light
```

### Anything in the logs that might be useful for us?

```txt
No.
```

ayufan, 2021-11-17:
Is there some easy way to run Motioneye?

coolduke666, 2021-11-18:
Ran motioneye as hassos add-on. Works like a charm but does not recognize my esp32 camera webserver with the latest esphome 2021.11

ayufan, 2021-11-18:
I have it reproduced. Will try to fix it.

coolduke666, 2021-11-19:
If you have a fix - I can give it a try

ayufan, 2021-11-21:
https://github.com/esphome/esphome/pull/2777

closed 2021-11-21 by jesserockz