esphome/issues #2709: Modbus Switch use write_multiple_command for writing

Jopand, 2021-11-14:
### The problem

Some systems only support the "write multiple registers" (modbus function 16) feature - even though only one register is to be written.
That is the case for the Nilan ventilation system.
The Modbus Switch component only uses write_single_command, which makes it unusable for the Nilan ventilation system

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Example YAML snippet

```yaml
switch:
  - platform: modbus_controller
    modbus_controller_id: nilan_modbus_controller
    id: nilan_on_off_state
    register_type: holding
    force_new_range: true
    address: 1000
    name: "On/Off state"
```

### Anything in the logs that might be useful for us?

```txt
[21:03:51][D][switch:017]: 'On/Off state' Turning OFF.
[21:03:51][D][switch:037]: 'On/Off state': Sending state OFF
[21:03:51][D][modbus:105]: Modbus error function code: 0x86 exception: 1
[21:03:51][E][modbus_controller:058]: Modbus error function code: 0x6 exception: 1 
[21:03:51][E][modbus_controller:067]: Modbus error - last command: function code=0x6  register adddress = 0x3E8  registers count=1 payload size=2
[21:03:51][D][modbus:105]: Modbus error function code: 0x86 exception: 1
[21:03:51][E][modbus_controller:058]: Modbus error function code: 0x6 exception: 1
```

martgras, 2021-11-14:
https://github.com/esphome/esphome/pull/2680 adds a write_lambda option.
Not the most elegant solution but allows you to use any modbus function code.
https://github.com/esphome/esphome-docs/blob/0bb007ca6cacbbd86543ab9ab2f3f3333ae56f23/components/switch/modbus_controller.rst shows how to use it. 

If you want to test use. 
```
external_components:
  # - source: github://martgras/esphome@modbus-customcmd
  - source: github://pr#2680
    components: ["modbus_controller"]
```
Let me know if it works for you

Jopand, 2021-11-15:
I would call that a workaround.
1: It would work for people, who knows c++ programming.
2: It would make the yaml file larger (much harder to understand)

martgras, 2021-11-15:
yes agreed 

For testing I added an option `use_write_multiple: True`  to switches 
(I don't like that name  too much - any better suggestion ) 
```
external_components:
   - source: github://martgras/esphome@test-multiplecoil
      components: ["modbus_controller"]
```
Does that work for you?

Jopand, 2021-11-15:
Yes, it seems to work 👍

closed 2021-11-25 by jesserockz