esphome/issues #2710: .h file custom sensor no longer compliling

Morbidbystandard, 2021-11-14:
### The problem

After updating to 2021.10.3 I can no longer compile and install my custom sensor esp 32 code. I get an error in ESPhome about the .h file and failed install message.
`Compiling /data/controller/.pioenvs/controller/src/main.cpp.o
Compiling /data/controller/.pioenvs/controller/FrameworkArduino/esp32-hal-dac.c.o
In file included from src/main.cpp:54:0:
src/uart_com.h: In member function 'virtual void scaleSensor::loop()':
src/uart_com.h:14:41: error: 'readStringUntil' was not declared in this scope
       String line = readStringUntil('\n');
                                         ^
Compiling /data/controller/.pioenvs/controller/FrameworkArduino/esp32-hal-gpio.c.o
Compiling /data/controller/.pioenvs/controller/FrameworkArduino/esp32-hal-i2c.c.o
*** [/data/controller/.pioenvs/controller/src/main.cpp.o] Error 1
========================== [FAILED] Took 5.01 seconds ==========================`

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.3 and supervisor-2021.10.8

### What platform are you using?

ESP32

### Board

Esp32 wrover

### Component causing the issue

custom uart sensor

### Example YAML snippet

```yaml
esphome:
  name: controller
  platform: ESP32
  board: esp-wrover-kit
  includes: uart_com.h

wifi:
  ssid: "****"
  password: "******"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "******"
    password: "******"

captive_portal:

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:

uart:
  id: uart_bus
  tx_pin: GPIO17
  rx_pin: GPIO16
  baud_rate: 9600
  
sensor:
- platform: custom
  lambda: |-
    auto EZO = new scaleSensor(id(uart_bus));
    App.register_component(EZO);
    return {EZO};
  sensors:
    name: "Ph Sensor"
    unit_of_measurement: Ph
    accuracy_decimals: 2
    filters:
      - offset: +0.22
    
switch:
  - platform: gpio
    pin: 32
    id: ph_relay
    inverted: yes
  - platform: template
    name: "Relay 1"
    icon: "mdi:trending-up"
    lambda: 'return id(ph_relay).state;'
    turn_on_action:
    - switch.turn_on: ph_relay
    - delay: 4000ms
    - switch.turn_off: ph_relay
    
  - platform: gpio
    pin: 33
    name: "Relay 2"
    inverted: yes
    
  - platform: gpio
    pin: 25
    name: "Relay 3"
    inverted: yes
    
  - platform: gpio
    pin: 26
    name: "Relay 4"
    inverted: yes
    
  - platform: gpio
    pin: 27
    name: "Relay 5"
    inverted: yes
    
  - platform: gpio
    pin: 14
    name: "Relay 6"
    inverted: yes
    
  - platform: gpio
    pin: 12
    name: "Relay 7"
    inverted: yes
    
  - platform: gpio
    pin: 13
    name: "Relay 8"
    inverted: yes
    
  - platform: uart
    name: "Mid Point Calibration"
    data: 'cal,mid,7'
    
  - platform: uart
    name: "Low Point Calibration"
    data: 'cal,low,4'
    
  - platform: uart
    name: "High Point Calibration"
    data: 'cal,high,10'
```

### Anything in the logs that might be useful for us?

```txt
#include "esphome.h"

class scaleSensor : public PollingComponent, public UARTDevice, public sensor::Sensor{
 public:
  scaleSensor(UARTComponent *parent) : PollingComponent(1000), UARTDevice(parent) {}
  float weight = 0;
  
  void setup() override {
    // nothing to do here
  }
  
  void loop() override {
    while (available()) {
      String line = readStringUntil('\n');
      weight = line.toFloat();
    }
  }
  
  void update() {
    publish_state(weight);
  }
};
```

oxan, 2021-11-14:
`UARTDevice` no longer inherits from the Arduino `Stream` class, so you need to change your code to use the [new API](https://github.com/esphome/esphome/blob/dev/esphome/components/uart/uart.h).

fabiojdsoliveira, 2022-01-02:
> `UARTDevice` no longer inherits from the Arduino `Stream` class, so you need to change your code to use the [new API](https://github.com/esphome/esphome/blob/dev/esphome/components/uart/uart.h).

You make it look so simple, and I do not know how to do it. Why change something that was working and do not give a concrete example of how to do it?

mmakaay, 2022-01-02:
The reason for changing things is normally: to move the code base forward. Keeping an API exactly the same normally leads to complexity in the long run. The reason for that is simply that adding features without deprecating old features increases the total amount of code. Old code has to be maintained next to new code.
In general API's are kept as-is, but with 2021.10.0, big changes had to be done in favor of making the code support direct use of ESP-IDF instead of using the Arduino library layer, for ESP32.

fabiojdsoliveira, 2022-01-05:
Thank you all. I thought it would be more difficult but after experimenting with the code I was able to get the program working.
My code changed from:

```
#include "esphome.h"

class myenergymetersensor : public PollingComponent, public UARTDevice, public Sensor   {
 public:
 
  // constructor - where we “initialize” the custom sensor
  myenergymetersensor(UARTComponent *parent) : UARTDevice(parent), PollingComponent(5000){}
  
  Sensor *activepower_sensor1    = new Sensor();
  Sensor *apparentpower_sensor1  = new Sensor();
  Sensor *voltage_sensor1        = new Sensor();
  Sensor *current_sensor1        = new Sensor();
  Sensor *powerfactor_sensor1    = new Sensor();  

  Sensor *activepower_sensor2    = new Sensor();
  Sensor *apparentpower_sensor2  = new Sensor();
  Sensor *voltage_sensor2        = new Sensor();
  Sensor *current_sensor2        = new Sensor();
  Sensor *powerfactor_sensor2    = new Sensor();  

  void setup() override {
    // This will be called by App.setup()
    setTimeout(2500);
  }

  void loop() override {
    // Use Arduino API to read data, for example
  }

  void update() override {
    // This will be called every "update_interval" milliseconds.
    
        float activepower1, activepower2;
        float apparentpower1, apparentpower2;
        float voltage1, voltage2;
        float current1, current2;
        float powerfactor1, powerfactor2;    

        String string_activepower;
        String string_apparentpower;
        String string_voltage;
        String string_current;
        String string_powerfactor;
        
    while(available()) read();
    
    String line = readStringUntil('\n');
    
    if(line.length()==70){
        string_activepower   = line.substring(0,6);
        string_apparentpower = line.substring(7,13);
        string_voltage       = line.substring(14,20);
        string_current       = line.substring(21,28);
        string_powerfactor   = line.substring(29,34);
    
        activepower1   = string_activepower.toFloat();
        apparentpower1 = string_apparentpower.toFloat();
        voltage1       = string_voltage.toFloat();
        current1       = string_current.toFloat();
        powerfactor1   = string_powerfactor.toFloat();
        
        activepower_sensor1      -> publish_state(activepower1);
        apparentpower_sensor1    -> publish_state(apparentpower1);
        voltage_sensor1          -> publish_state(voltage1);
        current_sensor1          -> publish_state(current1);
        powerfactor_sensor1      -> publish_state(powerfactor1);
[... 108 lines cut ...]
        powerfactor_sensor1      -> publish_state(powerfactor1);
    
        activepower2   = line.substring(35,41).toFloat();
        apparentpower2 = line.substring(42,48).toFloat();
        voltage2       = line.substring(49,55).toFloat();
        current2       = line.substring(56,63).toFloat();
        powerfactor2   = line.substring(64,69).toFloat();
        
        activepower_sensor2      -> publish_state(activepower2);
        apparentpower_sensor2    -> publish_state(apparentpower2);
        voltage_sensor2          -> publish_state(voltage2);
        current_sensor2          -> publish_state(current2);
        powerfactor_sensor2      -> publish_state(powerfactor2);        
        
    }else{
        //Does nothing
    }
  }
};

```

First tried reading directly to a string cariable, but it was not working, so I reformulated to using an array of char.

Thank you again.

closed 2022-01-06 by jesserockz

mmakaay, 2022-01-06:
Thanks for reporting back with your updated code. That can help others that run into the same issue.