esphome/issues #2711: Unable to Compile for ESP32

roverte, 2021-11-15:
### The problem

I've been using ESPHome with ESP8266 devices but cannot get it to work with an ESP32.  Specifically, when I try to compile the most basic code -- just connecting to my WiFi -- the compiling fails. I get an error "cannot find -lstdc++" (full logs below).

I've tried restarting my server multiple times and created several new devices to try compiling.  Nothing works for the ESP32.  I can still compile, upload and use ESP8266 devices without any issues.

Any ideas how to fix this?
Thanks,
Alex

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.10.8

### What platform are you using?

ESP32

### Board

ESP-WROOM-32

### Example YAML snippet

```yaml
esphome:
  name: ldksjaslkdfj
  platform: ESP32
  board: esp32dev

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "e636c7f31c40fa7e5eec217bb405655a"

wifi:
  ssid: "XXXXXX"
  password: "XXXXXX"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ldksjaslkdfj Fallback Hotspot"
    password: "G88I5SuakmZr"

captive_portal:
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/ldksjaslkdfj.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing ldksjaslkdfj (board: esp32dev; framework: arduino; platform: espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
[... 110 lines cut ...]
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/esp32-hal-spi.c.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/esp32-hal-time.c.o
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/esp32-hal-timer.c.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/esp32-hal-touch.c.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/esp32-hal-uart.c.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/libb64/cdecode.c.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/libb64/cencode.c.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/main.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/stdlib_noniso.c.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/wiring_pulse.c.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/FrameworkArduino/wiring_shift.c.o
Archiving /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/libFrameworkArduino.a
Linking /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
*** [/data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/firmware.elf] Error 1
========================= [FAILED] Took 93.08 seconds =========================
```

dixanl, 2021-11-19:
also such a problem

mmakaay, 2021-11-20:
Can you please retry this with the latest versions of ESPHome and Home Assistant?
I tried the exact configuration from above, and I could compile without issues using my command line ESPHome setup.

roverte, 2021-11-20:
I’m unfortunately still having the same issue after updating.  I’m running:
ESPHome 2021.11.1
Core 2021.11.4
Supervisor 2021.10.8

My HA is running on a Raspberry Pi 4.

Any other ideas on what I could try?

roverte, 2021-11-26:
I'm still having this issue with ESPHome 2021.11.2 on Core 2021.11.5.  The latest logs are pasted below.  Would really appreciate any help on how to solve this issue as I don't know where to begin trying to debug this one.

Thanks!

`INFO Reading configuration /config/esphome/ldksjaslkdfj.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing ldksjaslkdfj (board: esp32dev; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <AsyncTCP-esphome> 1.2.2
|-- <WiFi> 1.0
|-- <FS> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ESPmDNS> 1.0
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/api_connection.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/api_server.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/list_entities.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/proto.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/user_services.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/api/util.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/esp32/core.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/esp32/preferences.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/logger/logger.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/md5/md5.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/network/util.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/ota/ota_backend_arduino_esp32.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/ota/ota_backend_arduino_esp8266.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/ota/ota_backend_esp_idf.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/ota/ota_component.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/socket/bsd_sockets_impl.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/socket/lwip_raw_tcp_impl.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/web_server_base/web_server_base.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/wifi/wifi_component.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/wifi/wifi_component_esp32_arduino.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/wifi/wifi_component_esp8266.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/components/wifi/wifi_component_esp_idf.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/application.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/color.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/component.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/controller.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/entity_base.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/helpers.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/log.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/scheduler.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/esphome/core/util.cpp.o
Compiling /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/src/main.cpp.o
Linking /data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/firmware.elf
/data/cache/platformio/packages/toolchain-xtensa32/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lstdc++
collect2: error: ld returned 1 exit status
*** [/data/ldksjaslkdfj/.pioenvs/ldksjaslkdfj/firmware.elf] Error 1
========================= [FAILED] Took 51.28 seconds =========================`

paarak, 2021-11-27:
Hi I got the same error message. I took a backup of the ESPHOME folder before deleting all content. made a new confg uploaded over the air then everything worked fine!

roverte, 2021-11-27:
Thanks paarak -- I followed your suggestion and now it is compiling and uploading.  Thanks!

paarak, 2021-11-28:
> Thanks paarak -- I followed your suggestion and now it is compiling and uploading. Thanks!

Good. then you can close issue.

closed 2021-11-28 by roverte