esphome/issues #2712: Attribute invalid for xiaomi_lywsd03mmc

jli113, 2021-11-16:
### The problem

how should I set attribute for lywsd03mmc?
If I put it under platform, it has error that "attribute is an invalid option for sensor.xiaomi_lywsd03mmc"
platform: xiaomi_lywsd03mmc
mac_address: "A4:C1:38:B1:CD:7F"
bindkey: "eef418daf699a0c188f3bfd17e4565d9"
temperature:
name: "LYWSD03MMC Temperature"
humidity:
name: "LYWSD03MMC Humidity"
battery_level:
name: "LYWSD03MMC Battery Level"
attribute: currenttemperature
If put under temperature, it has error that "attribute is an invalid option for temperature"
temperature:
  name: "LYWSD03MMC Temperature"
  attribute: currenttemperature

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

jesserockz, 2021-11-16:
What are you trying to achieve exactly?
`attribute` is for pulling information **from** home assistant, not for exposing sensor data to it. You cannot add arbitrary attributes to sensors.

jli113, 2021-11-16:
> What are you trying to achieve exactly?
> `attribute` is for pulling information **from** home assistant, not for exposing sensor data to it. You cannot add arbitrary attributes to sensors.

Thanks for explaining. So do I need to create another config file along with the sensor configuration file?
Please show me how to create it.

jesserockz, 2021-11-17:
What are you trying to achieve exactly?

jli113, 2021-11-17:
I have several xiaomi_lywsd03mmc sensors installed, I want to link the live data with a 3D model, but I don't know how to set the sensor attributes.

jesserockz, 2021-11-17:
I still don't understand why you are talking about attributes.

The sensors are exposed to Home Assistant individually and you can use them there however you like.

[image]

jli113, 2021-11-18:
It's fine, I found other methods to visualize it.

closed 2021-11-18 by jli113