esphome/issues #2713: False Integration?

minimag450, 2021-11-16:
### The problem

For one device, the entities are not available. Why is the device 'kinderzimmerheizung' (ESPHome file) an integration of the Fritz!Box and not of ESPHome? see also issues2693? It also displays nodemcuv2. In the yaml I have specified the D1 mini. How can I delete the device so that it can no longer be found at the FritzBox?

### Which version of ESPHome has the issue?

2021.10.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.3

### What platform are you using?

ESP8266

### Board

D1 mini

### Component causing the issue

none

### Example YAML snippet

```yaml
esphome:
  name: heizungkinderzimmer
  platform: ESP8266
  board: d1_mini

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  networks:
   ssid: "FRITZ!Box 7490"
   password: "mypassword"
  ap:
    ssid: "TestHotspot"
    password: "andreas8"
  
  
  manual_ip:
   static_ip: 192.168.178.222
   gateway: 192.168.178.1
   subnet: 255.255.255.0
   
   
switch:
  - platform: gpio
    pin: 2
    name: "HeizungKinderzimmer"
```

### Anything in the logs that might be useful for us?

```txt
[... 294 lines cut ...]
[13:43:12][C][wifi:356]:   BSSID: [redacted]
[13:43:12][C][wifi:357]:   Hostname: 'heizungkinderzimmer'
[13:43:12][C][wifi:359]:   Signal strength: -90 dB ▂▄▆█
[13:43:12][C][wifi:363]:   Channel: 1
[13:43:12][C][wifi:364]:   Subnet: 255.255.255.0
[13:43:12][C][wifi:365]:   Gateway: 192.168.178.1
[13:43:12][C][wifi:366]:   DNS1: 0.0.0.0
[13:43:12][C][wifi:367]:   DNS2: 0.0.0.0
[13:43:12][C][switch.gpio:048]: GPIO Switch 'HeizungKinderzimmer'
[13:43:12][C][switch.gpio:049]:   Pin: GPIO2
[13:43:12][C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[13:43:12][C][logger:233]: Logger:
[13:43:12][C][logger:234]:   Level: DEBUG
[13:43:12][C][logger:235]:   Log Baud Rate: 115200
[13:43:12][C][logger:236]:   Hardware UART: UART0
[13:43:12][C][ota:082]: Over-The-Air Updates:
[13:43:12][C][ota:083]:   Address: 192.168.178.222:8266
[13:43:12][C][api:134]: API Server:
[13:43:12][C][api:135]:   Address: 192.168.178.222:6053
[13:43:12][C][api:139]:   Using noise encryption: NO
```

### Additional information

[image]

mmakaay, 2021-11-16:
Maybe the FRITZ!Box integration that you see is device tracking, provided by the FRITZ!Box?
I don't know why the deivce itself is not connected via the ESPHome API. This can be because of a number of reasons.
Could you try adding an ESPHome integration manually, using `192.168.178.222` or `heizungkinderzimmer.local` as the hostname?

minimag450, 2021-11-16:
Oh i'm sorry.  I don't know how to manually add an ESPHome integration.

mmakaay, 2021-11-16:
In Home Assistant, you go to Configuration -> Integrations.
Then click the "Add Integration" button at the bottom right of the screen.
You get a popup, where it's best to start typing "esphome" in the search box.
Then click the "ESPhome" item from the list.
You'll get a new popup, asking for a host and port.
In the host field, you can enter de hostname or IP address of you ESPhome device that you want to integrate.

minimag450, 2021-11-17:
Thanks!  I didn't know until now that it would work that way.  But then there is an error message, as can be seen in the picture.

[image]

mmakaay, 2021-11-17:
That suggests that the API is not available, however, it can be seen in your log output that the API has been setup:
```
[13:43:12][C][api:134]: API Server:
[13:43:12][C][api:135]:   Address: 192.168.178.222:6053
[13:43:12][C][api:139]:   Using noise encryption: NO
```

In the rest of the configuration, it looks like the WiFi connection is not very stable. There are some errors and multiple connect attempts. What you might want to try to test this hypothesis, is starting a ping to 192.168.178.222 to see if it responds consistently. I don't know that kind of system you are using. Your screenshot suggests a tablet, so in that case you'd have to install a ping app. On windows, open a command line terminal, and run `PING 192.168.178.222`.

minimag450, 2021-11-17:
[image]

mmakaay, 2021-11-17:
Okay, so you are getting timeouts for ping here.
If the device still has the 222 IP-address assigned to it, then it looks like the main issue here is a broken WiFi connection.
Because of that, Home Assistant will not be able to contact the API and therefore the integration will not be available there.

It's hard to tell why your WiFi is not working. Some options:
- Out of range of the WiFi router. Check if the device works if it's near your WiFi router to check this.
- The ESP-device WiFi is not functioning correctly. This can be checked by using a different ESP-device.
- Maybe the WiFi router has an issue. We've seen cases where rebooting the router solved connection issues.
Some additional options can be found [in the ESPHome FAQ](https://esphome.io/guides/faq.html#wifi-problems)

minimag450, 2021-11-17:
many thanks for your help!  Now I will first test your suggestions.  Would be nice if I finally got the problem solved.

mmakaay, 2021-11-27:
Managed to get any further with this?

minimag450, 2021-11-27:
oh yes, it helped me a lot. And I learned something again. Thank you once again! Thanks!
Many regards! Andreas

mmakaay, 2021-11-28:
Okay, great!
Then I will close this issue now.

closed 2021-11-28 by mmakaay