esphome/issues #2716: Atlas Ezo sensors noot report the value

egranto, 2021-11-17:
### The problem

today i update one of my builds and after the update i notice my ezo sensor both are coming with 0

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

ezo

### Example YAML snippet

```yaml
- platform: ezo
    id: eph
    name: "${upper_devicename} Agua pH"
    address: 99
    accuracy_decimals: 2
    unit_of_measurement: "pH" 
    filters:
    - offset: ${offset_ph}
    update_interval: 60s
```

### Anything in the logs that might be useful for us?

```txt
[06:18:03][I][i2c.arduino:048]: Scanning i2c bus for active devices...
[06:18:03][I][i2c.arduino:053]: Found i2c device at address 0x63
[06:18:03][I][i2c.arduino:053]: Found i2c device at address 0x64

[06:18:43][D][sensor:113]: 'R.Alita-0 Agua E.C': Sending state 0.00000 μS/cm with 2 decimals of accuracy
[06:18:44][D][sensor:113]: 'R.Alita-0 Agua pH': Sending state 0.00000 pH with 2 decimals of accuracy
```

### Additional information

was working on the with the previous build, this  was the first i upgrade my first board  

 
[image]

[image]

optikkore, 2021-11-17:
Same did the update to esphome today on home assistant and now after updating the sensor i get this
[15:37:53][D][sensor:113]: 'ph_ezo': Sending state 0.00000 pH with 2 decimals of accuracy

the bme280 is reporting properly though

esphome:
name: test-board
platform: ESP32
board: esp32dev

Enable logging
logger:

Enable Home Assistant API
api:

ota:
password: "....."

wifi:
ssid: "....."
password: "......"

Enable fallback hotspot (captive portal) in case wifi connection fails
ap:
ssid: "Test-Board Fallback Hotspot"
password: "....."

captive_portal:

i2c:
sda: 23
scl: 22
scan: true
id: bus_a
frequency: 100kHz

tca9548a:

address: 0x70
id: multiplex0
i2c_id: bus_a
channels:
bus_id: multiplex0channel0
channel: 0
bus_id: multiplex0channel1
channel: 1
sensor:

platform: bme280
i2c_id: multiplex0channel0
temperature:
name: "BME280 Temperature"
oversampling: 16x
pressure:
name: "BME280 Pressure"
humidity:
name: "BME280 Humidity"
address: 0x76
iir_filter: 16x
update_interval: 60s

platform: adc
pin: 35
name: "BME280 Battery"
update_interval: 60s
attenuation: 11db
filters:

multiply: 2.0
platform: ezo
i2c_id: multiplex0channel1
name: "ph_ezo"
id: ph_ezo
address: 99
unit_of_measurement: "pH"
accuracy_decimals: 2
update_interval: 10s

optikkore, 2021-11-17:
Also getting this error when building now

/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
                                       ^
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^

ssieb, 2021-11-18:
That's a warning, not an error.  It's been happening for a while now.  It's in the underlying platform, so there's nothing we can do.  You can ignore it.

ssieb, 2021-11-18:
Can you get a VERY_VEBOSE log of one sample interval?

optikkore, 2021-11-18:
@ssieb 
[23:46:39][VV][scheduler:152]: Running interval 'update' with interval=10000 last_execution=4294966827 (now=9531)
[23:46:39][VV][i2c.arduino:121]: 0x63 TX 52
[23:46:40][VV][i2c.arduino:097]: 0x63 RX 01362E3533360000000000000000000000000000
[23:46:40][V][sensor:062]: 'ph_ezo': Received new state 0.000000
[23:46:40][D][sensor:113]: 'ph_ezo': Sending state 0.00000 pH with 2 decimals of accuracy
[23:46:41][D][api:098]: Accepted IP_ADDRESS
[23:46:41][VV][api.service:291]: on_hello_request: HelloRequest {
  client_info: 'Home Assistant 2021.11.4'
}
[23:46:41][V][api.connection:731]: Hello from client: 'Home Assistant 2021.11.4 (IP_ADDRESS)'
[23:46:41][VV][api.service:013]: send_hello_response: HelloResponse {
  api_version_major: 1
  api_version_minor: 6
  server_info: 'test-board (esphome v2021.11.1)'
}
[23:46:41][VV][api.service:300]: on_connect_request: ConnectRequest {
  password: ''
}
[23:46:41][D][api.connection:747]: Home Assistant 2021.11.4 (192.168.2.80): Connected successfully
[23:46:41][VV][api.service:019]: send_connect_response: ConnectResponse {
  invalid_password: NO
}
[23:46:41][VV][api.service:345]: on_device_info_request: DeviceInfoRequest {}
[23:46:41][VV][api.service:049]: send_device_info_response: DeviceInfoResponse {
  uses_password: NO
  name: 'test-board'
  mac_address: ''
  esphome_version: '2021.11.1'
  compilation_time: 'Nov 17 2021, 23:46:01'
  model: 'featheresp32'
  has_deep_sleep: NO
  project_name: ''
  project_version: ''
  webserver_port: 0
}
[23:46:41][VV][api.service:354]: on_list_entities_request: ListEntitiesRequest {}
[23:46:41][VV][api.service:132]: send_list_entities_sensor_response: ListEntitiesSensorResponse {
  object_id: 'ph_ezo'
  key: 24725120
  name: 'ph_ezo'
  unique_id: 'test-boardsensorph_ezo'
  icon: ''
  unit_of_measurement: 'pH'
  accuracy_decimals: 2
  force_update: NO
  device_class: ''
  state_class: STATE_CLASS_NONE
  legacy_last_reset_type: LAST_RESET_NONE
  disabled_by_default: NO
  entity_category: ENTITY_CATEGORY_NONE
}
[23:46:41][VV][api.service:055]: send_list_entities_done_response: ListEntitiesDoneResponse {}
[23:46:41][VV][api.service:363]: on_subscribe_states_request: SubscribeStatesRequest {}
[23:46:42][VV][api.service:425]: on_subscribe_homeassistant_services_request: SubscribeHomeassistantServicesRequest {}
[23:46:42][VV][api.service:452]: on_subscribe_home_assistant_states_request: SubscribeHomeAssistantStatesRequest {}
[23:46:42][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 24725120
  state: 0
  missing_state: NO
}

ssieb, 2021-11-18:
This appears to be caused by https://github.com/esphome/esphome/pull/2659

egranto, 2021-11-26:
Fixed with Fix parsing numbers from null-terminated buffers esphome/esphome#2755

closed 2021-11-26 by egranto