esphome/issues #2719: Pms5003 returns 0

DavyRoswinkel, 2021-11-17:
### The problem

Since the latest update my pms5003 returns 0 as value when measuring. It worked before updating. Updated today to esphome 2021.11.0 and it still worked. Updated to the last version (esphome 2021.11.1) and the sensor returns 0 as value. Coincidence or something related to the update?

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP8266

### Board

Nodemcuv2

### Component causing the issue

Pmsx003

ssieb, 2021-11-18:
There was only a tiny change between those two versions and nothing that would affect this.  It's definitely not related to #2716.
Are you able to verify those versions again?

DavyRoswinkel, 2021-11-18:
> 
> 
> There was only a tiny change between those two versions and nothing that would affect this. It's definitely not related to #2716. Are you able to verify those versions again?
Maybe I'm mistaking...and the troubles already started at 2021.11.0 Not really sure ( I was pointing to hard, sorry)

ssieb, 2021-11-18:
Can you get a VERY_VERBOSE log of one update cycle?

DavyRoswinkel, 2021-11-18:
[09:46:24][I][app:099]: ESPHome version 2021.11.1 compiled on Nov 18 2021, 09:44:46
[09:46:24][C][wifi:488]: WiFi:
[09:46:24][C][wifi:350]:   Local MAC: C4:5B:BE:54:7A:84
[09:46:24][C][wifi:351]:   SSID: [redacted]
[09:46:24][C][wifi:352]:   IP Address: 192.168.1.176
[09:46:24][C][wifi:354]:   BSSID: [redacted]
[09:46:24][C][wifi:355]:   Hostname: 'airsensor'
[09:46:24][C][wifi:357]:   Signal strength: -44 dB ▂▄▆█
[09:46:24][C][wifi:361]:   Channel: 11
[09:46:24][C][wifi:362]:   Subnet: 255.255.255.0
[09:46:24][C][wifi:363]:   Gateway: 192.168.1.114
[09:46:24][C][wifi:364]:   DNS1: 0.0.0.0
[09:46:24][C][wifi:365]:   DNS2: 0.0.0.0
[09:46:24][C][logger:233]: Logger:
[09:46:24][C][logger:234]:   Level: VERY_VERBOSE
[09:46:24][C][logger:235]:   Log Baud Rate: 115200
[09:46:24][C][logger:236]:   Hardware UART: UART0
[09:46:24][C][uart.arduino_esp8266:095]: UART Bus:
[09:46:24][C][uart.arduino_esp8266:096]:   TX Pin: GPIO12
[09:46:24][C][uart.arduino_esp8266:097]:   RX Pin: GPIO13
[09:46:24][C][uart.arduino_esp8266:099]:   RX Buffer Size: 256
[09:46:24][C][uart.arduino_esp8266:101]:   Baud Rate: 9600 baud
[09:46:24][C][uart.arduino_esp8266:102]:   Data Bits: 8
[09:46:24][C][uart.arduino_esp8266:103]:   Parity: NONE
[09:46:24][C][uart.arduino_esp8266:104]:   Stop bits: 1
[09:46:24][C][uart.arduino_esp8266:108]:   Using software serial
[09:46:24][D][pmsx003:169]: Got PM1.0 Concentration: 0 µg/m^3, PM2.5 Concentration 0 µg/m^3, PM10.0 Concentration: 0 µg/m^3
[09:46:24][V][sensor:062]: 'Particulate Matter <1.0µm Concentration': Received new state 0.000000
[09:46:24][D][sensor:113]: 'Particulate Matter <1.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[09:46:24][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 593117542
  state: 0
  missing_state: NO
}
[09:46:24][V][sensor:062]: 'Particulate Matter <2.5µm Concentration': Received new state 0.000000
[09:46:24][D][sensor:113]: 'Particulate Matter <2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[09:46:24][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 930978678
  state: 0
  missing_state: NO
}
[09:46:24][V][sensor:062]: 'Particulate Matter <10.0µm Concentration': Received new state 0.000000
[09:46:24][D][sensor:113]: 'Particulate Matter <10.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[09:46:24][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 1460261972
  state: 0
  missing_state: NO
}
[09:46:24][V][component:186]: Component pmsx003.sensor took a long time for an operation (0.11 s).
[09:46:24][V][component:187]: Components should block for at most 20-30ms.
[09:46:24][C][switch.gpio:048]: GPIO Switch 'Start measuring'
[09:46:24][C][switch.gpio:049]:   Pin: GPIO0
[09:46:24][C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[09:46:24][C][homeassistant.time:010]: Home Assistant Time:
[09:46:24][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[09:46:24][C][pmsx003:221]: PMSX003:
[09:46:24][C][pmsx003:226]:   PM1.0 'Particulate Matter <1.0µm Concentration'
[09:46:24][C][pmsx003:226]:     State Class: 'measurement'
[09:46:24][C][pmsx003:226]:     Unit of Measurement: 'µg/m³'
[09:46:24][C][pmsx003:226]:     Accuracy Decimals: 0
[09:46:24][C][pmsx003:226]:     Icon: 'mdi:chemical-weapon'
[09:46:24][C][pmsx003:227]:   PM2.5 'Particulate Matter <2.5µm Concentration'
[09:46:24][C][pmsx003:227]:     State Class: 'measurement'
[09:46:24][C][pmsx003:227]:     Unit of Measurement: 'µg/m³'
[09:46:24][C][pmsx003:227]:     Accuracy Decimals: 0
[09:46:24][C][pmsx003:227]:     Icon: 'mdi:chemical-weapon'
[09:46:24][C][pmsx003:228]:   PM10.0 'Particulate Matter <10.0µm Concentration'
[09:46:24][C][pmsx003:228]:     State Class: 'measurement'
[09:46:24][C][pmsx003:228]:     Unit of Measurement: 'µg/m³'
[09:46:24][C][pmsx003:228]:     Accuracy Decimals: 0
[... 99 lines cut ...]
[09:46:27][D][sensor:113]: 'Particulate Matter <2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[09:46:27][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 930978678
  state: 0
  missing_state: NO
}
[09:46:27][V][sensor:062]: 'Particulate Matter <10.0µm Concentration': Received new state 0.000000
[09:46:27][D][sensor:113]: 'Particulate Matter <10.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[09:46:27][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 1460261972
  state: 0
  missing_state: NO
}
[09:46:27][V][component:186]: Component pmsx003.sensor took a long time for an operation (0.10 s).
[09:46:27][V][component:187]: Components should block for at most 20-30ms.
[09:46:27][VV][scheduler:152]: Running interval 'update' with interval=10000 last_execution=10168 (now=20226)
[09:46:27][V][sensor:062]: 'Airsensor WiFi Signal': Received new state -46.000000
[09:46:27][D][sensor:113]: 'Airsensor WiFi Signal': Sending state -46.00000 dBm with 0 decimals of accuracy
[09:46:27][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 3339098154
  state: -46
  missing_state: NO
}

You mean this?? Or do you need more?

ssieb, 2021-11-18:
Sorry, I just realized that the uart logging has been removed.  You can go back to the default logging level, but add:
```yaml
  debug:
    direction: BOTH
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ':');
```
to the `uart:` section.  And then get the log output again.

DavyRoswinkel, 2021-11-18:
> 
> 
> Sorry, I just realized that the uart logging has been removed. You can go back to the default logging level, but add:
> 
> ```yaml
>   debug:
>     direction: BOTH
>     sequence:
>       - lambda: UARTDebug::log_hex(direction, bytes, ':');
> ```
> 
> to the `uart:` section. And then get the log output again.

When trying to install I get this error:
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 817, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 804, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 364, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 156, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 168, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 613, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 148, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/uart/__init__.py", line 185, in to_code
    await debug_to_code(config[CONF_DEBUG], var)
  File "/esphome/esphome/components/uart/__init__.py", line 151, in debug_to_code
    after = config[CONF_AFTER]
KeyError: 'after'

Am I using this piece of code wrong???
uart:
  tx_pin: D6
  rx_pin: D7
  baud_rate: 9600
  debug:
    direction: BOTH
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ':');

ssieb, 2021-11-18:
Use three backquotes on the lines before and after the yaml lines or just select the lines and click the code button `<>`.  If you put `yaml` right after the first backquotes, it will syntax highlight as well.

DavyRoswinkel, 2021-11-18:
> 
> 
> Use three backquotes on the lines before and after the yaml lines or just select the lines and click the code button `<>`. If you put `yaml` right after the first backquotes, it will syntax highlight as well.

I corrected the previous message ( thanks learning everyday)

ssieb, 2021-11-18:
There's a bug in the parsing.  Try:
```yaml
  debug:
    direction: BOTH
    after:
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ':');
```

DavyRoswinkel, 2021-11-18:
> ```yaml
> debug:
>     direction: BOTH
>     after:
>     sequence:
>       - lambda: UARTDebug::log_hex(direction, bytes, ':');
> ```
Failed config

uart: [source /config/esphome/airsensor.yaml:37]
  - [source /config/esphome/airsensor.yaml:38]
    tx_pin: D6
    rx_pin: D7
    baud_rate: 9600
    debug:  [source /config/esphome/airsensor.yaml:42]
      direction: BOTH
      
      expected a dictionary.
      after:  [source /config/esphome/airsensor.yaml:43]
      sequence: 
        - lambda: UARTDebug::log_hex(direction, bytes, ':');

DavyRoswinkel, 2021-11-18:
Used:```
debug:
    direction: BOTH
    after:
      bytes: 60
    sequence:
      - lambda: UARTDebug::log_hex(direction, bytes, ':');
```

Log:
INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.1.176 using esphome API
INFO Successfully connected to 192.168.1.176
[11:11:58][I][app:099]: ESPHome version 2021.11.1 compiled on Nov 18 2021, 11:11:19
[11:11:58][C][wifi:488]: WiFi:
[11:11:58][C][wifi:350]:   Local MAC: C4:5B:BE:54:7A:84
[11:11:58][C][wifi:351]:   SSID: [redacted]
[11:11:58][C][wifi:352]:   IP Address: 192.168.1.176
[11:11:58][C][wifi:354]:   BSSID: [redacted]
[11:11:58][C][wifi:355]:   Hostname: 'airsensor'
[11:11:58][C][wifi:357]:   Signal strength: -44 dB ▂▄▆█
[11:11:58][C][wifi:361]:   Channel: 11
[11:11:58][C][wifi:362]:   Subnet: 255.255.255.0
[11:11:58][C][wifi:363]:   Gateway: 192.168.1.114
[11:11:58][C][wifi:364]:   DNS1: 0.0.0.0
[11:11:58][C][wifi:365]:   DNS2: 0.0.0.0
[11:11:58][C][logger:233]: Logger:
[11:11:58][C][logger:234]:   Level: DEBUG
[11:11:58][C][logger:235]:   Log Baud Rate: 115200
[11:11:58][C][logger:236]:   Hardware UART: UART0
[11:11:58][C][uart.arduino_esp8266:095]: UART Bus:
[11:11:58][C][uart.arduino_esp8266:096]:   TX Pin: GPIO12
[11:11:58][C][uart.arduino_esp8266:097]:   RX Pin: GPIO13
[11:11:58][C][uart.arduino_esp8266:099]:   RX Buffer Size: 256
[11:11:58][C][uart.arduino_esp8266:101]:   Baud Rate: 9600 baud
[11:11:58][C][uart.arduino_esp8266:102]:   Data Bits: 8
[11:11:58][C][uart.arduino_esp8266:103]:   Parity: NONE
[11:11:58][C][uart.arduino_esp8266:104]:   Stop bits: 1
[11:11:58][C][uart.arduino_esp8266:108]:   Using software serial
[11:11:58][C][switch.gpio:048]: GPIO Switch 'Start measuring'
[11:11:58][C][switch.gpio:049]:   Pin: GPIO0
[11:11:58][C][switch.gpio:071]:   Restore Mode: Restore (Defaults to OFF)
[11:11:58][C][homeassistant.time:010]: Home Assistant Time:
[11:11:58][C][homeassistant.time:011]:   Timezone: 'CET-1CEST,M3.5.0,M10.5.0/3'
[11:11:58][C][pmsx003:221]: PMSX003:
[11:11:58][C][pmsx003:226]:   PM1.0 'Particulate Matter <1.0µm Concentration'
[11:11:58][C][pmsx003:226]:     State Class: 'measurement'
[11:11:58][C][pmsx003:226]:     Unit of Measurement: 'µg/m³'
[11:11:58][C][pmsx003:226]:     Accuracy Decimals: 0
[11:11:58][C][pmsx003:226]:     Icon: 'mdi:chemical-weapon'
[11:11:58][C][pmsx003:227]:   PM2.5 'Particulate Matter <2.5µm Concentration'
[11:11:58][C][pmsx003:227]:     State Class: 'measurement'
[11:11:58][C][pmsx003:227]:     Unit of Measurement: 'µg/m³'
[11:11:58][C][pmsx003:227]:     Accuracy Decimals: 0
[11:11:58][C][pmsx003:227]:     Icon: 'mdi:chemical-weapon'
[11:11:58][C][pmsx003:228]:   PM10.0 'Particulate Matter <10.0µm Concentration'
[11:11:58][C][pmsx003:228]:     State Class: 'measurement'
[11:11:58][C][pmsx003:228]:     Unit of Measurement: 'µg/m³'
[11:11:58][C][pmsx003:228]:     Accuracy Decimals: 0
[11:11:58][C][pmsx003:228]:     Icon: 'mdi:chemical-weapon'
[11:11:58][D][pmsx003:169]: Got PM1.0 Concentration: 0 µg/m^3, PM2.5 Concentration 0 µg/m^3, PM10.0 Concentration: 0 µg/m^3
[11:11:58][D][sensor:113]: 'Particulate Matter <1.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:11:58][D][sensor:113]: 'Particulate Matter <2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:11:58][D][sensor:113]: 'Particulate Matter <10.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:11:58][C][dht:017]: DHT:
[11:11:58][C][dht:018]:   Pin: GPIO2
[11:11:58][C][dht:020]:   Auto-detected model: DHT11
[11:11:58][C][dht:027]:   Update Interval: 20.0s
[... 67 lines cut ...]
[11:12:06][D][sensor:113]: 'Particulate Matter <2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:12:06][D][sensor:113]: 'Particulate Matter <10.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:12:06][D][uart_debug:109]: <<< 42:4D:00:1C:00:00:00:00:00:00:00:00:00:00:00:00:00:D8:00:A1:00:00:00:00:00:00:00:00:12:00:02:36
[11:12:07][D][pmsx003:169]: Got PM1.0 Concentration: 0 µg/m^3, PM2.5 Concentration 0 µg/m^3, PM10.0 Concentration: 0 µg/m^3
[11:12:07][D][sensor:113]: 'Particulate Matter <1.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:12:07][D][sensor:113]: 'Particulate Matter <2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:12:07][D][sensor:113]: 'Particulate Matter <10.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:12:07][D][uart_debug:109]: <<< 42:4D:00:1C:00:00:00:00:00:00:00:00:00:00:00:00:00:D8:00:A0:00:00:00:00:00:00:00:00:12:00:02:35
[11:12:08][D][pmsx003:169]: Got PM1.0 Concentration: 0 µg/m^3, PM2.5 Concentration 0 µg/m^3, PM10.0 Concentration: 0 µg/m^3
[11:12:08][D][sensor:113]: 'Particulate Matter <1.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:12:08][D][sensor:113]: 'Particulate Matter <2.5µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:12:08][D][sensor:113]: 'Particulate Matter <10.0µm Concentration': Sending state 0.00000 µg/m³ with 0 decimals of accuracy
[11:12:08][D][uart_debug:109]: <<< 42:4D:00:1C:00:00:00:00:00:00:00:00:00:00:00:00:00:D7:00:9E:00:00:00:00:00:00:00:00:12:00:02:32
[11:12:09][D][pmsx003:169]: Got PM1.0 Concentration: 0 µg/m^3, PM2.5 Concentration 1 µg/m^3, PM10.0 Concentration: 1 µg/m^3
[... cut ...]
[11:12:09][D][sensor:113]: 'Airsensor WiFi Signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[11:12:19][D][sensor:113]: 'Airsensor WiFi Signal': Sending state -49.00000 dBm with 0 decimals of accuracy
[11:12:24][D][dht:048]: Got Temperature=22.1°C Humidity=42.0%
[11:12:24][D][sensor:113]: 'Air quality meter Temperature': Sending state 22.10000 °C with 1 decimals of accuracy
[11:12:24][D][sensor:113]: 'Air quality meter Humidity': Sending state 42.00000 % with 0 decimals of accuracy

ssieb, 2021-11-18:
Ok, you got a working config.  I'll take a better look later, but at a first look, it appears to be correct.  Maybe something is wrong with the sensor.

DavyRoswinkel, 2021-11-18:
Maybe sensor is broken??? Used it only for 3 days.....:)

MallocArray, 2021-11-19:
Have you checked what your outdoor air quality is?  If it is already low outside, possibly indoors it really is 0 or near?  I thought mine wasn't working originally, but according to AccuWeather it was only 5-7 um outside so I was properly reading 0 indoors.  I took my sensor outside for a while and saw it bounce around 5-7 so I knew it was reading data.

DavyRoswinkel, 2021-11-19:
> Have you checked what your outdoor air quality is?  If it is already low outside, possibly indoors it really is 0 or near?  I thought mine wasn't working originally, but according to AccuWeather it [...]

You got a point, but I did watch the outside values. Wasn't that low yesterday. Also I never expected 0 as value inside while next to a burning woodstove. ( or am I mistaken the readings from this sensor?)

Nevertheless...I left the sensor connected overnight....this morning it shows values 16, 28, 32.

So it looks like it works now.

closed 2021-12-13 by DavyRoswinkel