esphome/issues #2721: fatal error: ESP8266HTTPClient.h: No such file or directory

jsuanet, 2021-11-17:
### The problem

Compiling with the Nextion display ``` tft_url: ``` option enabled results in "a fatal error occurred"  

fatal error: ESP8266HTTPClient.h: No such file or directory

Compiling without this option is successful.

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

Nextion

### Example YAML snippet

```yaml
api:
  services:
    - service: update_nextion
      then:
        - lambda: 'id(next_disp)->upload_tft();'

display:
  - platform: nextion
    id: next_disp
    update_interval: 5s
    touch_sleep_timeout: 120
    auto_wake_on_touch: true
    brightness: 100%
    tft_url: http://xx.xx.xx.xx:8123/local/tft/design-1.tft
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/nextion.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Deleting /data/nextion-display/.pioenvs
INFO Compiling app...
Processing nextion-display (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESPAsyncTCP-esphome> 1.2.3
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <ESPAsyncTCP-esphome> 1.2.3
|   |-- <Hash> 1.0
|   |-- <ESP8266WiFi> 1.0
|-- <DNSServer> 1.1.1
|-- <ESP8266WiFi> 1.0
|-- <ESP8266mDNS> 1.2
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/api_connection.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/api_server.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/list_entities.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/proto.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/user_services.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/api/util.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/binary_sensor/filter.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/display/display_buffer.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/esp8266/core.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/esp8266/gpio.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/esp8266/preferences.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/homeassistant/binary_sensor/homeassistant_binary_sensor.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/homeassistant/sensor/homeassistant_sensor.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/homeassistant/text_sensor/homeassistant_text_sensor.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/homeassistant/time/homeassistant_time.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/logger/logger.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/md5/md5.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/mdns/mdns_component.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/mdns/mdns_esp32_arduino.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/mdns/mdns_esp8266.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/mdns/mdns_esp_idf.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/network/util.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/nextion/binary_sensor/nextion_binarysensor.cpp.o
Compiling /data/nextion-display/.pioenvs/nextion-display/src/esphome/components/nextion/nextion.cpp.o
In file included from src/esphome/components/nextion/nextion.cpp:1:0:
src/esphome/components/nextion/nextion.h:15:31: fatal error: ESP8266HTTPClient.h: No such file or directory

***************************************************************************
* Looking for ESP8266HTTPClient.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:ESP8266HTTPClient.h"
* Web  > https://platformio.org/lib/search?query=header:ESP8266HTTPClient.h
*
***************************************************************************

 #include <ESP8266HTTPClient.h>
                               ^
compilation terminated.
*** [/data/nextion-display/.pioenvs/nextion-display/src/esphome/components/nextion/nextion.cpp.o] Error 1
========================= [FAILED] Took 31.86 seconds =========================
```

mmakaay, 2021-11-18:
This could be solved by manually adding `http_request:` to the YAML configuration file.
The issue might be because no libraries are included for ESP8266 here in `display.py`;
```python
   if CONF_TFT_URL in config:
        cg.add_define("USE_NEXTION_TFT_UPLOAD")
        cg.add(var.set_tft_url(config[CONF_TFT_URL]))
        if CORE.is_esp32:
            cg.add_library("WiFiClientSecure", None)
            cg.add_library("HTTPClient", None)
```

comet424, 2022-01-12:
so i have tried the  http_request:
but it doesnt work.. it complies now and uploads the yaml  but it wont upload the nextion screen file to the screen

HFsi, 2022-02-24:
I'm facing up the same issue.
Only I can add is that the problem is not present with an ESP32 platform

sanderlv, 2022-11-19:
Here as well as for 8266 it is an issue and esp32 not

closed 2022-11-24 by jesserockz