esphome/issues #2722: Could not find the package platformio/toolchain-riscv-esp for linux_aarch64

nicari92, 2021-11-17:
### The problem

During installation on ESP32-C3 board this problem appears:

[image]

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.9.7

### What platform are you using?

ESP32

### Board

ESP32-C3-DevKitC-02

### Example YAML snippet

```yaml
esphome:
  name: esp32-c3
  platform: ESP32
  board: esp32-c3-devkitm-1
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/esp32-c3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32-c3 (board: esp32-c3-devkitm-1; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-riscv-esp @ ~1.80400.0
Error: Could not find the package with 'platformio/toolchain-riscv-esp @ ~1.80400.0' requirements for your system 'linux_aarch64'
```

nicari92, 2021-11-17:
HomeAssistant running in Raspberry Pi 4 with Debian 10

oxan, 2021-11-18:
There's no RISC-V toolchain for ARM64 available yet on PlatformIO. Not much we can do about that.

Bascht74, 2022-03-13:
@oxan it seems that this changed now:
https://github.com/espressif/esp-idf/issues/6432

> Hi @SebastianBar, aarch64 toolchains (for Xtensa and RISC-V) are now provided and should be installed automatically when you run install.sh. This is available in master branch and in release v4.4.
> We still don't have an aarch64 toolchain for ESP32 ULP coprocessor, so it won't be installed when running install.sh on aarch64. Aside from that, installation on aarch64 using install.sh should succeed. If you run into any issue, please share some details about your OS and the IDF version you are using.

If i try now to compile I get: 
```
INFO Reading configuration /config/esphome/nilan.yaml...
INFO Updating https://github.com/Jopand/esphome_components.git@main
INFO Generating C++ source...
INFO Compiling app...
Processing nilan (board: m5stack-atom; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Platform Manager: Installing platformio/espressif32 @ 3.5.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32 @ 3.5.0 has been installed!
The platform 'platformio/espressif32 @ 3.5.0' has been successfully installed!
The rest of the packages will be installed later depending on your build environment.
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```

I am not shure if esphome needs the esp32ulp package in my case. Could it be "skipped" in some configurations? (e.g. if the ULP is not used based on the yaml file).

Seb

kaystrobach, 2022-10-08:
```
INFO Generating C++ source...
INFO Compiling app...
Processing bhkw (board: esp32dev; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
INFO Installing platformio/toolchain-esp32ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```

Hardware:

* Home Assistant Blue / Hardkernel Odroid-N2 

Software:

* ESPHome by Nabu Casa | [Fund development](https://esphome.io/guides/supporters.html) | [v2022.9.4 Documentation](https://www.esphome.io/)

It works with the arduino framework, but i can not used MQTT with SSL, wenn using the arduino framework. I also think the BLE proxy only works with the esp-idf framework.

There seems to be a related issue on platform.io

https://github.com/platformio/platform-espressif32/issues/753

According to that package the library support is now given:

https://registry.platformio.org/tools/espressif/toolchain-riscv32-esp/compatibility

kaystrobach, 2022-10-08:
Using docker on mac i get the exact same problem:

```
Platform Manager: Installing platformio/espressif32 @ 3.5.0
INFO Installing platformio/espressif32 @ 3.5.0
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Platform Manager: espressif32@3.5.0 has been installed!
INFO espressif32@3.5.0 has been installed!
Tool Manager: Installing platformio/toolchain-esp32ulp @ ~1.22851.0
INFO Installing platformio/toolchain-esp32ulp @ ~1.22851.0
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ ~1.22851.0' requirements for your system 'linux_aarch64'
```

Seems to be that this is really needed: https://github.com/platformio/platform-espressif32/pull/924

Jason2866, 2022-10-08:
@kaystrobach You are right. Without the changes made in PR 924 there is no way to use the linux_aarch64 nor the Apple Arm native toolchains.

kaystrobach, 2022-10-08:
Can you provide a timeframe for that?

I would be glad to sponsor 100€ to get this in an official esphome release for HACS in HomeAssistent. Thank you

Jason2866, 2022-10-08:
I am **not** a maintainer from platformio. Just did some PRs there.

kaystrobach, 2022-10-26:
Just updated to `v2022.10.1` still have the same problem. There are also tons of similar issues, with compile errors and the bluetooth proxy ... it's really a pity, that the reference platform (home assistant blue) is failing so hard when compiling with esp-idf

ssieb, 2022-10-26:
esphome can't move to a newer platform until issues caused by platform changes have been resolved.  You can specify the platform version in the config.  I can't test it because I don't have any AARCH64 devices available to test on right now.

ssieb, 2022-10-26:
You're also using a device that the support is still in development.

kaystrobach, 2022-10-26:
which one do you mean?

 the odroid-n2? It's the original old home assistant blue 🤔

Or do you mean my ESP Hardware?

I just need to find a way to use MQTT with SSL, if that's possible with the arduino framework, i would be happy for any advice. thank you.

ssieb, 2022-10-26:
I mean the ESP C3, unless you're using something different than the original poster.
SSL support is [broken](https://github.com/esphome/issues/issues/2660) on 8266.  I don't  know about ESP32.  What issue are you having and how does it work with IDF?

kaystrobach, 2022-10-26:
I have a normal esp32, can link the board lateron. According to the docs ingot the impression, that ssl should work on the esp32 with esp-idf.

ssieb, 2022-10-26:
If it's not working with Arduino, file an issue with the details or comment in the one I linked.  I only know about the issue for 8266, but maybe that also applies to ESP32.

Jason2866, 2022-10-27:
@ssieb esp8266 and esp32 have nothing common. Of there is a issue with esp8266 it does not affect in any way the esp32x. There is no issue in Arduino ESP32 nor in latest Platformio. It is only esphome which does not support the newer version.

ssieb, 2022-10-27:
@Jason2866, the esphome code is mostly the same between them, so that's really not true.  And you're also missing the conversation here.  This is not about the original topic, which is why I asked him to make a new issue.

kaystrobach, 2022-10-27:
Main reason for going with esp-idf:

* https://esphome.io/components/mqtt.html#mqtt-ssl-fingerprints
* https://esphome.io/components/mqtt.html

And this is also related

* https://github.com/esphome/feature-requests/issues/17

So basically we have all the issues here. Maybe only an example of mqtt with ssl f.e. via hiveMQ is missing.

Jason2866, 2022-10-27:
@ssieb Well the esphome code is (maybe) the same. BUT this just the last layer of code. The underlying layers for esp8266 and esp32 ARE completely different.

spn91, 2022-11-16:
Compiling also fails for me on a raspberry pi 4 (64 bit OS) with these esp config
```yaml
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
```
I thought with the release of esp-idf 4.4.3 this should get fixed, but it still fails.

Jason2866, 2022-11-16:
@spn91 Platformio has to release.

kaystrobach, 2022-11-16:
@Jason2866 can we somehow help / push that on the platformio end?

Jason2866, 2022-11-16:
@kaystrobach ask in [Platformio github](https://github.com/platformio/platform-espressif32/issues) for a release. I am not related to. Just did some PRs.

kaystrobach, 2022-11-17:
Here is a working fork, now we need to wait for the update in esphome 😅. Thank you @Jason2866

spn91, 2022-11-17:
I have tried the fork from @Jason2866  a few week ago which was mentioned in another issues I think. Didn’t get it to work with esphome, got the same error as the other user there (I can’t remember the error now).

bwsoftinc, 2022-11-23:
There is no platformio/toolchain-riscv-esp build for linux-aarch64, its a year old and probably not maintained any longer https://registry.platformio.org/tools/platformio/toolchain-riscv-esp/compatibility

There is a toolchain espressif/toolchain-riscv32-esp for linux-aarch64 that does work
https://registry.platformio.org/tools/espressif/toolchain-riscv32-esp/compatibility

Not an external issue any more?  Maybe esphome can fix the board/variant mappings to use the correct toolchain

`esphome:
  name: esp32c3
  platformio_options:
    board_build.flash_mode: dio  
    platform_packages:
     - toolchain-riscv32-esp # @8.4.0+2021r2-patch5
     - toolchain-xtensa-esp32s2 # @8.4.0+2021r2-patch5

esp32:
  board: esp32-c3-devkitm-1
  variant: esp32c3
  framework:
    type: esp-idf`

Jason2866, 2022-11-23:
@bwsoftinc  You are wrong, The uploaded riscv (Linux aarch64) patch5 from espressif does work. The fact that it can not be used to compile esphome is ONLY a problem from  esphome (HA).

kaystrobach, 2022-11-23:
Now it’s getting interesting, we are getting close to a solution - right 🤩?

bwsoftinc, 2022-11-23:
@Jason2866 "can not"? try those yml options
```
INFO Reading configuration /config/esp32c3.yaml...
INFO Generating C++ source...
INFO Compiling app...
Processing esp32c3 (board: esp32-c3-devkitm-1; framework: espidf; platform: platformio/espressif32 @ 3.5.0)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40302.0 (4.3.2) 
 - tool-cmake @ 3.16.9 
 - tool-ninja @ 1.10.2 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
Reading CMake configuration...

...

INFO Uploading /config/.esphome/build/esp32c3/.pioenvs/esp32c3/firmware.bin (781280 bytes)
Uploading: [============================================================] 100% Done...

INFO Waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from 192.168.100.144 using esphome API
INFO Successfully connected to 192.168.100.144
[00:44:10][I][app:102]: ESPHome version 2022.11.1 compiled on Nov 23 2022, 00:41:05
[00:44:10][C][wifi:504]: WiFi:
[00:44:10][C][wifi:362]:   Local MAC: 60:55:F9:79:82:98
[00:44:10][C][wifi:363]:   SSID: [redacted]
[00:44:10][C][wifi:364]:   IP Address: 192.168.100.144
[00:44:10][C][wifi:366]:   BSSID: [redacted]
[00:44:10][C][wifi:367]:   Hostname: 'esp32c3'
[00:44:10][C][wifi:369]:   Signal strength: -41 dB ▂▄▆█
```

spn91, 2022-11-23:
@bwsoftinc is this something special for the esp32c3? Because I can't get it to work.
```yaml
esphome:
  name: ${device_name}
  platformio_options:
    board_build.flash_mode: dio
    platform_packages:
     - toolchain-riscv32-esp # @8.4.0+2021r2-patch5
     - toolchain-xtensa-esp32s2 # @8.4.0+2021r2-patch5
     - toolchain-esp32ulp # @2.35.0-20220830
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```
`Error: Could not find the package with 'platformio/toolchain-esp32ulp @ *' requirements for your system 'linux_aarch64'`

Jason2866, 2022-11-23:
@bwsoftinc  I do not use EspHome nor HomeAssistant.

spn91, 2022-11-23:
@bwsoftinc I have uncomment the versions, but it is failing again, because of esp32ulp

```Tool Manager: Installing platformio/toolchain-esp32ulp @ 2.35.0-20220830
INFO Installing platformio/toolchain-esp32ulp @ 2.35.0-20220830
Error: Could not find the package with 'platformio/toolchain-esp32ulp @ 2.35.0-20220830' requirements for your system 'linux_aarch64'```

Jason2866, 2022-11-23:
This is still the old IDF 4.4.2 release. 
`- framework-espidf @ 3.40302.0 (4.3.2)`
Actual is IDF 4.4.3. You may run in issues since IDF 4.4.2 is not official ready for the patch5 toolchains. It will probably work. ;-). Patch5 build is a bug fix release, so no breaking changes.

Jason2866, 2022-11-23:
@spn91 
Not
```
platformio/toolchain-esp32ulp
```
the new toolchains are all from espressif
```
espressif/toolchain-esp32ulp
```

spn91, 2022-11-23:
@Jason2866 I`m just using the config provided by bwsoftinc -> https://github.com/esphome/issues/issues/2722#issuecomment-1324783147

Maybe I have to reset something.

Jason2866, 2022-11-23:
@spn91 The owner of the packages should be provided, if it is not platformio.
Platformio tries to search (if not provided) by default for owner platformio. If there are **any** packages found, it will not search for others.

Jason2866, 2022-11-23:
So this should work always
```
esphome:
  name: ${device_name}
  platformio_options:
    board_build.flash_mode: dio
    platform_packages:
     - espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
     - espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
     - espressif/toolchain-esp32ulp @ ~2
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```

spn91, 2022-11-23:
Got it to compile with this settings

```yaml
esphome:
  name: ${device_name}
  platformio_options:
    board_build.flash_mode: dio
    platform_packages:
     - espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5
     - espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
     - espressif/toolchain-xtensa-esp32
     - espressif/toolchain-esp32ulp @ ~2
```
But know it is failing while compiling.
<details>
  <summary>Error</summary>

  ```
In file included from /data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c:16:
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c: In function 'xtensa_perfmon_init':
/data/cache/platformio/packages/framework-espidf/components/xtensa/include/xtensa-debug-module.h:80:29: error: 'XCHAL_NUM_PERF_COUNTERS' undeclared (first use in this function)
 #define ERI_PERFMON_MAX     XCHAL_NUM_PERF_COUNTERS
                             ^~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c:22:15: note: in expansion of macro 'ERI_PERFMON_MAX'
     if (id >= ERI_PERFMON_MAX) {
               ^~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/xtensa/include/xtensa-debug-module.h:80:29: note: each undeclared identifier is reported only once for each function it appears in
 #define ERI_PERFMON_MAX     XCHAL_NUM_PERF_COUNTERS
                             ^~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c:22:15: note: in expansion of macro 'ERI_PERFMON_MAX'
     if (id >= ERI_PERFMON_MAX) {
               ^~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c: In function 'xtensa_perfmon_dump':
/data/cache/platformio/packages/framework-espidf/components/xtensa/include/xtensa-debug-module.h:80:29: error: 'XCHAL_NUM_PERF_COUNTERS' undeclared (first use in this function)
 #define ERI_PERFMON_MAX     XCHAL_NUM_PERF_COUNTERS
                             ^~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c:42:26: note: in expansion of macro 'ERI_PERFMON_MAX'
     for (int i = 0 ; i < ERI_PERFMON_MAX ; i++) {
                          ^~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c: In function 'xtensa_perfmon_reset':
/data/cache/platformio/packages/framework-espidf/components/xtensa/include/xtensa-debug-module.h:80:29: error: 'XCHAL_NUM_PERF_COUNTERS' undeclared (first use in this function)
 #define ERI_PERFMON_MAX     XCHAL_NUM_PERF_COUNTERS
                             ^~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c:52:15: note: in expansion of macro 'ERI_PERFMON_MAX'
     if (id >= ERI_PERFMON_MAX) {
               ^~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c: In function 'xtensa_perfmon_value':
/data/cache/platformio/packages/framework-espidf/components/xtensa/include/xtensa-debug-module.h:80:29: error: 'XCHAL_NUM_PERF_COUNTERS' undeclared (first use in this function)
 #define ERI_PERFMON_MAX     XCHAL_NUM_PERF_COUNTERS
                             ^~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c:71:15: note: in expansion of macro 'ERI_PERFMON_MAX'
     if (id >= ERI_PERFMON_MAX) {
               ^~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c: In function 'xtensa_perfmon_overflow':
/data/cache/platformio/packages/framework-espidf/components/xtensa/include/xtensa-debug-module.h:80:29: error: 'XCHAL_NUM_PERF_COUNTERS' undeclared (first use in this function)
 #define ERI_PERFMON_MAX     XCHAL_NUM_PERF_COUNTERS
                             ^~~~~~~~~~~~~~~~~~~~~~~
/data/cache/platformio/packages/framework-espidf/components/perfmon/xtensa_perfmon_access.c:80:15: note: in expansion of macro 'ERI_PERFMON_MAX'
     if (id >= ERI_PERFMON_MAX) {
               ^~~~~~~~~~~~~~~
*** [/data/deckenleuchte-arbeitsplatz/.pioenvs/deckenleuchte-arbeitsplatz/perfmon/xtensa_perfmon_access.o] Error 1
========================= [FAILED] Took 357.28 seconds =========================
  ```
</details>

jfparis, 2022-11-23:
@Jason2866 this is really helpful for me.

This begs the question though. How do you identify those packages and their latest version?

Thanks

Jason2866, 2022-11-23:
@jfparis You can search for: https://registry.platformio.org/
I knew the release because i asked the maintainer of the toolchains from espressif to release.

Jason2866, 2022-11-23:
@spn91 Delete everything in folder `data/cache/platformio`. If it is still not working, i cant help since the setup is esphome/HA customized.

spn91, 2022-11-23:
@Jason2866 I'm jusing an ESP32, not C3.

Jason2866, 2022-11-23:
Install only needed toolchain(s)
```
esphome:
  name: esp32
  platformio_options:
    board_build.flash_mode: dio
     - espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
     - espressif/toolchain-esp32ulp @ ~2
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
```

spn91, 2022-11-23:
But if I don't add  `- espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5` it is failing during dependency installation.

Jason2866, 2022-11-23:
Ah, okay. Dumb check, since the S2 toolchain is not needed to compile for an ESP32.

spn91, 2022-11-23:
And it seems that the esp32s package wants  `- espressif/toolchain-riscv32-esp @ 8.4.0+2021r2-patch5`
But after all, it fails with the same error as mentioned here https://github.com/esphome/issues/issues/2722#issuecomment-1324934288

jfparis, 2022-11-23:
I have mine running with the below header

Please note
* Arduino framework 
* patch level 3 which seems to be what esphome is pulling when running on my AMD64 computer

```
esphome:
  name: blablabla
  platformio_options: 
    platform: espressif32
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
    platform_packages:
     - espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
     - espressif/toolchain-esp32ulp @ ~2

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: arduino
    version: 2.0.0
```

Jason2866, 2022-11-23:
@jfparis  This will not work for aarch64 since why this issue was opened.

jfparis, 2022-11-23:
This definitely work on aarch64. Just run it on my home assistant 64bit raspi 

I think we are discussing different esp though 
My config should work for @spn91

Jason2866, 2022-11-23:
Yup, orig issue is for riscv-esp on aarch64 (C3). Your setup is for ESP32.
Reading this https://esphome.io/components/esp32.html 
i would try to use latest release IDF v4.4. version. NOT the default IDF 4.3.

spn91, 2022-11-23:
@jfparis this is not working, because of the missing dependencies which were already discussed above. I'm not using arduino as framework type at the moment (don't know if I can just change it?). 

@Jason2866 changed it to 4.4.3, which gives me `INFO Installing platformio/framework-espidf @ ~3.40403.0
Error: Could not find the package with 'platformio/framework-espidf @ ~3.40403.0' requirements for your system 'linux_aarch64'
`
Changing it to latest (5.0-rc1) fails during python dependencies installation. 

This is just a big dependency hell :grin:

jfparis, 2022-11-23:
@spn91 why are you using esp-idf ?

FWIW, the following works on an ESP32 (still patch level 3 for the latest espidf framework 4.4.2) 

```
esphome:
  name: xxxxx
  platformio_options: 
    platform: espressif32
    board_build.mcu: esp32
    board_build.f_cpu: 240000000L
    platform_packages:
     - framework-espidf @ 3.40402.0
     - espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
     - espressif/toolchain-esp32ulp @ ~2

esp32:
  board: az-delivery-devkit-v4
  framework:
    type: esp-idf
    version: recommended
    # Custom sdkconfig options
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
```

spn91, 2022-11-23:
@jfparis I can't really give a answer to you. This is a Yeelight which I converted from Yeelight Firmware to esphome. I got the configuration from [here](https://github.com/syssi/esphome-yeelight-ceiling-light).

Ok, it seems, that this now successfully compiled, but now my light stays offline :thinking:

bwsoftinc, 2022-11-23:
@spn92 try another flash mode? I tried a couple before my specific c3 board took the flash write correctly.  The bootloader logs may give more info what the esp32 is doing, like unable to make sense of what was programmed to flash
https://docs.platformio.org/en/latest/platforms/espressif32.html#flash-mode
```board_build.flash_mode: dio```

spn91, 2022-11-23:
@bwsoftinc Yes, I have removed the flash mode before my last try. This was a mistake. 
The problem is, that this is a ceiling light and I have now first to gain access to the esp32 in it and then try to reflash it via serial connection.

Jason2866, 2022-11-23:
You have a softbrick, since yeelight is a single core esp32.
Afaik esphome has no Arduino support for single core esp32.
So you have to use IDF.

spn91, 2022-11-23:
I don’t use arduino.

Jason2866, 2022-11-23:
Have you added
```
type: esp-idf
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```
without it will build for dual core and the efuse crc mac fix is not there.

spn91, 2022-11-23:
```yaml
esphome:
  name: ${device_name}
  platformio_options:
    platform: espressif32
    platform_packages:
     - framework-espidf @ 3.40402.0
     - espressif/toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
     - espressif/toolchain-esp32ulp @ ~2
 #   platform:
 #     - https://github.com/Jason2866/platform-espressif32/releases/download/v2.0.5-esp/platform-espressif32-2.0.5-esp.zip

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
    version: recommended
    sdkconfig_options:
      CONFIG_FREERTOS_UNICORE: y
    advanced:
      ignore_efuse_mac_crc: true
```

This is the config which was applied  to the esp

Jason2866, 2022-12-01:
Did a quick hack to use latest Arduino 2.0.5 / IDF 4.4.3 and toolchains patch5.
So it should work on aarch64 devices.
The fork is [here](https://github.com/Jason2866/esphome)
Git clone, change into esphome folder and install with `pip3 install -e . `

[10 more replies]

closed 2022-12-14 by ssieb