esphome/issues #2725: Wrong timezone offset when sync time with home assistant

igogold, 2021-11-18:
### The problem

When ESP device boots it shows in log such info:
```
...
[09:51:59][C][homeassistant.time:010]: Home Assistant Time:
[09:51:59][C][homeassistant.time:011]:   Timezone: '<+05>-5'
...
[09:52:12][D][api.connection:747]: Home Assistant 2021.11.4 (192.168.5.75): Connected successfully
[09:52:12][D][time:039]: Synchronized time: 2021-11-17 23:52:12
```
In real my timezone has offset +5 and time of boot was 2021-11-18 09:52:12.
I don't know where the error is - in esphome code or home assistant api answer but I have timezone "Asia/Yekaterinburg" setting in my HA server settings.

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP8266

### Board

sonoff th16

### Component causing the issue

time

### Example YAML snippet

```yaml
time:
  - platform: homeassistant
    id: homeassistant_time
```

### Anything in the logs that might be useful for us?

```txt
[09:51:59][C][homeassistant.time:010]: Home Assistant Time:
[09:51:59][C][homeassistant.time:011]:   Timezone: '<+05>-5'
...
[09:52:12][D][api.connection:747]: Home Assistant 2021.11.4 (192.168.5.75): Connected successfully
[09:52:12][D][time:039]: Synchronized time: 2021-11-17 23:52:12
```

### Additional information

Timezone configured in the Home Assistant is "Asia/Yekaterinburg", +5 hours from UTC.

oxan, 2021-11-18:
It seems newlib [doesn't support](https://sourceware.org/git/?p=newlib-cygwin.git;a=blob;f=newlib/libc/time/tzset_r.c;h=9e0cf834bd6beda8353144f49bb432c6c9855feb;hb=c8b779aff447966ec467c3a52edab34a6ab76277#l47) enclosing the timezone name in angle brackets to extend the character set, and misparses the timezone string for Asia/Yekaterinburg.

I recommend setting the timezone manually in your config (`timezone: YEKT+5`), this unfortunately isn't likely to get fixed anytime soon.

igogold, 2021-11-19:
@oxan, thank you, I'll add this option in config as workaround.

mmakaay, 2022-05-11:
@igogold This fixed the issue for you? In that  case, we can close this issue.

mmakaay, 2022-05-11:
Perfect! Thanks for the feedback.

closed 2022-05-11 by mmakaay