esphome/issues #2726: Esphome devices are not connecting to Home Assistant

hvddrift, 2021-11-18:
### The problem

 I have for the last few days been running into all types of problems. One-minute ESPHOME devices will work. On a restart of HA all of the devices are unavailable

On the integrations page the device will say "Not loaded"

The device shows as Online in the esphome integrations page.
[image]

The Web UI works
[image]

But in the integrations, it shows it as "Not Loaded"

[image]

[image]

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP8266

### Board

All Devices ESP32, Shelly, Sonoff, Zeinsmart etc

### Component causing the issue

All

### Example YAML snippet

```yaml
API are in all my files.
```

### Anything in the logs that might be useful for us?

```txt
No errors in my logs other than this. This is from the web server

[E][api:123]: No client connected to API. Rebooting...
[I][ota:102]: Boot seems successful, resetting boot loop counter.
```

oxan, 2021-11-18:
Take a look inside your HA logs, there's probably something going wrong on that side that delays loading of the ESPHome integration.

hvddrift, 2021-11-18:
Could this be the error?

`2021-11-19 08:01:53 ERROR (MainThread) [homeassistant] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 197, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 215, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 709, in async_create_flow
    integration.get_platform("config_flow")
  File "/usr/src/homeassistant/homeassistant/loader.py", line 530, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 535, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "", line 1030, in _gcd_import
  File "", line 1007, in _find_and_load
  File "", line 972, in _find_and_load_unlocked
  File "", line 228, in _call_with_frames_removed
  File "", line 1030, in _gcd_import
  File "", line 1007, in _find_and_load
  File "", line 986, in _find_and_load_unlocked
  File "", line 680, in _load_unlocked
  File "", line 850, in exec_module
  File "", line 228, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 10, in 
    from aioesphomeapi import (
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/__init__.py", line 2, in 
    from .client import APIClient
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/client.py", line 17, in 
    from .api_pb2 import (  # type: ignore
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/api_pb2.py", line 16, in 
    from . import api_options_pb2 as api__options__pb2
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/api_options_pb2.py", line 24, in 
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 197, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 215, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 709, in async_create_flow
    integration.get_platform("config_flow")
  File "/usr/src/homeassistant/homeassistant/loader.py", line 530, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 535, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "", line 1030, in _gcd_import
  File "", line 1007, in _find_and_load
  File "", line 972, in _find_and_load_unlocked
  File "", line 228, in _call_with_frames_removed
  File "", line 1030, in _gcd_import
  File "", line 1007, in _find_and_load
  File "", line 986, in _find_and_load_unlocked
  File "", line 680, in _load_unlocked
  File "", line 850, in exec_module
  File "", line 228, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 10, in 
    from aioesphomeapi import (
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/__init__.py", line 2, in 
    from .client import APIClient
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/client.py", line 17, in 
    from .api_pb2 import (  # type: ignore
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/api_pb2.py", line 16, in 
    from . import api_options_pb2 as api__options__pb2
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/api_options_pb2.py", line 24, in 
    create_key=_descriptor._internal_create_key,
[... 554 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/loader.py", line 535, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/usr/local/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "", line 1030, in _gcd_import
  File "", line 1007, in _find_and_load
  File "", line 972, in _find_and_load_unlocked
  File "", line 228, in _call_with_frames_removed
  File "", line 1030, in _gcd_import
  File "", line 1007, in _find_and_load
  File "", line 986, in _find_and_load_unlocked
  File "", line 680, in _load_unlocked
  File "", line 850, in exec_module
  File "", line 228, in _call_with_frames_removed
  File "/usr/src/homeassistant/homeassistant/components/esphome/__init__.py", line 10, in 
    from aioesphomeapi import (
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/__init__.py", line 2, in 
    from .client import APIClient
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/client.py", line 17, in 
    from .api_pb2 import (  # type: ignore
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/api_pb2.py", line 16, in 
    from . import api_options_pb2 as api__options__pb2
  File "/usr/local/lib/python3.9/site-packages/aioesphomeapi/api_options_pb2.py", line 24, in 
    create_key=_descriptor._internal_create_key,
AttributeError: module 'google.protobuf.descriptor' has no attribute '_internal_create_key'`

oxan, 2021-11-18:
> Could this be the error?

Yes, though I don't immediately see what causes this...

hvddrift, 2021-11-18:
I have opened an issue in Home Assistant. I'll see if they have a solution.

mmakaay, 2021-11-18:
How do you run Home Assistant? Within a venv by any chance, or are you running some container version?
The error might indicate that the protobuf Python library requires an upgrade. 
I have protobuf 3.17.3 and with that version, I see:
```
>>> from google.protobuf import descriptor
>>> descriptor._internal_create_key
<object object at 0x7fa5f6e57640>
```
So the attribute is available for me in that version of google.protobuf.

hvddrift, 2021-11-18:
It’s home assistant os. Can I still upgrade?

## System Health

version | core-2021.11.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.75
arch | x86_64
timezone | Australia/Brisbane

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4971
Installed Version | 1.17.2
Stage | running
Available Repositories | 976
Installed Repositories | 61

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 6.6
-- | --
update_channel | stable
supervisor_version | supervisor-2021.10.8
docker_version | 20.10.8
disk_total | 228.5 GB
disk_used | 22.9 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Node-RED (10.1.1), Samba share (9.5.1), SSH & Web Terminal (9.1.0), Vaultwarden (Bitwarden) (0.14.0), WireGuard (0.6.0), Log Viewer (0.12.1), Studio Code Server (3.7.0), Home Assistant Google Drive Backup (0.105.2), Duck DNS (1.14.0), ADB - Android Debug Bridge (0.8.0), NGINX Home Assistant SSL proxy (3.0.2), Check Home Assistant configuration (3.9.0), Assistant Relay (0.7.4), Frigate NVR (2.3), Mosquitto broker (6.0.1), TasmoAdmin (0.16.0), ESPHome (2021.11.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 34
views | 40
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

closed 2022-03-19 by hvddrift