esphome/issues #2728: pm1006 uses software UART when following the docs

mmakaay, 2021-11-18:
### The problem

I see that the pm1006 component documentation references D2 as the pin to use on D1 mini, but that results in software UART right away. Pin D7 with logger `baud_rate: 0` would make it use hardware UART instead (just like it's done for DSMR).

For the required baud rate of 9600, this doesn't cause read issues, but when flashing the device the software UART activity does interfere with the OTA process, causing OTA to fail part of the time. While this might be considered an OTA bug, maybe it's worth making the described configuration for pm1006 make use of hardware UART support?

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

pm1006

### Anything in the logs that might be useful for us?

```txt
[13:10:17][C][uart.arduino_esp8266:095]: UART Bus:
[13:10:17][C][uart.arduino_esp8266:097]:   RX Pin: GPIO4
[13:10:17][C][uart.arduino_esp8266:099]:   RX Buffer Size: 256
[13:10:17][C][uart.arduino_esp8266:101]:   Baud Rate: 9600 baud
[13:10:17][C][uart.arduino_esp8266:102]:   Data Bits: 8
[13:10:17][C][uart.arduino_esp8266:103]:   Parity: NONE
[13:10:17][C][uart.arduino_esp8266:104]:   Stop bits: 1
[13:10:17][C][uart.arduino_esp8266:108]:   Using software serial <---
```

Habbie, 2021-11-18:
Indeed, the software UART quickly breaks down. I noticed this when I tried using one towards the PM1006 and another one towards the original IKEA PCB. If my instructions lead to software UART use, we should definitely fix that.

mmakaay, 2021-11-18:
I'll open up one of my devices, probably tomorrow, to use D7 instad of D2. That ought to provide me with hardware UART. I'll post an update when I tested that.

Habbie, 2021-11-18:
I just now reread https://esphome.io/components/uart.html (it's been a while) and I think I made this choice on purpose. However, if you have a config that makes the logger -and- pm1006 work, both on hardware, that would be perfect.

Habbie, 2021-11-18:
(I think I did not realise that logger is TX-only and thus we could claim that RX-pin.)

Habbie, 2021-11-18:
Having now also read https://esphome.io/components/logger.html, I'm not sure the RX/TX split I had in mind is possible. In that case, indeed we should document the two choices (hw uart logging + sw uart pm1006, or no logging + hw uart pm1006)

mmakaay, 2021-11-18:
Yes, those are the two routes that are possible I think.
Pin D7 + baud rate 0 for the logger bring the goodness of hw uart.

Habbie, 2021-11-18:
It would be nice though if logger could get hardware TX (on the pin connected to USB serial) while pm1006 could get hardware RX, but my impression is that that would need some core esphome changes.

MallocArray, 2021-11-19:
I think you are running into the same bug that I have with a PMS5003 and UART causing OTA failures.  Using an older version of ESPHome 1.18.0 does not run into this, but the PMS1006 isn't supported on that version:
https://github.com/esphome/issues/issues/2629
#2629