esphome/issues #2730: total_daily_energy has strange spikes

irekzielinski, 2021-11-18:
### The problem

I have an optical pulse conter that counts flashes in my electricity meter.
- platform: pulse_meter (with total), works as expected and value increases over time as pulses are counted.
- platform: total_daily_energy - set up to track daily usage (value of this sensor sometimes will jump up, and count more than real usage)

In other words: total_daily_energy seems to count too much, more than the pulse_meter itself

### Which version of ESPHome has the issue?

2021.10.2

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.11.2

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

total_daily_energy

### Example YAML snippet

```yaml
sensor:
  - platform: pulse_meter
    pin: D2
    unit_of_measurement: 'kW'
    device_class: 'power'
    name: 'Spa Power Consumption'
    id: spa_power_consumption
    internal_filter: 100ms
    timeout: 300s
    accuracy_decimals: 2
    filters:
      - multiply: 0.06
      - delta: 0.008
    total:
      name: "Spa Consumed Energy"
      unit_of_measurement: "kWh"
      accuracy_decimals: 2
      device_class: 'energy'
      state_class: total_increasing
      filters:
        - multiply: 0.001
        
  - platform: total_daily_energy
    name: "Spa Consumed Energy Today"
    power_id: spa_power_consumption
    accuracy_decimals: 2
    unit_of_measurement: "kWh"
    device_class: 'energy'
    id: spa_consumed_energy
    min_save_interval: 30s
```

irekzielinski, 2021-11-18:
pulse_meter: device was rebooted yesterday. But looking at today usage we can see a slow rise in the number of counted pulses. Day started at 0.6kwh and we used today around: 3.32kwh
[image]

I would expect total_daily_energy to show value of around 3.32kwh (today usage), but there are those strnge spikes in the graph at 5:13am and 15:10 - total count for today is 5.68kwh. This is more than the pulse_meter.
[image]

mmakaay, 2021-11-18:
What pops to mind when seeing that, would be bouncing behaviour when the sensor is halted right at the point between a high and low signal. You count light flashes, so I can't easily relate that to bouncing. 
How long is the led active during such flash?

irekzielinski, 2021-11-18:
> What pops to mind when seeing that, would be bouncing behaviour when the sensor is halted right at the point between a high and low signal. You count light flashes, so I can't easily relate that to bouncing. How long is the led active during such flash?

pulse_counter has internal_filter: 100ms and this seem to count pulses corectly (no creeping up or unstable operation).
is total_daily_energy taking into account internal_filter of pulse_counter? I though total_daily_energy should be using outut of pulse_counter right?

irekzielinski, 2021-11-18:
Here is an output of the pulse_meter:
[image]

NOTE: at 5:15 and 15:10 we had a very short lived spike in energy consumption. Here is a zoomed in one at 5 am:
[image]

at 5am we see:
(1.4kw for 1 minute = 23wh)  +  (0.61kw for 2 minutes = 20wh) = total: 43wh (or 43 pulses seen by pulse_counter

43wh matches what we see in pulse_meter total sensor (goes up from 1.08kwh to 1.11kwh - so around 30wh, close to 43 we calculated, we might see some rounding errors):
[image]

but for the same period when we look into total_daily_energy sensor we see that it goes up by some crazy high value: from 0.25kwh to 1.19kwh. So registers increase of 0.94kwh (940wh, where in fact only around 43 wh were registered by pulse_meter)
[image]

My theory: total_daily_energy gets confused by a high, short lived spike in energy consumption. This sensor seems to think energy was cunsumed at high rate for much longer than it really was.

mmakaay, 2021-11-18:
That could be caused by `timeout: 300s`.
The timeout is used to let the device decide after how much time to consider no change in the pulse as the end of the consumption period. That timeout should be somewhere near the maximum time that you see between pulses in normal operation. I assume thats not 1 pulse per 5 minutes?
Right now, you might get a few pulses, triggering the high usage, but possibly the code is lingering after that power peak for 5 minutes before the usage is considered at its end.

irekzielinski, 2021-11-19:
I think a combination of "delta" and maybe "timeout" starved the daily sensor.
This should not happen in my oppinion as "total" sensor inside "pulse_meter" was not affected.

**Yesterday I added a "heartbeat: 30s" and this seem to resolve problem - so clearly feeding more data to the sensor solves the problem.**

So maybe something can be improved here, if not then at least I have a work-around.

closed 2022-04-03 by github-actions[bot]