esphome/issues #2734: RGB/RGBW showing as just a toggle light

pete1450, 2021-11-19:
### The problem

I've been trying to put together a RGBW strip and it works in that I can toggle it from both HomeAssistant and the web interface but that's it. No color control. The platform is set to RGBW so I would think even if I messed everything else up and the lights didn't work shouldn't it at least think it's an RGBW device and show the relevant controls?

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

rgbw

### Example YAML snippet

```yaml
esphome:
  name: rgbwstrip
  platform: ESP8266
  board: nodemcuv2

# Enable logging
logger:

# Enable Home Assistant API
api:
  password: ""

ota:
  password: ""

wifi:
  ssid: "something"
  password: "somethingelse"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Rgbwstrip Fallback Hotspot"
    password: "mmxFm7fjii0b"

captive_portal:

light:
  - platform: rgbw
    name: "Kitchen Floor Lights"
    red: output_component1
    green: output_component2
    blue: output_component3
    white: output_component4
    
output:
  - platform: esp8266_pwm
    id: output_component1
    pin: D1
  - platform: esp8266_pwm
    id: output_component2
    pin: D2
  - platform: esp8266_pwm
    id: output_component3
    pin: D3
  - platform: esp8266_pwm
    id: output_component4
    pin: D4
    
web_server:
  port: 80
```

oxan, 2021-12-01:
Which HA version are you using?

pete1450, 2021-12-06:
2021.6.0 which is admittedly a bit old but as I mentioned this problem is shown when just using the sensors web interface so separate from HA.

oxan, 2021-12-20:
Try a newer HA version, it works fine for me with 2021.12. The web interface is limited to just toggling lights at the moment.

closed 2022-05-12 by github-actions[bot]