esphome/issues #2736: Not connect to WPA2-PSK/WPA3-SAE mixed mode encryption SSID

jlazkano, 2021-11-19:
### The problem

I am new in ESPHome.

I compile my first image for Xiaomi Bed Lamp 2, and I notice that the lamp doesn't connect to my SSID.

My actual SSID has WPA2-PSK/WPA3-SAE mixed mode encryption, if I change to WPA2-PSK it works.

I think that the device is trying to connect first to WPA3 and can not connect and doesn't try the second WPA2 mode.

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP32

### Board

Xiaomi Bed Lamp 2

### Component causing the issue

wifi

### Example YAML snippet

```yaml
packages:
  bslamp2:
    url: https://github.com/mmakaay/esphome-xiaomi_bslamp2
    ref: dev
    files:
      - packages/core.yaml
      - packages/behavior_default.yaml
      - packages/ota_feedback.yaml
      - packages/activate_preset_svc.yaml
    refresh: 0s

# Enable logging
logger:

# Enable Home Assistant API
api:

ota:
  password: "xxx"

wifi:
  ssid: "xxx"
  password: "xxx"

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "XLamp Fallback Hotspot"
    password: "xxx"
```

MallocArray, 2021-12-19:
I just tried WPA3 only and was not able to get a D1 Mini to connect.  Is WPA3 supported by ESPHome?

JeffResc, 2022-04-26:
It's been awhile, but I just tested this and for "WPA2-PSK/WPA3-SAE Mixed Mode," this is working for me on a variety of ESP8266 and ESP32 devices running ESPHome.

SerialVelocity, 2022-08-23:
I seem to be able to get it working with wpa2/wpa3 mixed mode when using the standard framework, but not with the esp-idf framework. (version 2022.8.0). It says:
```
[W][wifi:442]: No matching network found!
```