esphome/issues #2738: ADC attenuation issue (ESP32)

derwolff, 2021-11-20:
### The problem

After update to ESPHome 2021.11.1 and firmware update of 2 ESP32, the measured voltage is different from previous one. In my case 2 CT-Clamp sensors and a moisture sensor with calibration data are not useable anymore.
In one case for example the measured voltage before update was 2.35V and after update 2. 17V.
All ADC attenuation are set to 11db.
I found some issues reported from the mid of  this year, due to IDE changes. I think it`s not complete solved.
Switch back to older code, solves the issue for me:

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: 2021.10.3
    components: [ adc ]

### Which version of ESPHome has the issue?

2021.11.1 

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP32

### Board

mhetesp32minikit

### Component causing the issue

adc

### Example YAML snippet

```yaml
sensor:
  - platform: adc
    pin: GPIO32
    attenuation: 11db
    #attenuation: auto
    #raw: true
    name: "${devicename} Feuchtigkeit"
    id: ADC32
    unit_of_measurement: "%"
    icon: "mdi:water-percent"
    accuracy_decimals: 1
    update_interval: 360s
    filters:
      - median:
          window_size: 5
          send_every: 5
          send_first_at: 5
      - lambda: !lambda |-
          id(val_raw) = x;
          float val = (x - id(val_max)) * 100 / (id(val_min) - id(val_max)) + id(val_max);
          if (val < 0) {
            val = 0.00 ;
          }
          else if (val > 100) {
              val = 100.00 ;
          }
          return val;

--------------------------------------------------------------------------

sensor:
  - platform: adc
    pin: GPIO32
    attenuation: 11db
    #attenuation: auto
    id: ADC32
  - platform: adc
    pin: GPIO33
    attenuation: 11db
    #attenuation: auto
    id: ADC33  
    
  - platform: ct_clamp
    sensor: ADC32
    name: "Leistung TR"
    id: power_tr
    update_interval: 10s
    sample_duration: 1000ms
    filters:
      - calibrate_linear:
          # Measured value of 0 maps to 0A
          #- 0 -> 0
          - 0.00522 -> 0
          # Known load: 4.0A
          # Value shown in logs: 0.1333A
          - 0.03904 -> 0.451
      - median:
          window_size: 7
          send_every: 6
      #- lambda: return x * 230.0;
      - lambda: if ((x * 230) < 1) return 0; else return (x * 230.0);
    unit_of_measurement: "W"
    accuracy_decimals: 0
    
  - platform: ct_clamp
    sensor: ADC33
    name: "Leistung WM"
    id: power_wm
    update_interval: 10s
    sample_duration: 1000ms
    filters:
      - calibrate_linear:
          # Measured value of 0 maps to 0A
          #- 0 -> 0
    #      - 0.00459 -> 0
          - 0.00253 -> 0
          # Known load: 4.0A
          # Value shown in logs: 0.1333A
    #      - 0.03918 -> 0.451
          - 0.02663 -> 0.451
      - median:
          window_size: 7
          send_every: 6    
      - lambda: if ((x * 230) < 1) return 0; else return (x * 230.0);
    unit_of_measurement: "W"
    accuracy_decimals: 0
```

oxan, 2021-11-20:
Can you try `raw: true`?

derwolff, 2021-11-20:
Yes I try raw: true but then I get no voltage format value but a integer value like 2500.

oxan, 2021-11-20:
Ah, see [this](https://github.com/esphome/esphome/pull/2663#issue-1044780732), you need to add a multiply filter.

@CarlosGS @jesserockz I think we need to update the docs, both to note the breaking change in ADC behaviour in the changelog (esphome/esphome#2574 wasn't labelled, unfortunately), and to note that with the `raw` option you get a value 0-4095 instead of a voltage, possibly with an example to convert it back to a voltage.

CarlosGS, 2021-11-20:
Good point, working on that documentation PR now. Not sure of how to change the release log, maybe it would be nice to mention the breaking change in the next release, linking to the docs.

derwolff, 2021-11-20:
Thanks for the information. I have set "raw: true" and add the correct filter value in yaml for 11db.
From my side the issue can be closed.

closed 2021-11-20 by jesserockz