esphome/issues #2739: UART DEBUG AFTER optional/mandatory issue

Tonguc-Endem, 2021-11-20:
### The problem

I have an ESP-01 module talking UART to a Sony Receiver.
I'm using reversed I/O pins:
`  tx_pin: GPIO3 #reversed to avoid boot debug message problems`
`  rx_pin: GPIO1 #reversed to avoid boot debug message problems`
I'm using a Custom UART Text Sensor.
Also, I have:
`logger:`
`  level: VERY_VERBOSE`
`  baud_rate: 0 #disable logging over uart`

Before the update, I was able to sniff all the UART communications with logger level set to VERY_VERBOSE.
Since I don't want any debug messages going to the UART lines, debug is disabled over UART with baud_rate: 0
I only need to see the UART communication over the wireless log output.

After the update, I tried the following exactly as shown in the documentation but I'm getting compilation errors.
`uart:`
`  id: uart_bus`
`  #tx_pin: GPIO1 #standard`
`  #rx_pin: GPIO3 #standard`
`  tx_pin: GPIO3 #reversed to avoid boot debug message problems`
`  rx_pin: GPIO1 #reversed to avoid boot debug message problems`
`  baud_rate: 115200`
`  rx_buffer_size: 80`
`  stop_bits: 1`
`  data_bits: 8`
`  parity: NONE`
`  debug:`
`    direction: both`
`    sequence:`
`      - lambda: UARTDebug::log_hex(direction, bytes, ':');`

If I don't add the last 4 lines (starting with debug:), it compiles ok but then I can not view the UART bytes.
So looking at the compilation error output (included below) I thought the problem might be something to do with the 'after' keyword. However the documentation lists 'after' as optional, so that's why I didn't include it.

The problems went away after adding the 'after' keyword as the following:
`  data_bits: 8`
`  parity: NONE`
`  debug:`
`    direction: both`
`    after:`
`      timeout: 100ms`
`    sequence:`
`      - lambda: UARTDebug::log_hex(direction, bytes, ':');`

I'm not sure if the documentation is wrong or if the 'after' keyword is not coded to be optional.
I hope this helps.

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.4,  supervisor-2021.10.8

### What platform are you using?

ESP8266

### Board

ESP-01

### Component causing the issue

UART, DEBUG

### Anything in the logs that might be useful for us?

```txt
Traceback (most recent call last):
  File "/usr/local/bin/esphome", line 33, in <module>
    sys.exit(load_entry_point('esphome', 'console_scripts', 'esphome')())
  File "/esphome/esphome/__main__.py", line 817, in main
    return run_esphome(sys.argv)
  File "/esphome/esphome/__main__.py", line 804, in run_esphome
    rc = POST_CONFIG_ACTIONS[args.command](args, config)
  File "/esphome/esphome/__main__.py", line 364, in command_run
    exit_code = write_cpp(config)
  File "/esphome/esphome/__main__.py", line 156, in write_cpp
    generate_cpp_contents(config)
  File "/esphome/esphome/__main__.py", line 168, in generate_cpp_contents
    CORE.flush_tasks()
  File "/esphome/esphome/core/__init__.py", line 613, in flush_tasks
    self.event_loop.flush_tasks()
  File "/esphome/esphome/coroutine.py", line 245, in flush_tasks
    next(task.iterator)
  File "/esphome/esphome/__main__.py", line 148, in wrapped
    await coro(conf)
  File "/esphome/esphome/components/uart/__init__.py", line 185, in to_code
    await debug_to_code(config[CONF_DEBUG], var)
  File "/esphome/esphome/components/uart/__init__.py", line 151, in debug_to_code
    after = config[CONF_AFTER]
KeyError: 'after'
```

oxan, 2021-11-20:
This is fixed by esphome/esphome#2753.

(Also, next time please put your code in code-blocks (triple backticks), as that preserves indentation and makes it readable).

closed 2021-11-20 by oxan

Tonguc-Endem, 2021-11-20:
> This is fixed by [esphome/esphome#2753](https://github.com/esphome/esphome/pull/2753).
thnx a lot for adding this as a milestone for 2021.11.2
> (triple backticks)
yes, I was looking for that too. sorry, this was my first issue post ever.