esphome/issues #2740: ESP-IDF Dallas scratch pad invalid checksum

cmezh, 2021-11-20:
### The problem

After switching to ESP-IDF framework Dallas component has started throwing a lot of warnings about scratch pad invalid checksum.
There are no such warnings if Arduino framework is used.
With Dallas hub update interval of 1s it throws warnings every 3-5 seconds.

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.4

### What platform are you using?

ESP32-IDF

### Board

ESP32-WROOM-32U custom hardware setup

### Component causing the issue

Dallas

### Example YAML snippet

```yaml
esp32:
  board: esp32dev
  framework:
    type: esp-idf
    version: latest

dallas:
  - pin: 26
    id: "dallas_fast"
    update_interval: 1s

sensor:
  - platform: dallas
    dallas_id: "dallas_fast"
    address: 0x4400000D6A383928
    name: "Boiler temperature"
    id: "boiler_temperature"
    internal: true
```

### Anything in the logs that might be useful for us?

```txt
INFO Reading configuration /config/esphome/boiler.yaml...
INFO Starting log output from 192.168.1.3 using esphome API
INFO Successfully connected to 192.168.1.3
[15:50:59][I][app:099]: ESPHome version 2021.11.1 compiled on Nov 20 2021, 15:22:40
[15:51:06][W][dallas.sensor:129]: 'Boiler temperature' - Scratch pad checksum invalid!
[15:51:15][W][dallas.sensor:129]: 'Boiler temperature' - Scratch pad checksum invalid!
[15:51:23][W][dallas.sensor:129]: 'Boiler temperature' - Scratch pad checksum invalid!
[15:51:28][W][dallas.sensor:129]: 'Boiler temperature' - Scratch pad checksum invalid!
[15:51:38][W][dallas.sensor:129]: 'Boiler temperature' - Scratch pad checksum invalid!
[15:51:47][W][dallas.sensor:129]: 'Boiler temperature' - Scratch pad checksum invalid!
[15:51:49][W][dallas.sensor:129]: 'Boiler temperature' - Scratch pad checksum invalid!
[15:52:00][W][dallas.sensor:129]: 'Boiler temperature' - Scratch pad checksum invalid!

...and much more similar lines...
```

### Additional information

DS18B20 sensor is powered from 3v3 line and its data line is pulled up to 3v3 externally with 2k2 resistor.

cmezh, 2021-11-20:
[image]
Here is Home Assistant screenshot. Blue line represents temperature readings. Left side - Arduino framework, right side - ESP-IDF framework, gap in the middle - OTA update.

closed 2022-04-04 by github-actions[bot]

WiebKastanje, 2022-04-25:
Exact Same problem here:
[21:06:33][W][dallas.sensor:137]: 'Temperature: wandlicht-west' - **Scratch pad checksum invalid!**

ESPHome version: 2022.4.0
"device_info": {
        "uses_password": false,
        "name": "wandlicht-west",
        "mac_address": "**REDACTED**",
        "compilation_time": "Apr 25 2022, 20:53:14",
        "model": "mhetesp32devkit",
        "has_deep_sleep": false,
        "esphome_version": "2022.4.0",
        "project_name": "",
        "project_version": "",
        "webserver_port": 0
      },

"sensor": [
        {
          "object_id": "temperature_wandlicht-west",
          "key": 3408443978,
          "name": "Temperature: wandlicht-west",
          "unique_id": "dallas-0700000003df8d28",
          "disabled_by_default": false,
          "icon": "",
          "entity_category": 0,
          "device_class": "temperature",
          "unit_of_measurement": "\u00b0C",
          "accuracy_decimals": 1,
          "force_update": false,
          "state_class": 1,
          "last_reset_type": 0
        },

lucasimons, 2022-07-30:
i have the same issue when i turn on the light dimmer, when i turn off the temperature sensor works fine
```

status_led:
  pin:
    number: GPIO13              # Led di stato
    inverted: true

switch:
  - platform: gpio
    pin: GPIO12
    name: "Valvola Solare"      # Modificare nome a piacimento da esporre in Home Assistant
    id: rele

binary_sensor:
  - platform: gpio
    pin:
      number: GPIO0
      inverted: True
    name: "Pulsante manuale 1ch3"    # Modificare nome a piacimento da esporre in Home Assistant
    on_click:
      min_length: 100ms
      max_length: 1000ms
      then:
        - switch.toggle: rele   # Versione con comando a pulsante

    
output:
  - platform: ac_dimmer
    id: dimmer1
    gate_pin: GPIO05   
    zero_cross_pin:
      number: GPIO04 
      mode: INPUT
      inverted: yes

light:
  - platform: monochromatic
    output: dimmer1
    name: Resistenza Dimmer    
    
    
dallas:
  - pin: GPIO2
    update_interval: 30s
    
    
    
# Individual sensors   

sensor:

  - platform: dallas
    address: 0xde3c01f095899c28 
    name: "Temperatura Boiler Solare"    
    resolution: 12      
  - platform: uptime
    name: Uptime Sensor    
```

```
[10:06:32][D][dallas.sensor:144]: 'Temperatura Boiler Solare': Got Temperature=46.2°C
[10:06:32][D][sensor:124]: 'Temperatura Boiler Solare': Sending state 46.25000 °C with 1 decimals of accuracy
[10:06:47][D][sensor:124]: 'Uptime Sensor': Sending state 100.48800 s with 0 decimals of accuracy
[10:07:02][D][dallas.sensor:144]: 'Temperatura Boiler Solare': Got Temperature=46.2°C
[10:07:02][D][sensor:124]: 'Temperatura Boiler Solare': Sending state 46.25000 °C with 1 decimals of accuracy
[10:07:19][D][light:035]: 'Resistenza Dimmer' Setting:
[10:07:19][D][light:046]:   State: ON
[10:07:19][D][light:050]:   Brightness: 5%
[10:07:19][D][light:084]:   Transition length: 1.0s
[10:07:32][W][dallas.sensor:137]: 'Temperatura Boiler Solare' - Scratch pad checksum invalid!
[10:07:33][D][sensor:124]: 'Temperatura Boiler Solare': Sending state nan °C with 1 decimals of accuracy
[10:07:47][D][sensor:124]: 'Uptime Sensor': Sending state 160.49001 s with 0 decimals of accuracy
[10:08:02][W][dallas.sensor:137]: 'Temperatura Boiler Solare' - Scratch pad checksum invalid!
[10:08:02][D][sensor:124]: 'Temperatura Boiler Solare': Sending state nan °C with 1 decimals of accuracy
[10:08:32][W][dallas.sensor:137]: 'Temperatura Boiler Solare' - Scratch pad checksum invalid!
[10:08:32][D][sensor:124]: 'Temperatura Boiler Solare': Sending state nan °C with 1 decimals of accuracy
[10:08:47][D][sensor:124]: 'Uptime Sensor': Sending state 220.49500 s with 0 decimals of accuracy
[10:09:02][W][dallas.sensor:137]: 'Temperatura Boiler Solare' - Scratch pad checksum invalid!
[10:09:02][D][sensor:124]: 'Temperatura Boiler Solare': Sending state nan °C with 1 decimals of accuracy
[10:09:20][D][light:035]: 'Resistenza Dimmer' Setting:
[10:09:20][D][light:046]:   State: OFF
[10:09:20][D][light:084]:   Transition length: 1.0s
[10:09:32][D][dallas.sensor:144]: 'Temperatura Boiler Solare': Got Temperature=46.2°C
[10:09:33][D][sensor:124]: 'Temperatura Boiler Solare': Sending state 46.18750 °C with 1 decimals of accuracy
```

cmezh, 2022-11-26:
Solved someday somehow. It's been long since I've tested it, but now it works without any issues.

Ulrar, 2023-01-02:
Interesting, just noticed the same here since I moved to esp-idf.
I'll swap back to arduino for now but I'd love to know how you solved it, if you can remember any changes you made

jazzmonger, 2023-02-08:
I have the exact same problem with this sensor and a Pwm dimmer. Turn on the dimmer and every response is lScratch pad error". Turn off the dimmer and it works fine.  I've moved gpios, shielded wires, changed power supplies, even looked at the wpdallas waveform while adjusting the pull-up resistor to be optimal.  Nothing solves it. 

How do u use the Arduino framework? I using a D1mini esp8266.

jazzmonger, 2023-02-08:
How do you enable the Arduino framework?

Ulrar, 2023-02-08:
```
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf
```

That's for ESP32.
For esp8266 I believe it's the only option so you don't do anything, it's already using arduino as it doesn't support esp-idf anyway

jazzmonger, 2023-02-08:
Ok, that makes sense.  But the problem remains...

WimImmelman, 2023-02-25:
I have the exact same issue.  When using a single temp sensor, all is good. But as soon as I connect 2, then I get "Scratch pad config register invalid!".  I've also tried, multiple power supplies, dedicate power supplies for the sensors, a magnitude of different wires, wire lengths, etc.  What is very interesting, is that when my sensors are on very short wires (30 cm) they work without this error.  As soon as I extend the wires to 150cm for each sensor I get this error.

Any advice?

pgolawsk, 2023-03-10:
I have the same issue on esp8266:( - randomly Scratch pach error. 2 Dallas sensors connected 1m and 2m length. Pull-up resistor 4.7K

vantop77, 2023-03-22:
Just to confirm, I have the same issue. The setup is 4.7K pull-up on +5V, no voltage translator for the IO pin. Two DS18B20 sensors, one is 1m cable length which returns "Scratch pad config register invalid!", the other sensor is with 15m cable and has no issues.

jazzmonger, 2023-03-22:
Looks like this is a serious issue affecting a LOT of people.

jazzmonger, 2023-03-29:
ok, we've duplicated the problem and are working on a fix.  I've supplied detailed logic probe timing data along w/ info from my intermittent sensor.  I'm also testing various updates to the code as they come in.  We'll post here once we nail this bug.

Lepelot, 2023-04-03:
Hello, I have been struggling with this same issue for the last few days. Same problem with 3 dallas sensors which were working before I started to change the ESP32 boards with a more definitive wiring. I don't know if this will help, but here is a copy of my log on a test setup. Binary sensors on the same boards are woring fine. 
Thank you for your help. Rgds. 

[image]

INFO Reading configuration /config/esphome/esp-test.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from esp-test.local using esphome API
INFO Successfully connected to esp-test.local
[11:20:46][I][app:102]: ESPHome version 2023.3.2 compiled on Apr  3 2023, 10:37:31
[11:20:46][C][wifi:504]: WiFi:
[11:20:46][C][wifi:362]:   Local MAC: C8:F0:9E:F2:8B:60
[11:20:46][C][wifi:363]:   SSID: 'Freebox-xxxxxx'[redacted]
[11:20:46][C][wifi:364]:   IP Address: 192.168.1.xx
[11:20:46][C][wifi:366]:   BSSID: DC:00:B0:FF:A4:08[redacted]
[11:20:46][C][wifi:367]:   Hostname: 'esp-test'
[11:20:46][C][wifi:369]:   Signal strength: -76 dB ▂▄▆█
[11:20:46][C][wifi:373]:   Channel: 8
[11:20:46][C][wifi:374]:   Subnet: 255.255.255.0
[11:20:46][C][wifi:375]:   Gateway: 192.168.1.254
[11:20:46][C][wifi:376]:   DNS1: 192.168.1.254
[11:20:46][C][wifi:377]:   DNS2: 0.0.0.0
[11:20:46][C][logger:293]: Logger:
[11:20:46][C][logger:294]:   Level: DEBUG
[11:20:46][C][logger:295]:   Log Baud Rate: 115200
[11:20:46][C][logger:296]:   Hardware UART: UART0
[11:20:46][C][dallas.sensor:075]: DallasComponent:
[11:20:46][C][dallas.sensor:076]:   Pin: GPIO4
[11:20:46][C][dallas.sensor:077]:   Update Interval: 60.0s
[11:20:46][D][dallas.sensor:082]:   Found sensors:
[11:20:46][D][dallas.sensor:084]:     0xea70e8071e64ff28
[11:20:46][C][dallas.sensor:089]:   Device 'Temperature bureau'
[11:20:46][C][dallas.sensor:089]:     Device Class: 'temperature'
[11:20:46][C][dallas.sensor:089]:     State Class: 'measurement'
[11:20:46][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[11:20:46][C][dallas.sensor:089]:     Accuracy Decimals: 1
[11:20:46][C][dallas.sensor:097]:     Address: 0xbc012152c1d3a128
[11:20:46][C][dallas.sensor:098]:     Resolution: 12
[11:20:46][C][captive_portal:088]: Captive Portal:
[11:20:46][C][mdns:108]: mDNS:
[11:20:46][C][mdns:109]:   Hostname: esp-test
[11:20:46][C][ota:093]: Over-The-Air Updates:
[11:20:46][C][ota:094]:   Address: esp-test.local:3232
[11:20:46][C][ota:097]:   Using Password.
[11:20:46][C][api:138]: API Server:
[11:20:46][C][api:139]:   Address: esp-test.local:6053
[11:20:46][C][api:143]:   Using noise encryption: NO
[11:21:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:21:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy
[11:22:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:22:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy
[11:23:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:23:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy
[11:24:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:24:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy
[11:25:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:25:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy

Lepelot, 2023-04-03:
Hello, it looks like my previous comment didn't register, so here we go again: I have been struggeling with the same issue, (I think)  for the last few days. Same issue with 3 Dallas sensors on ESP32 boards which were working before I started to clean up my wiring . I don't know if it will help, but here is a copy of the log on a test setup I just ran this morning. 
Thank you for your help. Rgrds 

INFO Reading configuration /config/esphome/esp-test.yaml...
WARNING GPIO4 is a Strapping PIN and should be avoided.
Attaching external pullup/down resistors to strapping pins can cause unexpected failures.
See https://esphome.io/guides/faq.html#why-am-i-getting-a-warning-about-strapping-pins
INFO Starting log output from esp-test.local using esphome API
INFO Successfully connected to esp-test.local
[11:20:46][I][app:102]: ESPHome version 2023.3.2 compiled on Apr  3 2023, 10:37:31
[11:20:46][C][wifi:504]: WiFi:
[11:20:46][C][wifi:362]:   Local MAC: C8:F0:9E:xxxxxxxxxxx
[11:20:46][C][wifi:363]:   SSID: 'Freebox-6Axxxxx'[redacted]
[11:20:46][C][wifi:364]:   IP Address: 192.168.1.xx
[11:20:46][C][wifi:366]:   BSSID: DC:00:B0:FF:A4:08[redacted]
[11:20:46][C][wifi:367]:   Hostname: 'esp-test'
[11:20:46][C][wifi:369]:   Signal strength: -76 dB ▂▄▆█
[11:20:46][C][wifi:373]:   Channel: 8
[11:20:46][C][wifi:374]:   Subnet: 255.255.255.0
[11:20:46][C][wifi:375]:   Gateway: 192.168.1.254
[11:20:46][C][wifi:376]:   DNS1: 192.168.1.254
[11:20:46][C][wifi:377]:   DNS2: 0.0.0.0
[11:20:46][C][logger:293]: Logger:
[11:20:46][C][logger:294]:   Level: DEBUG
[11:20:46][C][logger:295]:   Log Baud Rate: 115200
[11:20:46][C][logger:296]:   Hardware UART: UART0
[11:20:46][C][dallas.sensor:075]: DallasComponent:
[11:20:46][C][dallas.sensor:076]:   Pin: GPIO4
[11:20:46][C][dallas.sensor:077]:   Update Interval: 60.0s
[11:20:46][D][dallas.sensor:082]:   Found sensors:
[11:20:46][D][dallas.sensor:084]:     0xea70e8071e64ff28
[11:20:46][C][dallas.sensor:089]:   Device 'Temperature bureau'
[11:20:46][C][dallas.sensor:089]:     Device Class: 'temperature'
[11:20:46][C][dallas.sensor:089]:     State Class: 'measurement'
[11:20:46][C][dallas.sensor:089]:     Unit of Measurement: '°C'
[11:20:46][C][dallas.sensor:089]:     Accuracy Decimals: 1
[11:20:46][C][dallas.sensor:097]:     Address: 0xbc012152c1d3a128
[11:20:46][C][dallas.sensor:098]:     Resolution: 12
[11:20:46][C][captive_portal:088]: Captive Portal:
[11:20:46][C][mdns:108]: mDNS:
[11:20:46][C][mdns:109]:   Hostname: esp-test
[11:20:46][C][ota:093]: Over-The-Air Updates:
[11:20:46][C][ota:094]:   Address: esp-test.local:3232
[11:20:46][C][ota:097]:   Using Password.
[11:20:46][C][api:138]: API Server:
[11:20:46][C][api:139]:   Address: esp-test.local:6053
[11:20:46][C][api:143]:   Using noise encryption: NO
[11:21:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:21:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy
[11:22:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:22:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy
[11:23:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:23:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy
[11:24:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:24:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy
[11:25:12][W][dallas.sensor:261]: 'Temperature bureau' - Scratch pad checksum invalid!
[11:25:12][D][sensor:127]: 'Temperature bureau': Sending state nan °C with 1 decimals of accuracy

jazzmonger, 2023-04-12:
Guys, try this alternate Dallas sensor driver and see if it fixes any of your problems.
It works on my ESP32-S2 WITHOUT errors.

```
external_components:
  - source: github://nrandell/dallasng

dallasng:
  - pin: GPIO6
    update_interval: 3s

sensor:
  - platform: dallasng
    name: Temperature
    index: 0
    resolution: 12
    filters:
      - filter_out: NAN
      - 
   ```

Quantum-37, 2023-04-13:
> Guys, try this alternate Dallas sensor driver and see if it fixes any of your problems. It works on my ESP32-S2 **WITHOUT** errors.
>
> ```
> external_components:
>   - source: github://nrandell/dallasng [...]

I get this:

`[05:38:36][E][dallasng.sensor:090]: Failed to reset the bus: 1`
`[05:38:40][W][dallasng.sensor:111]: 'Temperature' failed to read scratchpad: 3`
`[05:38:43][D][dallasng.sensor:119]: 'Temperature' got temperature as 25562 (25.562000)`
`[05:38:43][D][sensor:126]: 'Temperature': Sending state 25.56200 °C with 1 decimals of accuracy`
`[05:38:43][D][switch:013]: 'Relay' Turning ON.`
`[05:38:43][D][climate:392]: 'Thermostat' - Sending state:`
`[05:38:43][D][climate:395]:   Mode: HEAT`
`[05:38:43][D][climate:397]:   Action: HEATING`
`[05:38:43][D][climate:415]:   Current Temperature: 25.56°C`
`[05:38:43][D][climate:418]:   Target Temperature: Low: 28.00°C High: 32.00°C`
`[05:38:43][D][climate:392]: 'Thermostat' - Sending state:`
`[05:38:43][D][climate:395]:   Mode: HEAT`
`[05:38:43][D][climate:397]:   Action: HEATING`
`[05:38:43][D][climate:415]:   Current Temperature: 25.56°C`
`[05:38:43][D][climate:418]:   Target Temperature: Low: 28.00°C High: 32.00°C`
`[05:38:46][W][dallasng.sensor:111]: 'Temperature' failed to read scratchpad: 3`
`[05:38:49][D][dallasng.sensor:119]: 'Temperature' got temperature as 25562 (25.562000)`
`[05:38:49][D][sensor:126]: 'Temperature': Sending state 25.56200 °C with 1 decimals of accuracy`

jazzmonger, 2023-04-13:
Did you comment out references to the other sensor?? It looks like you're trying to use both...

Lepelot, 2023-04-13:
See update in my earlier post. Thanks for helping.

jazzmonger, 2023-04-13:
@Quantum-37 
Try index =0 instead of address.
```
sensor:
  - platform: dallasng
    name: Temperature
    index: 0
    resolution: 12
    filters:
      - filter_out: NAN```

Quantum-37, 2023-04-13:
> @Quantum-37 Попробуйте index = 0 вместо адреса.
> 
> ```
> sensor:
>   - platform: dallasng
>     name: Temperature
>     index: 0
>     resolution: 12
>     filters:
>       - filter_out: NAN
> ```

Thank you very much. I will test it.

Quantum-37, 2023-04-13:
> @Quantum-37 Try index =0 instead of address.
> 
> ```
> sensor:
>   - platform: dallasng
>     name: Temperature
>     index: 0
>     resolution: 12
>     filters:
>       - filter_out: NAN
> ```

It's a pity, but the error remained, although it began to manifest itself less often. Periodically, the error began to appear: "[dallasng.sensor:090]: Failed to reset the bus: 1"

haifeng-x, 2023-05-08:
I met the same problem and solved it by changing the address to index.
external_components:
  - source: github://nrandell/dallasng

dallasng:
  - pin: GPIO5
    update_interval: 3s

# Individual sensors
sensor:
  - platform: dallasng
    index: 0
    name: "Living Room Temperature"
    resolution: 12
    filters:
      - filter_out: NAN

manonfgoo, 2023-05-15:
I had the same issue, and it went away after I changed the Power Supply.

MasterCATZ, 2023-08-20:
> Guys, try this alternate Dallas sensor driver and see if it fixes any of your problems. It works on my ESP32-S2 **WITHOUT** errors. Make sure you comment out all other references to other Dallas drivers.
> 
> ```
> external_components:
>   - source: github://nrandell/dallasng
> 
> dallasng:
>   - pin: GPIO6
>     update_interval: 3s
> 
> sensor:
>   - platform: dallasng
>     name: Temperature
>     index: 0
>     resolution: 12
>     filters:
>       - filter_out: NAN
> ```

THANK YOU 

this worked perfectly with my 

```
esp8266:
  board: esp01_1m

```

so why is the default  "Dallas" still stuffed all these years later !!! 
normal code kinda works on 1 of the 10 esp's  but my others keep throwing the errors regardless what power source or probes were connected to it 
changed to alternate code and it worked , now tempted to change code on the working one to see if the long hangups it gets randomly gets stops 

```
external_components:
  - source: github://nrandell/dallasng

dallasng:
  - pin: GPIO2
    update_interval: 3s

#dallas:
#  - pin: GPIO2  #D4
#    id: hub_D4
#    update_interval: 30s
#  - pin: GPIO3 #D?
#    id: hub_D
#    update_interval: 30s
sensor:
  - platform: dallasng
    index: 0
    name: "Air Temp N2"
    resolution: 12
    filters:
      - filter_out: NAN
#  - platform: dallas
#    address: 0xc205170112bdff28
#    resolution: 9
#    name: "Reefer 1 Air Temp N1"
#    id:  R1ATN1
```
```
[18:16:20][D][sensor:093]: 'Air Temp N2': Sending state 26.37500 °C with 1 decimals of accuracy
[18:16:23][D][dallasng.sensor:119]: 'Air Temp N2' got temperature as 26375 (26.375000)
[18:16:23][D][sensor:093]: 'Air Temp N2': Sending state 26.37500 °C with 1 decimals of accuracy
[18:16:26][D][dallasng.sensor:119]: 'Air Temp N2' got temperature as 26375 (26.375000)
[18:16:26][D][sensor:093]: 'Air Temp N2': Sending state 26.37500 °C with 1 decimals of accuracy
```