esphome/issues #2741: DSMR: Invalid identification string

RonNabuurs, 2021-11-20:
### The problem

After reading multiple post about some DSMR issue and unable to resolve it, here's my post.

I've tried all the possible versions you can imagine on my ESP32 and al of the possible combinations of configuration, as far as I know.

Slimmemeter: Kaifa MA304C

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

nodemcu

### Component causing the issue

dsmr

### Example YAML snippet

```yaml
esphome:
  name: slimmelezer
  platform: ESP32
  board: nodemcu-32s
  platformio_options:
    lib_deps: glmnet/Dsmr@0.5
    lib_ignore: glmnet/Dsmr@0.3

external_components:
  - source:
      type: git
      url: https://github.com/esphome/esphome
      ref: dev
    components: [ "dsmr" ]
    refresh: 60s

# Enable logging
logger:
  level: DEBUG
  baud_rate: 0

captive_portal:

# Slimmelezer
uart:
  baud_rate: 115200
  rx_buffer_size: 3000
  rx_pin: 
    number: GPIO16
    inverted: true
  # debug:
  #   direction: RX
  #   after:
  #     delimiter: "\r\n"
  #   sequence:
  #     - lambda: UARTDebug::log_string(direction, bytes);

dsmr:
  id: dsmr_instance
[... 49 lines cut ...]
[15:00:42][E][dsmr:171]: /
 ^
Invalid identification string
[15:00:42][D][api.connection:157]: Home Assistant 2021.9.7 (<redacted>) requested disconnected
[15:00:42][D][sensor:113]: 'SlimmeLezer Uptime': Sending state 3664.61694 s with 0 decimals of accuracy
[15:00:42][D][api:098]: Accepted <redacted>
[15:00:42][D][api.connection:747]: Home Assistant 2021.9.7 (<redacted>): Connected successfully
[15:00:43][D][uart_debug:152]: <<< "@\xFF\n\x00\x80\xFF\x03\x00\x00\xE7\x02\x01\x80\xFE\t\x00\xF0\x00\x00\x00q\x04\x00\xF8\x05\x00\x00\x00\xF8\x00\x00\x80\v\x00\x00\xE0\x00\x00\x00\xFE-\x00\x00\xE0\x05\x00\x00\xA0\x16\x10\x80\xFF\x01\x00\x00\xE0\v\x00\x00\x00\xF8\x00\x00\x00\x00\xF8\x00\x00\x00\xF9\x02\x00\xD0\x00\x00\x00\xA0\x00\x00\x80\xFFA\x00\x00\x00\xFC\x00\x00\xD0\a\x00\xC0\x10\x00\x00\xC0\x03\x00\xE0\xFF\x12\x00\x00\x00\xF8\x00\x00\x00\xE4\x06\x00\x00\xE0\x00\x00\x00\x90\xFF\x00\x00\x00\xFE\x00\x00\xE8
[15:00:45][D][uart_debug:152]: <<< "\xFA\x00\x00\xF0\x06\x00\xE0\x00\x00\x00\xFA\x06\x00\x00\xFE\x0F\x00\x00\xFE\x06\xF0\x00\xDA\x00\xA0\xFF\x00\xF01\x00\xE0\x00\x00\xF8\x00\x00\x00\xF0\x02\x00\x00\xFA\x00\xC2\n\x00\x00\xFC\x00\x00\xFA\x00\x00\xF8\x00\xC0\x04\xE0\x01\x00\x00\xD0\x00\x80\x00\x00\x00\xFA\n\x00\xF0\x02\x00\xF0\x00\"\x03\x00p\x00\xE0\x00\xF0\x02\x00\x00\xA0\xFF\x03\x00\xFA\x01\xB0\x00\xF0\x00\xC0\x00\x12\xFF\x00\x00\x00\xF8\x00\x00\x00\xD0\x04\xDC\x00\xA0\x02\x00\xE0\x00\x00\xFA\x0E\x00\x80\x00\x
[15:00:47][D][uart_debug:152]: <<< "o \xFF\n\x00\x00\xFC\x06\x00\x00\xF8\x01\x00\x00\xFE\x00\x00\xFE\x03\x00\xFE\b\xCE\x00P\x00\x00\x80\xFC\x00\x00\xFF\x17\x00\x00\xF8\x00\x00\xFE\x06\x00x\x00\x00\x00\x00\xF8\x03\x00\xA0\x04\x00\x00@\x04\xF0\x02\x00\x00\xF0\x1F\x00\x00\xA0\x00\x00\xFF\x00\x00\xFE\x05\x00\xF8\x02\x00\xF0\x00\xF0\x020\x00\x80\x00\x00\xFF\x00\x00\xFF\x02\x00\x00\xF0\x04\x00\x04\x00\x00\x00\xFF\x00\x00\xF0\x03\xD0\x00\x00\xF4\x00\x00\x00\xF8\x01\xF0\xFF\x04\x00 \x00\x00\xFA\x01\x00\xFE\x00\xE0\x0
[15:00:48][D][uart_debug:152]: <<< "\xFA\x00\x00\x00\xF8\x00\x00\x00\x00\xF5\x04\x00\x00\xE0\x00\x00\x00\x00\xFA\x00\x80\x01\x00\x00\xFA\r\x00 \x00\x00\xFE\x00\x00\xF0\xFF\x01\x00\xFA\x00\xFE\x17\x00\x00\xF0\x04\xB0\x01\x00\xA0\x00\xC0\x02\x00\xE0\x04\x00\x00\xFE\x00\xDA\x06\xC8\x02\xC0\x00\xE0\a\x00\xF0\x02\x94\x01\x00\xF0\x00\x00\x00\x00\xFE\x00\x00\x80\xFB\x04\x00\x00\xFE\x02\x00\x00\x00\x00\xFE\x04\x00\x00\xF0\x00\x00\x00\xC0\xFF\x00\x00\x00\xF0\x02\xF0\x02\xF0\x01z\x02\x00\x00\xF0\x01\x00\xE0\x02\xE0\x00
[15:00:50][D][uart_debug:152]: <<< "\xFE\x05\x00\x00\xFD\x05\x00\x00\x00\x00\x00\xF0=\x00\xE0\b\xF0\x00\x00\xFE\x02\x00\x00\xFE\x16\x00\x00x\x04\xE0\x86\xE0\x00\xE0\x04\x00\x00\xFF\x03\x00\xF0\x00\x00X\x04\xD4\x01\xD0\xFF\x04\xE0\x01\x00\xE0\x00\xE0\x00\x00\xFF\x14\x00\xFA\x00\x00\x00\xF0\v\x00\x00\xFF\n\x00\x00\x00\xFE\x02\x00\x00\xFE\x00\x00\x00\xF0\x00\xF0\a\x00\xFE\x02\x00 \xFE\x06\x10\x00\xFE\x04\xF0\x00\xF0\x00\x00\x00\xF4\n\x00\x00\x00\xE0\x00\xA0\x02\x00\x00\x00\x00\xFA\x10\x00\x00\x00\xF0\x00\x00\xFA
[15:00:52][D][uart_debug:152]: <<< "\x00\xFF\b\x00\x00\x02\x00\x00\xE0\x00\x00\x00\xE0\x04\x00\x00\xFA\xFF\x00\x00\xFE\x06\x00\xF0\x02\xA0\x00\xF0\x00\xE0\v\x00\x00\x00\x00\x00\x00\x00\xF0\x00\x00\xF0\x00\xF0\x00\xE0\x00\x80\x00\x00\x00\xF0\x00\x88\x03\x92\x00\x00\xFA\x02\x00\x00\x80\x04\x00\xE1\x06\x00\xFA\x00\xFA\xFF\b\x00\xFF\x06\xF8\x00\x00\xFF\x06\x00\xF8\x04\xF0\x02@\x00\x00\xB0\x03\x00\xFF$\xF0\x06\x00x \xCA\x00\xF0\x03\x00\xF8\x06\x00\xCC\x03 \x03\x00\x00\xE2\x00\xE0\x02\x00\x00\xFE\x04\xC0\x00\xE0\x0
[15:00:53][D][uart_debug:152]: <<< "\xFE\x04\x00\x00\xFC\n\x00\x00\xF2\x00\x00\x00\x80\"\x00\x00\x00\xFE\x00\x00\x00\xFF\x02\xF8\x04\x00\xF0\x00\x00\x01\x00\xF0\x02\x00\xC0\x03\x00\xE0\x00\x00\xF0\xFF\x04\x00\xF8\x02\x00\x00\xB0\x00\x00\x00\xC0\x02\x00\x00\xF0\x00\x00\x80\x00\x00\xF0\x02\xF0\x04\xE0\x03\xC0\x03\x00\xE0\x02\x80\x00@\xF4\x03\x00\xF0\x00\x00\x00\xF0\x00\xD0\x00\x00\xFE\x02\x00\xF0\x00\x00\xC0\x00\xC0\x00@\x00\x00\x00\xFE\x00\x00\xF0\x01\x00\xFA\x03\x00\x00\x06\x00\x00\x00\xFF\x00\xD0\x00\xFA\x00
[15:00:55][D][uart_debug:152]: <<< "\xF0\x00\x00\x80\xFE\x00\x00\x00\xE0\x04\x00\x00\xF9\x00\xA0\x00\xE0L\x00\x00\x00\xFF\x00\x00\xF0\x04\xF0\x02\x00\xF0\x02\x000\x05\x00\x00\xC0\x00\xC0\x02\x00\xFA\x02\xF0\x00\x00\x00\xFF\x0F\x00\x00\x00\xFF\x02\x00\xC0\x00\xD2\x00\x00p\xFF\x00\x80\xFF\x00\x00\xE0\x02\x00\x00\x00\xFF\x04\x00\x00\xF0\x02\xF8\x00\xCA\x06\x00\x00\xFF\x04\x00\x02\x00\x0F\x00\xF0\x04\x00\x00\xF0\x00\x00\x80\x00\x00\x00\x80\x02\x00\x00\xF8\x00\x00\x00\xE0\xFF\x02\xE0\x00\xF0\x02\x00D\x0E\x00\x00\x
[15:00:57][D][uart_debug:152]: <<< "\xFF\x04\x00\x00\x00\xFF\n\x00\x00\xF0\x00\x00\x00\xFF\n\x00\xE0\x00\xE0\x00\x00\x00\xFC\x00\xF0\x02\x00\x00\xFF\x00\x00\xFA\x04\xDA\x00\x00\xF0\x02\xF0 \x00\xD0\x00\x80\x00\xC0\x0E\x00|\x00\xF0\x00\x00\xE0\x00\xC8@\xE1\x00\x00\xFE\x04\xE0\x00\x80\x00\x00\x00 \x03\x00\x00\xF1\x00\x00\x00\xF0\x00\x00\xE0\x04\xF8\x03\x00\xF0\x03\x00\xC0\a\x00\xF0\x02\xE0\x06\xF0\x00\xF8\x00\xE0\xFF\x02\x00\xE8\x00\xF0\x03\xE0\x04\xF0\x010\x00\xF0\x00\x00\xF8\x00\xF0\vp\xFF\x00\x00\xF8\r\x00\x
[15:00:59][D][uart_debug:152]: <<< "\x01\x04\x00\xFE\v\x00\x00\x00\x80\xFF\x03\x00\x00\xF0\x00\x00\xFE\x01\x00\xF8\x03\x00\xF0\x00\xF0\x04\x00\x000\x00\xD0\x02\xC0\x02\x00\x00\x00\xFE\x00\x00\x00\xE0\x02\xD0\x04\x00\xC0\x00\x00\xFA\x02\x00\x00\x00\x00\x80\xFF\x00\x80\x00\x00 \x04\x00\x00\xFE\x04\x00\x00\xF0\x00\x00\x00\xFF\x01\x00\xF0\x00\x00\x00\xFF\x00\xD0\x10\xE0\x06\x00\x00\xE8\x00\x00\xFF\x00\xA0\xFF\x00\xE2\x06\xF0\x02\xF0\x00\xF0\x03\x00\x00\xF8\x00\x00/\x00\xE0\x00\x00\x00\xD0\x1E\x00\xC0\x00\x00\x00\
[15:01:00][D][uart_debug:152]: <<< "D\x00\x00\x00\xC0\xAF\x00\x00\xFE\x00\x00\x00\xF8\x04\x00\xF0\x02\x00\x00\xFE\x03\x00\xF0\x00\x00\x00\xFE\x12\xC0\x00\x00\xF0\x00\xE0\x00\x00\x00\x00\xF8\x05\x00\x80\x00\x00\xFA\x00\xE0\x02\x00\xFA\x04\x00\xE0\x00\x00\xFA\x03\x00\xDA\x00\x00\xFE\x00\x00\x00\xFA\x00\x00\xF8\x03\x00p\xFF\x04\x00\x00\x00\xF0\x00\xC0\x00\x00\xFE\x00\x00\x00\xFE\x00\xD8\x02\x00\xB0\x04\xC0\x00\x00\xFE\x04\x00\x80\xF7\x06\x00\xF0\xFF\v\x00\x00\xFF\x00\x00\xFA\x00\x00\xE0\x03\xC0\n\x00\xF0\x00\x00
[15:01:02][D][uart_debug:152]: <<< "\xFE\x0F\x00\x80\x00\x00\x00\xFC\x00\x00\x00\xFC\xE7\x00\x00\xC0\xFF\x00\x00\xFF\n\x00\x00\x06\x00\n\x00\x00\xF0\x03\xC0\x00\x00\x01\x000\x00\x00\x00\xF0\x00\xC0\x00`\x00\x00\xF0\x04\x00\x00\xE0\x02\x00\xF0\x02\xE0\x00\x00\xF0\x10\x00\xFF\t\x00\x00\xF0\x00\x00\xF8\x00\xC0\x00\x00\xFE\x04\x00\xE2\x00\x00\xE0\x1F\x00\x80\x00\x00\x03\x00\xF0\x01\xE0\x00\x00\x00\xF0\x02\x00\x00\x00\xFC\x03\x00\x00\x00\xF8\x00\xE8\x00\x00\xFF\x00\x00\xD0\x01\x00\x00\x00\xFA\x00\x00X\x06\xC4\x04\
[15:01:04][D][uart_debug:152]: <<< "\x00\xF0\x01\x00\xA0\x03\x00\x00\xF0\x06\x00\x00\xF0\x02\x00\xC0\x06\x00\x00p\x00\xC0\x02\x00\xF0\x03\x00\x00p\x00\xE1\x02\x00\x00\xF0\x00\xE0\x04\x00\x00\x00\xFA\x00\x00\x00}\x03\x00\x00\xF0\x00\xE8\x00\x00\xFF\x02\xE4\x00\x00\xFA\x06\x00\x00\xFE\x00\xF0\b\x00\xFA\x00\x00\xE0\x00\x00\xE0\x02\x00\xE0\x00\x00\x00\xFD\x00\x00P\xFF\x00\x00\x00\xFC\x04\x00\x00\xF8\x00\x00\xC0\xFF\x00\xE0\x00\x00\xB0\x00\x00\xFA\x06\x000\xFF\xEF\x00\x00\x00\xA0\xFF\x00\x00\x80\xFF\x00\x00\x88\x0
[15:01:06][D][uart_debug:152]: <<< "(\xFE\x04\x00\xE0\f\x00\x00\x00\xF8\x00\x00\x00H\x03\x04\xC0\x02\x00\xF0\x03\xD8\x02\x00\x00\xFE\x00\xF0\n\x00\x00\x80\xFF\x02\x00\x02\x00\xF8\x01\xE0\x01\x00\x00\xF0\x00\x00\xF0\x00 \x04\x00\x00\xF8\x02\x00\xF8\x01\x00\xFA\x02\x00\xF0\x04\xE2\x05\x00\x00\xF0\x06\xE0\x00\x00\xFA\x06\x00\x80\xFC\x06\x00\xF0\x00\x00\x00\x00\x00\xFE@\xF0!\x00\xC0\x02\x00\xF2\x00\xF0\x04\xD0\x01\xE0\x04\x00\x00\xFA\x00\xF0\x00\x00\xFF\x02\x00\xF0\x00\xC0\x00\x00\xF8\x00\x00\xE0\x04\x00\xF0\x00\
[15:01:06][E][dsmr:171]: /
 ^
Invalid identification string
```

mmakaay, 2021-11-20:
That data does not look like unencrypted dsmr data at all. That smart meter should be sending dsmr 4.2.2 data though.
My first thought was dat possibly the uart signal wasn't inverted, but I see that this was added to the config.

RonNabuurs, 2021-11-20:
@mmakaay Could it be my temporary breadboard solution to connect the p1 port with my ESP32. Maybe a bad connection? I would expect no data at all when I have a bad connection though. 

And as you said, I first also thought to invert the signal, but still got the "same" result as in the post above. Even without it being inverted I get hex values. So I'm not sure what to do here 🤔

mmakaay, 2021-11-20:
How did you connect the data pin to your ESP32?
The signal is inverted, with the P1 being an open drain. 
Therefore, the pin needs a pull up resistor to force the signal go HIGH when the P1 port is open.

RonNabuurs, 2021-11-20:
At the moment I have the data pin directly connected to the ESP32 gpio and the 5v from te P1 port connected to the RTS pin(so I get data). I'll try to use a pull up resistor tomorrow and report back here. Any idea of how many ohms the pull up resistor should be?

RonNabuurs, 2021-11-21:
Tested with a 10k resistor (I did not have anything else) from the data line to 3.3v on de ESP32, but now I get only empty data.

```
[17:01:46][D][uart_debug:152]: <<< "\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x
```

RonNabuurs, 2021-11-21:
This wall of `\x00` stops When I use a seperate 3.3v (from the esp32) or 5v (from a random arduino I had lying around) on the RTS pin of the P1 port. So I guess you cannot use the 5v supply from the p1 port it self to power the RTS pin?

mmakaay, 2021-11-21:
You can power the P1 RTS pin from its own +5V. That is how most devices are wired.

mmakaay, 2021-11-22:
Maybe checkout this one?: https://trmm.net/Smartmeter/ 
The circuit design there includes the pull up resistor as I suggested, but also a capacitor to GND for filtering noise.

RonNabuurs, 2021-11-22:
After using the schematic from the link you send, I got a working reader! Though I didn't use the capacitor. I think my problem was that I didnt have the data ground connected to ground from the P1 port.

Here the final schematic I used so other people can use it.
[image]

closed 2021-11-22 by RonNabuurs

mmakaay, 2021-11-24:
Ah yes, data ground is very important to have connected to the MCU ground. Well done, thanks for posting the schematic 👍