esphome/issues #2742: PSK doesn't work for password

sotnikov-link, 2021-11-20:
### The problem

I cannot use PSK password for WiFi connection, but simple password works.

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

pip

### Which version of Home Assistant has the issue?

None

### What platform are you using?

ESP32

### Board

nodemcu-32s

### Component causing the issue

wifi

### Example YAML snippet

```yaml
esphome:
  name: test-psk
  platform: ESP32
  board: nodemcu-32s

logger:

wifi:
  ssid: my-net
  # Next line doesn't work
  password: da8c0012575d312157b972b8cfac617e114b7c54d9d79c3412b94bac9549ade7

  # Line without PSK will work
  # password: my-pass
```

### Anything in the logs that might be useful for us?

```txt
(esp32-TBEhM836-py3.8) bash-5.1$ esphome run test.yml 
INFO Reading configuration test.yml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
********************************************************************************************************************************************************************************************************
Obsolete PIO Core v5.2.2 is used (previous was 5.2.3)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-platformio-cores-in-a-system
********************************************************************************************************************************************************************************************************
Processing test-psk (board: nodemcu-32s; framework: arduino; platform: platformio/espressif32 @ 3.3.2)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
Dependency Graph
|-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
Compiling .pioenvs/test-psk/src/esphome/components/esp32/core.cpp.o
Compiling .pioenvs/test-psk/src/esphome/components/esp32/gpio_arduino.cpp.o
Compiling .pioenvs/test-psk/src/esphome/components/esp32/gpio_idf.cpp.o
Compiling .pioenvs/test-psk/src/esphome/components/esp32/preferences.cpp.o
[... 150 lines cut ...]
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
[01:06:28][W][wifi:554]: WiFi Unknown connection status 0
```

### Additional information

OS: macOS Monterey 12.0.1 (Intel)

sotnikov-link, 2021-11-20:
I use https://www.wireshark.org/tools/wpa-psk.html for PSK.

oxan, 2021-12-01:
I think this is just unclear documentation (PSK is sometimes used as synonym for the passphrase), as the ESP-IDF doesn't seem to support supplying the key directly.

sotnikov-link, 2021-12-01:
One person from Russian chat checked and confirmed this bug only for his ESP32 but his ESP8286 works well with PSK: https://t.me/esphome/86851

closed 2022-04-16 by github-actions[bot]