esphome/issues #2743: External regression - Si1145 Project doesn't compile anymore

Lefuneste83, 2021-11-21:
### The problem

I have noticed that I cannot compile my weather station project anymore. It has been working for years but probably due to a change in the libraries dependencies, it refuses to compile anymore, It seems to miss one library since the last updates in Esphome.

`/data/weather/.piolibdeps/weather/Adafruit SI1145 Library/Adafruit_SI1145.h:25:32: fatal error: Adafruit_I2CDevice.h: No such file or directory`

I have tried to force previous versions of the Adafruit SI1145 Library but with no success. If I place this library Adafruit_I2CDevice.h by hand in the /config/esphome directory, it then complains about Wire.h missing.

I think something has changed with the Adafruit SI1145 Library as it now installs the Adafruit BusIO as a dependency which I believe was not the case.

If anyone has some idea on how to solve this as it seems a widespread issue with the Adafruit SI1145 Library.

### Which version of ESPHome has the issue?

v2021.12.0-dev

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.5

### What platform are you using?

ESP8266

### Board

D1 Mini

### Component causing the issue

SI1145, I2C, Wire

### Example YAML snippet

```yaml
esphome:
  name: Weather
  platform: ESP8266
  board: d1_mini
  
  includes:
    - SI1145.h
  libraries:
    - "Adafruit SI1145 Library"
    
wifi:
  ssid: "Bla"
  password: "blabla"
  fast_connect: on

  manual_ip:
    static_ip: 192.168.0.x
    gateway: 192.168.0.x
    subnet: 255.255.255.0
    dns1: 192.168.0.x
    dns2: 8.8.8.8
    
# Enable logging
logger:

# Enable Home Assistant API
ota:
  password: 'blablabla'

mqtt:
  broker: 192.168.0.x
[... 271 lines cut ...]
  sensor::Sensor *ir_sensor = new sensor::Sensor();
  sensor::Sensor *uvindex_sensor = new sensor::Sensor();

  SI1145_sensor() : PollingComponent(1000) { }

float visible;
float irlight;
float uvindex;

  void setup() override {
    uv.reset();
    uv.begin();
  }

  void update() override {
    visible = uv.readVisible();
    visible_sensor->publish_state(visible);

    irlight = uv.readIR();
    ir_sensor->publish_state(irlight);

    uvindex = uv.readUV();
    uvindex_sensor->publish_state(uvindex / 100.0);
  }
};

Lefuneste83, 2021-11-21:
I have tried both
 - adafruit/Adafruit BusIO @ 1.9.6
 - mertmechanic/Adafruit BusIO @ 1.7.3
 Without any change is the error

raphis, 2021-11-21:
I assume that this might be an issue affecting multiple libraries - in my case it's missing SPI.h and EEPROM.h

`.piolibdeps/d1mini-somfy-wireless/SmartRC-CC1101-Driver-Lib/ELECHOUSE_CC1101_SRC_DRV.cpp:16:17: fatal error: SPI.h: No such file or directory`

`.piolibdeps/d1mini-somfy-wireless/Somfy_Remote_Lib/src/EEPROMRollingCodeStorage.h:4:20: fatal error: EEPROM.h: No such file or directory`

I see this starting in 2021.11.1 running in docker - if I go one version back compilation works again.

Lefuneste83, 2021-11-21:
This is interesting. So the regression would be linked to HA...

oxan, 2021-11-22:
We turned off the PlatformIO LDF in 1.11, so libraries need to be explicitly specified now.

> I have tried both
> 
> * adafruit/Adafruit BusIO @ 1.9.6

Did you add this one to the `libraries` key, before the SI1145 library? I should expect that to solve the problem.

> I assume that this might be an issue affecting multiple libraries - in my case it's missing SPI.h and EEPROM.h

Try adding `SPI` and `EEPROM` to the `libraries` key.

Lefuneste83, 2021-11-22:
Yes issue is solved with the following additions

 ```
 includes:
    - SI1145.h

  libraries:
    - SPI
    - Wire
    - adafruit/Adafruit BusIO @ 1.9.6
    - "Adafruit SI1145 Library"
```
  
Many thanks for the help !

Lefuneste83, 2021-11-23:
I believe this has to do with my custom library which does not interact properly anymore with the other ones

```
#include "esphome.h"
#include "Adafruit_SI1145.h"

using namespace esphome;

class SI1145_sensor : public PollingComponent {
 public:

  Adafruit_SI1145 uv = Adafruit_SI1145();

  sensor::Sensor *visible_sensor = new sensor::Sensor();
  sensor::Sensor *ir_sensor = new sensor::Sensor();
  sensor::Sensor *uvindex_sensor = new sensor::Sensor();

  SI1145_sensor() : PollingComponent(1000) { }

float visible;
float irlight;
float uvindex;

  void setup() override {
    uv.reset();
    uv.begin();
  }

  void update() override {
    visible = uv.readVisible();
    visible_sensor->publish_state(visible);

    irlight = uv.readIR();
    ir_sensor->publish_state(irlight);

    uvindex = uv.readUV();
    uvindex_sensor->publish_state(uvindex / 100.0);
  }
};
```

For reference the installed Adafruit_SI1145.h

```
/***************************************************
  This is a library for the Si1145 UV/IR/Visible Light Sensor

  Designed specifically to work with the Si1145 sensor in the
  adafruit shop
  ----> https://www.adafruit.com/products/1777

  These sensors use I2C to communicate, 2 pins are required to
  interface
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  BSD license, all text above must be included in any redistribution
 ****************************************************/
#ifndef _SI1145_H_
#define _SI1145_H_

#if (ARDUINO >= 100)
#include "Arduino.h"
#else
#include "WProgram.h"
#endif
#include <Adafruit_I2CDevice.h>

[... 121 lines cut ...]
 *
 */
class Adafruit_SI1145 {
public:
  Adafruit_SI1145(void);
  ~Adafruit_SI1145();
  boolean begin(uint8_t addr = SI1145_ADDR, TwoWire *pBus = &Wire);
  boolean begin(TwoWire *pBus);
  void reset();

  uint16_t readUV();
  uint16_t readIR();
  uint16_t readVisible();
  uint16_t readProx();

private:
  uint16_t read16(uint8_t addr);
  uint8_t read8(uint8_t addr);
  void write8(uint8_t reg, uint8_t val);
  uint8_t readParam(uint8_t p);
  uint8_t writeParam(uint8_t p, uint8_t v);
  Adafruit_I2CDevice *i2c_dev = NULL; ///< Pointer to I2C bus interface
};
#endif
```

Lefuneste83, 2021-11-23:
Ok I have rewritten the custom library carefully following the instructions from the [custom sensor components](https://esphome.io/components/sensor/custom.html) and it compiles and seems to work now. I have also rearranged the variable declaration and calls so that they match the order of the Adafruit Library (doubt this has any effect). I have removed the reset call that was necessary before and did some othe minor changes. Hope it's good now... Will see how it stands in the next couple of hours.

```
#include "esphome.h"
#include "Adafruit_SI1145.h"

using namespace esphome;

class SI1145_sensor : public PollingComponent {
 public:

  Adafruit_SI1145 uv;

  Sensor *uvindex_sensor = new Sensor();
  Sensor *ir_sensor = new Sensor();
  Sensor *visible_sensor = new Sensor();
  
  SI1145_sensor() : PollingComponent(2000) { }

  void setup() override {
    uv.begin();
  }

  void update() override {
    float uvindex = uv.readUV();
    uvindex_sensor->publish_state(uvindex / 100.0);

    float irlight = uv.readIR();
    ir_sensor->publish_state(irlight);
    
    float visible = uv.readVisible();
    visible_sensor->publish_state(visible);
  }
};
```

closed 2021-11-25 by Lefuneste83

raphis, 2021-11-25:
> libraries:
>    - SPI
>    - Wire

Indeed - the problem was the order of the libraries - after I put SPI + EEPROM to the top it's working with the latest esphome as well.  Thanks for the pointer.