esphome/issues #2744: ezo sensors reporting 0.00000 after 2021.11.1 update

rchamp26, 2021-11-21:
### The problem

after updating to the current version 2021.11.1, my ezo sensors arent reporting properly. i always receive a value of 0.00000 for both my rtd and ph sensors (i2c mode). 

I connected the same board, isolated carrier and probe to an arduino (also using i2c mode) and took readings from the serial monitor and verified that the sensor module and probe are still reading accurately themselves. 

not sure if something changed in the code with the last update.

i'm ordering another identical esp module so i can look to do another test via a new module and esphome, just for giggles.

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP32

### Board

hiletgo esp-wroom-32

### Component causing the issue

i2c

### Example YAML snippet

```yaml
esphome:
  platform: ESP32
  board: esp32dev

# Example configuration entry for ESP32
i2c:
  sda: 21
  scl: 22
  scan: true
  id: bus_a

sensor:  
  - platform: ezo
    id: ph_ezo
    name: "pH sensor"
    address: 99
    unit_of_measurement: "pH"
    accuracy_decimals: 2
    update_interval: 10s
    
  - platform: ezo
    id: rtd_ezo
    name: "RTD Temperature"
    address: 102
    accuracy_decimals: 2
    unit_of_measurement: "°C"
    device_class: temperature
    update_interval: 10s
```

### Anything in the logs that might be useful for us?

```txt
not really. this is what is reporting in the logs

[11:25:59][I][i2c.arduino:053]: Found i2c device at address 0x66
[11:26:00][C][ezo.sensor:015]: EZO 'RTD Temperature'
[11:26:00][C][ezo.sensor:015]:   Device Class: 'temperature'
[11:26:00][C][ezo.sensor:015]:   State Class: ''
[11:26:00][C][ezo.sensor:015]:   Unit of Measurement: '°C'
[11:26:00][C][ezo.sensor:015]:   Accuracy Decimals: 2
[11:26:00][C][ezo.sensor:016]:   Address: 0x66
[11:26:00][C][ezo.sensor:019]:   Update Interval: 10.0s
[11:26:03][D][sensor:113]: 'RTD Temperature': Sending state 0.00000 °C with 2 decimals of accuracy
```

rchamp26, 2021-11-21:
just tested with a brand new hiletgo esp-wroom-32 module and receive the same results

oxan, 2021-11-22:
Duplicate of #2716.

closed 2021-11-22 by oxan

rchamp26, 2021-11-22:
Rolled back to 2021.10.3 and sensors are reporting as expected