esphome/issues #2745: Wyze Outdoor Outlet Disconnecting From WiFi

556duckvader, 2021-11-22:
### The problem

I have 3 Wyze Outdoor Outlets and one of them is giving me issues where it will connected to the WiFi and work for a while then it will disconnect completely and none of the buttons on it work to turn the outlets on and off.  The only way to get it back working again is to unplug it and plug it back in and then it works again for a while then stops again.  I have tried the same config on another one of my Wyze outlets and it does the same thing.  I don't know if its the config causing the issue or what because I have another one running the exact same config with only the name changing and its rock solid.

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.1

### What platform are you using?

ESP32

### Board

Wyze Outdoor Outlet

### Component causing the issue

wifi

### Example YAML snippet

```yaml
substitutions:
  display_name: WyzeOutdoorPlug1
  # Higher value gives lower watt readout
  current_res: "0.001"
  # Lower value gives lower voltage readout
  voltage_div: "770"  
  
esphome:
  name: wyze-outdoor-plug1
  platform: ESP32
  board: esp-wrover-kit

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  
  ap: 
    ssid: ${display_name}_AP
    password: !secret wifi_password
    ap_timeout: 3min
    
captive_portal:

logger:

#MQTT
mqtt:
  broker: !secret mqtt_broker
  username: !secret mqtt_username
  password: !secret mqtt_password

#OTA
ota:

# Enable Home Assistant API
api:
  password: "XXXXXXXXX"

web_server:
[... 198 lines cut ...]
[22:19:13][C][mqtt:074]:   Log Topic: 'wyze-outdoor-plug1/debug'
[22:19:13][C][mqtt:077]:   Availability: 'wyze-outdoor-plug1/status'
[22:19:13][C][api:134]: API Server:
[22:19:13][C][api:135]:   Address: wyze-outdoor-plug1.local:6053
[22:19:13][C][api:139]:   Using noise encryption: NO
[22:19:13][C][mdns:084]: mDNS:
[22:19:13][C][mdns:085]:   Hostname: wyze-outdoor-plug1
[22:19:13][C][mqtt.switch:041]: MQTT Switch 'WyzeOutdoorPlug1 Restart': 
[22:19:13][C][mqtt.switch:042]:   State Topic: 'wyze-outdoor-plug1/switch/wyzeoutdoorplug1_restart/state'
[22:19:13][C][mqtt.switch:042]:   Command Topic: 'wyze-outdoor-plug1/switch/wyzeoutdoorplug1_restart/command'
[22:19:13][C][mqtt.sensor:027]: MQTT Sensor 'WyzeOutdoorPlug1 LUX':
[22:19:13][C][mqtt.sensor:031]:   State Topic: 'wyze-outdoor-plug1/sensor/wyzeoutdoorplug1_lux/state'
[22:19:13][C][mqtt.sensor:027]: MQTT Sensor 'WyzeOutdoorPlug1 Volts':
[22:19:13][C][mqtt.sensor:031]:   State Topic: 'wyze-outdoor-plug1/sensor/wyzeoutdoorplug1_volts/state'
[22:19:13][C][mqtt.sensor:027]: MQTT Sensor 'WyzeOutdoorPlug1 Amps':
[22:19:13][C][mqtt.sensor:031]:   State Topic: 'wyze-outdoor-plug1/sensor/wyzeoutdoorplug1_amps/state'
[22:19:13][C][mqtt.sensor:027]: MQTT Sensor 'WyzeOutdoorPlug1 Watts':
[22:19:13][C][mqtt.sensor:031]:   State Topic: 'wyze-outdoor-plug1/sensor/wyzeoutdoorplug1_watts/state'
[22:19:13][C][mqtt.switch:041]: MQTT Switch 'WyzeOutdoorPlug1 Relay1': 
[22:19:13][C][mqtt.switch:042]:   State Topic: 'wyze-outdoor-plug1/switch/wyzeoutdoorplug1_relay1/state'
[22:19:13][C][mqtt.switch:042]:   Command Topic: 'wyze-outdoor-plug1/switch/wyzeoutdoorplug1_relay1/command'
[22:19:14][C][mqtt.switch:041]: MQTT Switch 'WyzeOutdoorPlug1 Relay2': 
[22:19:14][C][mqtt.switch:042]:   State Topic: 'wyze-outdoor-plug1/switch/wyzeoutdoorplug1_relay2/state'
[22:19:14][C][mqtt.switch:042]:   Command Topic: 'wyze-outdoor-plug1/switch/wyzeoutdoorplug1_relay2/command'
```

vmsguy, 2021-11-23:
I've experienced the same issue with a newly flashed plug using the instructions and yaml from the digiblur.com site.  
Unplugging and plugging back in and it connects right up to HA.  

I am going to pick up another plug at Home Depot and flash it too and see if is more stable.

556duckvader, 2021-11-23:
I have since reflashed the original outlet and removed everything associated with that outlet from HA and HA rediscovered it and it’s been stable since.  I have left if plugged in for over 3 days now and it’s still connected to the Wi-Fi.

vmsguy, 2021-11-24:
I reflashed mine as well after I updated ESPHOME, and so far it has been running fine for the past couple of days.

closed 2022-04-08 by github-actions[bot]