esphome/issues #2746: GPS component and OTA update

hippieua, 2021-11-22:
### The problem

Using GPS component causes  ESP reboot. Sometimes during work. Always during OTA Update. If TX pin is unplugged, everything works ok ( except GPS ). Tried on differect boards ( d1 mini, nodemcu, nodemcu v3)

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.4

### What platform are you using?

ESP8266

### Board

modemcu, d1_mini

### Component causing the issue

GPS

### Example YAML snippet

```yaml
esphome:
  name: gps
  platform: ESP8266
  board: d1_mini

logger:
api:
ota:
wifi:
  ssid: "secret"
  password: "secret"

uart:
  rx_pin: D7
  baud_rate: 9600

gps:
  latitude:
    name: "Latitude"
  longitude:
    name: "Longitude"
  altitude:
    name: "Altitude"
  speed:
    name: "Speed"
  course:
    name: "Course"
  satellites:
    name: "Satellites"
  update_interval: 10s

time:
  - platform: gps
```

### Anything in the logs that might be useful for us?

```txt
[17:21:39][D][ota:136]: Starting OTA Update from 192.168.5.9...
[... 123 lines cut ...]
[17:21:40]3fffff60:  00003194 3fff0144 3fff0600 3fff0644  
[17:21:40]3fffff70:  3fff1968 3fff04ac 3fff03ec feefeffe  
[17:21:40]3fffff80:  00000000 00000000 00000001 3ffeed64  
[17:21:40]3fffff90:  3fffdad0 00000000 3ffeed24 4020e190  
[17:21:40]3fffffa0:  3fffdad0 00000000 3ffeed24 402164a0  
[17:21:40]3fffffb0:  feefeffe feefeffe 3ffe8530 4010084d  
[17:21:40]<<<stack<<<
[17:21:40]
[17:21:40]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[17:21:40]
[17:21:40] ets Jan  8 2013,rst cause:4, boot mode:(3,6)
[17:21:40]
[17:21:40]wdt reset
[17:21:40]load 0x4010f000, len 3584, room 16 
[17:21:40]tail 0
[17:21:40]chksum 0xb0
[17:21:40]csum 0xb0
[17:21:40]v2843a5ac
[17:21:40]~ld
[17:21:40][I][logger:214]: Log initialized
[17:21:40][C][ota:452]: There have been 7 suspected unsuccessful boot attempts.
[17:21:40][I][app:029]: Running through setup()...
[17:21:40][C][uart.arduino_esp8266:052]: Setting up UART bus...
[17:21:40][C][wifi:037]: Setting up WiFi...
```

oxan, 2021-11-23:
Could you please obtain the logs through ESPHome itself? That would decode the exception, which is necessary to fix it.

hippieua, 2021-11-23:
> Could you please obtain the logs through ESPHome itself? That would decode the exception, which is necessary to fix it.

Is there any guide how to do it ?
On ESPHome addon it just starting update, and then after 10-20% just loosing connection

oxan, 2021-11-23:
Connect the device to the serial port, open the logs through serial in ESPHome, and then run the OTA (from another tab).

hippieua, 2021-11-23:
> Could you please obtain the logs through ESPHome itself? That would decode the exception, which is necessary to fix it.

`
INFO Reading configuration /config/esphome/gps.yaml...
INFO Detected timezone 'Europe/Kiev'
INFO Generating C++ source...
INFO Compiling app...
Processing gps (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Dependency Graph
|-- <ESP8266WiFi> 1.0
|-- <TinyGPSPlus> 1.0.2
|-- <ESP8266mDNS> 1.2
Compiling /data/gps/.pioenvs/gps/src/main.cpp.o
Linking /data/gps/.pioenvs/gps/firmware.elf
Building /data/gps/.pioenvs/gps/firmware.bin
RAM:   [====      ]  36.2% (used 29624 bytes from 81920 bytes)
Flash: [====      ]  35.4% (used 369784 bytes from 1044464 bytes)
========================= [SUCCESS] Took 4.25 seconds =========================
INFO Successfully compiled program.
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 84:0d:8e:8a:8e:9a
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00000000 to 0x0005bfff...
Compressed 373936 bytes to 263703...
Wrote 373936 bytes (263703 compressed) at 0x00000000 in 6.1 seconds (effective 493.0 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
INFO Successfully uploaded program.
INFO Starting log output from /dev/ttyUSB0 with baud rate 115200
[11:35:57]sd$ܟ<\x84l\xe0<\x84l\xe4#|\x87\x83\xec\033\xdbs\x9bc\x84#\x8c\xf3og\x9fl'o\xdc\xe3\xec#x\x8c\x8fl;l{lx\xf3n\xe0\xc3$\x84\x9cb'\xe3|\xe4l\x8f\x8ec\x84\xfbgn\xeed\x8c\x8e$`\x90\033ogl o;Ǜ\x92'cd`;\x93\x9bgcl`\x9cc\x8f\x9bls\xdb`\xfc\x83o\xdc[I][logger:214]: Log initialized
[11:35:57][C][ota:452]: There have been 4 suspected unsuccessful boot attempts.
[11:35:57][I][app:029]: Running through setup()...
[11:35:57][C][uart.arduino_esp8266:052]: Setting up UART bus...
[11:35:57][C][wifi:037]: Setting up WiFi...
[11:35:57][I][wifi:245]: WiFi Connecting to 'G-HOME'...
[11:35:57][D][sensor:113]: 'Latitude': Sending state -1.00000 ° with 6 decimals of accuracy
[11:35:57][D][sensor:113]: 'Longitude': Sending state -1.00000 ° with 6 decimals of accuracy
[11:35:57][D][sensor:113]: 'Speed': Sending state -1.00000 km/h with 6 decimals of accuracy
[11:35:57][D][sensor:113]: 'Course': Sending state -1.00000 ° with 2 decimals of accuracy
[11:35:57][D][sensor:113]: 'Altitude': Sending state -1.00000 m with 1 decimals of accuracy
[11:35:57][D][sensor:113]: 'Satellites': Sending state -1.00000  with 0 decimals of accuracy
[11:35:57][D][gps:062]: Satellites:
[11:35:57][D][gps:063]:   0
[11:35:58][D][gps:062]: Satellites:
[11:35:58][D][gps:063]:   0
[11:35:59][D][gps:062]: Satellites:
[11:35:59][D][gps:063]:   0
[11:36:01][I][wifi:502]: WiFi Connected!
[11:36:01][C][wifi:350]:   Local MAC: 84:0D:8E:8A:8E:9A
[11:36:01][C][wifi:351]:   SSID: [redacted]
[11:36:01][C][wifi:352]:   IP Address: 192.168.5.215
[11:36:01][C][wifi:354]:   BSSID: [redacted]
[11:36:01][C][wifi:355]:   Hostname: 'gps'
[11:36:01][C][wifi:357]:   Signal strength: -59 dB ▂▄▆█
[11:36:01][C][wifi:361]:   Channel: 8
[... 279 lines cut ...]
[11:36:15][C][wifi:364]:   DNS1: 192.168.5.1
[11:36:15][C][wifi:365]:   DNS2: 0.0.0.0
[11:36:15][C][logger:233]: Logger:
[11:36:15][C][logger:234]:   Level: DEBUG
[11:36:15][C][logger:235]:   Log Baud Rate: 115200
[11:36:15][C][logger:236]:   Hardware UART: UART0
[11:36:15][C][uart.arduino_esp8266:095]: UART Bus:
[11:36:15][C][uart.arduino_esp8266:097]:   RX Pin: GPIO5
[11:36:15][C][uart.arduino_esp8266:099]:   RX Buffer Size: 256
[11:36:15][C][uart.arduino_esp8266:101]:   Baud Rate: 9600 baud
[11:36:15][C][uart.arduino_esp8266:102]:   Data Bits: 8
[11:36:15][C][uart.arduino_esp8266:103]:   Parity: NONE
[11:36:15][C][uart.arduino_esp8266:104]:   Stop bits: 1
[11:36:15][C][uart.arduino_esp8266:108]:   Using software serial
[11:36:15][C][ota:082]: Over-The-Air Updates:
[11:36:15][C][ota:083]:   Address: gps.local:8266
[11:36:15][W][ota:092]: Last Boot was an unhandled reset, will proceed to safe mode in 5 restarts
[11:36:15][C][api:134]: API Server:
[11:36:15][C][api:135]:   Address: gps.local:6053
[11:36:15][C][api:139]:   Using noise encryption: NO
[11:36:15][C][mdns:084]: mDNS:
[11:36:15][C][mdns:085]:   Hostname: gps
[11:36:15][D][gps:062]: Satellites:
[11:36:15][D][gps:063]:   0
`

rradar, 2021-11-30:
As a "quick fix" you might want to consider enter the "safe mode" using the [safe mode switch](https://esphome.io/components/switch/safe_mode.html) before starting the ota update

closed 2021-12-01 by oxan