esphome/issues #2747: Entity category: CONFIG seems not processed for switch

jgverweij, 2021-11-22:
### The problem

My smartplug exposed 2 switches for On/off and Restart.
To avoid accidentially restarting with all switches in one room I try to use the new entity_category in 2021.11.

However, after adding entity_category to the code the Restart switch does not move to the Config card as expected.
`switch:
  # Switch to restart the plug
  - platform: restart
    name: "${plug_name} Restart"
    entity_category: config`

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

esp8285

### Additional information

[image]

andrew-codechimp, 2021-11-28:
Been struggling with this myself.  If you are adding the entity_category to an existing integration you must reload the integration for it to change.
In HA, Integrations, click on the ESP device you need to update, then using the burger menu reload that integration.

[image]

jgverweij, 2021-11-28:
Confirmed -> solved

closed 2021-11-28 by jgverweij