esphome/issues #2748: espidf ledc fails to compile for ESP32C3 board

rsumner, 2021-11-22:
### The problem

Trying to use LEDC to control a H-Bridge for PWM fan control:

```
output:
  - platform: ledc
    pin: 4
    id: fan1
    frequency: "100Hz"
```

It fails saying that LEDC_HIGH_SPEED_MODE isn't declared.  I do know that the ESP32C3 only has LEDC_LOW_SPEED_MODE, so maybe that's the problem? 

```
src/esphome/components/ledc/ledc_output.cpp: In member function 'virtual void esphome::ledc::LEDCOutput::write_state(float)':
src/esphome/components/ledc/ledc_output.cpp:51:36: error: 'LEDC_HIGH_SPEED_MODE' was not declared in this scope
   auto speed_mode = channel_ < 8 ? LEDC_HIGH_SPEED_MODE : LEDC_LOW_SPEED_MODE;
                                    ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/ledc/ledc_output.cpp:51:36: note: suggested alternative: 'LEDC_LOW_SPEED_MODE'
   auto speed_mode = channel_ < 8 ? LEDC_HIGH_SPEED_MODE : LEDC_LOW_SPEED_MODE;
                                    ^~~~~~~~~~~~~~~~~~~~
                                    LEDC_LOW_SPEED_MODE
src/esphome/components/ledc/ledc_output.cpp: In member function 'virtual void esphome::ledc::LEDCOutput::setup()':
src/esphome/components/ledc/ledc_output.cpp:66:36: error: 'LEDC_HIGH_SPEED_MODE' was not declared in this scope
   auto speed_mode = channel_ < 8 ? LEDC_HIGH_SPEED_MODE : LEDC_LOW_SPEED_MODE;
                                    ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/ledc/ledc_output.cpp:66:36: note: suggested alternative: 'LEDC_LOW_SPEED_MODE'
   auto speed_mode = channel_ < 8 ? LEDC_HIGH_SPEED_MODE : LEDC_LOW_SPEED_MODE;
                                    ^~~~~~~~~~~~~~~~~~~~
                                    LEDC_LOW_SPEED_MODE
src/esphome/components/ledc/ledc_output.cpp: In member function 'virtual void esphome::ledc::LEDCOutput::update_frequency(float)':
src/esphome/components/ledc/ledc_output.cpp:117:36: error: 'LEDC_HIGH_SPEED_MODE' was not declared in this scope
   auto speed_mode = channel_ < 8 ? LEDC_HIGH_SPEED_MODE : LEDC_LOW_SPEED_MODE;
                                    ^~~~~~~~~~~~~~~~~~~~
src/esphome/components/ledc/ledc_output.cpp:117:36: note: suggested alternative: 'LEDC_LOW_SPEED_MODE'
   auto speed_mode = channel_ < 8 ? LEDC_HIGH_SPEED_MODE : LEDC_LOW_SPEED_MODE;
                                    ^~~~~~~~~~~~~~~~~~~~
                                    LEDC_LOW_SPEED_MODE
*** [.pioenvs/datacloset-temp/src/esphome/components/ledc/ledc_output.o] Error 1
```

### Which version of ESPHome has the issue?

2021.12.0-dev20211122

### What type of installation are you using?

Docker

### What platform are you using?

ESP32-IDF

### Board

ESP32-C3

### Component causing the issue

output/ledc

rsumner, 2021-11-23:
looking at the source code for framework-espidf/components/hal/include/hal/ledc_types.h, you can see:

```
 25 typedef enum {
 26 #if SOC_LEDC_SUPPORT_HS_MODE
 27     LEDC_HIGH_SPEED_MODE = 0, /*!< LEDC high speed speed_mode */
 28 #endif
 29     LEDC_LOW_SPEED_MODE,      /*!< LEDC low speed speed_mode */
 30     LEDC_SPEED_MODE_MAX,      /*!< LEDC speed limit */
 31 } ledc_mode_t;
```

In the case of the ESP32-C3, since it doesn't have HIGH speed mode it doesn't make it into the enum.  esphome should probably have a similar comparator on SOC_LEDC_SUPPORT_HS_MODE.

rsumner, 2021-11-23:
Something like this:

```
#ifdef USE_ESP_IDF
#if SOC_LEDC_SUPPORT_HS_MODE
  auto speed_mode = channel_ < 8 ? LEDC_HIGH_SPEED_MODE : LEDC_LOW_SPEED_MODE;
#else
  auto speed_mode = LEDC_LOW_SPEED_MODE;
#endif
  auto chan_num = static_cast<ledc_channel_t>(channel_ % 8);
  ledc_set_duty(speed_mode, chan_num, duty);
  ledc_update_duty(speed_mode, chan_num);
#endif
```

I wrapped the three occurrences of the use of LEDC_HIGH_SPEED_MODE in this conditional and the code does compile now.  This is my first time using this output, so I still have to see if it actually works 😄 

Once I confirm, I'll submit a pull request.  Disclosure, I haven't tested this on a ESP32 or ESP32-S2/3.

martgras, 2021-11-23:
You were faster :) 
Just pushed https://github.com/martgras/esphome/blob/led-c3/esphome/components/ledc/ledc_output.cpp#L19 to my repository :)
Not yet tested either but compiles

rsumner, 2021-11-23:
Hrmm...I tried my code and yours @martgras (since yours was noticeably different than that from the master branch) and neither produce any voltage on the PWM pin when I increase the speed of a fan component:

```
output:
  - platform: ledc
    pin: GPIO5
    id: fan1
    channel: 0
    frequency: "100Hz"

fan:
  - platform: speed
    output: fan1
    name: "Data Closet Fan 1"
```

Unfortunately, I don't have an oscilloscope to checkout the wavelength on the output pin.

rsumner, 2021-11-23:
I tested using the [ledc_basic example ](https://github.com/espressif/esp-idf/tree/a18cb70/examples/peripherals/ledc/ledc_basic) in espidf and I get 3.3v out the GPIO.  I modified it a little to iterate through various duty cycles and the speed of the fan does change.  So, something is wacky on the esphome side.

rsumner, 2021-11-23:
Using higher frequencies allow things to start to work.  I noticed the voltage on the GPIO when I'm below 100% falls well below 3V and I think my H-Bridge has a mimimum of 3V on the enable pin.    So, in summary, I think @martgras 's fix works.  I vote for moving it to a pull request and merging into the next dev.

closed 2021-11-24 by jesserockz