esphome/issues #2749: Status LED Light, still not playing well with other light on the same pin.

Lovely-Maisonette, 2021-11-23:
### The problem

I stopped using the status_led high level component when I realized it did not work simultaneously with a light under the same pin. I started using it again when it became a platform under light, and thus, the same pin could be shared. But it is not really working.

My case:

A powerplug with a template switch to enable/disable the power LED light when the relay is active (at night the light is too strong). This works well, and it restores its states (LED state and Template Switch state) even after reboot or power loss.

But if I enable the light led_status platform AND there is a reboot or power loss (it would probably be the same with WIFI loss), after the blinking, the light stays always off regardless of the relay or template switch status (I presume the blinking finished with off, and it stays there). The template switch does show the right state, so this is clearly just the light platforms not playing nicely with each-other, as it was the case with the original status_led component.

### Which version of ESPHome has the issue?

ESPHome version 2021.11.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

stand alone problem

### What platform are you using?

ESP8266

### Board

https://templates.blakadder.com/bakibo_TP22Y.html

### Component causing the issue

status_led light platform

### Example YAML snippet

```yaml
light:

  # Commenting this out, all behaviors work (except the status blinking of course)
  - platform: status_led
    id: Status_led
    pin:
      number: ${LED1R}
      inverted: ${LED1Ri}

  # LED1
  - platform: rgb
    id: LED1
    icon: "mdi:led-on"
    blue: pwm1b
    red: pwm1r
    green: pwm1g
    default_transition_length: 1s
    effects:
      - pulse:
      - pulse:
          name: "Fast Pulse"
          transition_length: 0.5s
          update_interval: 0.4s
      - pulse:
          name: "Slow Pulse"
          transition_length: 1s     # default 1s
          update_interval: 0.4s

output:

  # LED1 B/R/G
  - platform: esp8266_pwm
    id: pwm1b
    pin:
      number: ${LED1B}
      inverted: ${LED1Bi}
  - platform: esp8266_pwm
    id: pwm1r
    pin:
      number: ${LED1R}
      inverted: ${LED1Ri}
  - platform: esp8266_pwm
    id: pwm1g
    pin:
      number: ${LED1G}
      inverted: ${LED1Gi}

  # Power-LED Enable/Disable
  - platform: template
    name: ${nice_name} Status LED (Disable/Enable)
    id: LED1_enabled
    restore_state: true
    entity_category: config
    icon: "mdi:led-on"
    # Publish the switch state, but turn on the LED only if the relay is on
    turn_on_action:
      - switch.template.publish:
          id: LED1_enabled
          state: ON
      - if:
          condition:
            switch.is_on: Relay1
          then:
            - light.turn_on: LED1
    turn_off_action:
      - switch.template.publish:
          id: LED1_enabled
          state: OFF
      - light.turn_off: LED1

esp8266:
  restore_from_flash: true
preferences:
  flash_write_interval: 1min
```

### Additional information

Commenting this block out, fixes the interference problem, but it removes the status feature
```
  - platform: status_led
    id: Status_led
    pin:
      number: ${LED1R}
      inverted: ${LED1Ri}
```

oxan, 2021-11-23:
We don't support using the same pin for two components, as you are doing with `${LED1R}` on `LED1` and `Status_led`.

I think what you really want is a status_led platform that's non-monochromatic.

Lovely-Maisonette, 2021-11-23:
> We don't support using the same pin for two components, as you are doing with `${LED1R}` on `LED1` and `Status_led`.
> 
> I think what you really want is a status_led platform that's non-monochromatic, which unfortunately doesn't exist at the moment.

If I understand correctly, that was right with the old status_led COMPONENT (https://esphome.io/components/status_led.html),
but it is incorrect for the NEW status_led LIGHT (https://esphome.io/components/light/status_led.html?highlight=status_led)

From the documentation:

```
The status_led light platform allows to share a single LED for indicating the status of the device (when on error/warning state) or as binary light (when on OK state). This is useful for devices with only one LED available.

It provides the combined functionality of status_led component and a binary light component **over a single shared GPIO led**.
```

oxan, 2021-11-23:
No, it's still true for the new status_led light. The status_led light functions as a binary light (so you can control it from automations and HA), but also uses that light to show the status information. You can't use a pin that's used as a status_led light for another light as well.

Lovely-Maisonette, 2021-11-24:
I am confused: "A single shared GPIO LED" = You can used one led as a light for whatever, and as a light for status. Same LED, same pin. The status light should stop the light, show status, and return to the original light after no status is required. That is the problem, it is not returning, it stops blinking and stays in status off.

oxan, 2021-11-24:
The documentation isn't very clear, but the point with the status_led LED light is that _one_ light platform has both functions, not that you can use two light platforms on one pin.

Lovely-Maisonette, 2021-11-26:
Yes, now it is clear, the docu was not. Then this bug report applies to the docu, and not to the functionality :)

closed 2022-04-13 by github-actions[bot]