esphome/issues #2750: ESP32-IDF not work

liaukouski, 2021-11-23:
### The problem

esp-c3-32s and board esp-c3-01m not firmware using esphome and usb-com port

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP32-IDF

### Board

esp-c3-32s kit, esp-c3-01m

### Example YAML snippet

```yaml
esphome:
  name: esp-c3-32s
esp32:
  board: esp32-c3-devkitm-1
  framework:
    version: recommended
    type: esp-idf
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
```

### Anything in the logs that might be useful for us?

```txt
ERROR message
Installation failed: Error: Configuration does not match the platform of the connected device. Expected a ESP32 device.
```

rsumner, 2021-11-24:
@liaukouski, add variant like this:

```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
  # currently needed, but not in the future
  variant: ESP32C3
```

liaukouski, 2021-11-24:
not resolve problem, same error:
Installation failed: Error: Configuration does not match the platform of the connected device. Expected a ESP32 device.

iMike78, 2021-11-26:
Hi lads.
Any solution for this? I have the same problem, and the "variant: ESP32C3" won't help either. Got the same error message.
[image]

rsumner, 2021-11-26:
The C3/S2/S3 code is still in development, so make sure you're using the dev release and not a current production release for C3 chip support.

iMike78, 2021-11-26:
2021.12.0 dev version installed 
In the code I'm using the dev version as well.
[image]
The result is even more error during the compile
[image]
I was happy when I found the new esp32 version, but I think I'll leave them in the drawer for a while 😃

rsumner, 2021-11-26:
@iMike78.  just make sure you are using the latest dev release of ESPhome and don't change the espidf version in your config.yaml.  this works perfectly fine for me:

```
esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
  # currently needed, but not in the future
  variant: ESP32C3
```

iMike78, 2021-11-26:
I'm using the latest dev release of ESPhome
Unfortunately still I'm struggling with this error message:
Installation failed: Error: Configuration does not match the platform of the connected device. Expected a ESP32 device.
My code now is exactly what you posted.
[image]

rsumner, 2021-11-26:
What kind of board are you using?  Is there more output in your "run" command above the error?  What is it detecting your device as?  All of my testing has been using an official Espressif ESP32-C3-DevKitC-02 board and everything seems to work fine for me.  Here's a sample of my output where the ESP32-C3 chip is properly detected:

```
esptool.py v3.2
Serial port /dev/ttyUSB0
Connecting....
Chip is ESP32-C3 (revision 3)
Features: Wi-Fi
Crystal is 40MHz
MAC: 
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
```

iMike78, 2021-11-26:
User error, sorry. I tried to flash trough browser (chrome), not to connect to the ESPhome host computer. 
Now I plugged the board to the rPi, and the flash was complete.
Thanks for your help.

jddonovan, 2021-12-03:
> User error, sorry. I tried to flash trough browser (chrome), not to connect to the ESPhome host computer. Now I plugged the board to the rPi, and the flash was complete. Thanks for your help.

Weirdly, this is true. I have the same experience with a C3 board, NodeMCU ESP-C3-13. Will not work over Chrome, but will start flashing if connected physically.

southseaboy, 2021-12-14:
Same issue for me with NodeMCU esp32-c3 boards (4 different variants, all using CH430 usb to serial adaptors).  Interestingly Tasmota's chrome upload tool (also using ESP Web Tools) works fine with these boards, so would seem to be an ESPHome issue. I'm using esphome 2021.12.0-dev Home Assistant Addon on a linux supervised platform, the browser is Chrome on a Windows 10 PC.

metrafonic, 2021-12-23:
I get this issue:
```
*** [.pioenvs/esp32c3/.pioenvs/esp32c3/x509_crt_bundle.o] Source `.pioenvs/esp32c3/x509_crt_bundle.S' not found, needed by target `.pioenvs/esp32c3/.pioenvs/esp32c3/x509_crt_bundle.o'.
```
Using:
```
esphome:
  name: esp32c3

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
  # currently needed, but not in the future
  variant: ESP32C3
```

gmarcc, 2022-05-01:
I have been experiencing a similar issue.

nmajin, 2022-06-24:
I am trying to use my Adafruit ESP32-S3 chip and I can get it to compile and even flash (holding the non-restart button while plugging into USB) and then hitting install/flash. However, after the flash nothing happens...LEDs don't even turn on anymore lol. Here is my config I used with ESPHome Web:
```
esphome:
  name: esphome

esp32:
  board: esp32dev
  variant: esp32s3
  framework:
    type: esp-idf
```

nliaudat, 2022-11-14:
With esphome 2022.10.0, the following config works

```
Esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    version: recommended
    # Custom sdkconfig options
    sdkconfig_options:
      CONFIG_COMPILER_OPTIMIZATION_SIZE: y
    # Advanced tweaking options
    advanced:
      ignore_efuse_mac_crc: false
```

n3on4s, 2022-11-22:
> On what platform are you compiling, I have noticed that it will not work on rasbery mashine. Compile just not work.