esphome/issues #2752: Docker image does not allow running rootless

petrm, 2021-11-24:
### The problem

The official ESPHome docker image does not allow setting user/group to run the application with. While it is possible to pass to docker username and group, some of the functionality is then broken.
The web interface functions, it is possible to edit and save files, but impossible to compile. The compilation fails, because platformio is using for cache /root, which is owned by root and therefore not writable by the user passed to docker.
An simple fix would be to use a different folder than /root for the temporary files, for example by setting $HOME to this folder in the docker image or as an ugly workaround chmod /root to RW for all.

### Which version of ESPHome has the issue?

2021.11.0

### What type of installation are you using?

Docker

### What platform are you using?

ESP32

### Board

N/A

### Component causing the issue

Docker image

sgofferj, 2022-03-06:
I ran into the same issue. It's a potential security issue but also an inconvenience because the yaml files can't be edited by anyone but root. I am using the docker container with a bind mount and would like to make this bind mount available via NFS, so I can edit the files from any computer in my network with a nicer editor.

ilarrain, 2022-04-17:
Seems that I've managed to implement a quick fix to make it work:
```
    environment:
      - 'HOME=/config'
      - 'PLATFORMIO_GLOBALLIB_DIR=/config/.piolibs'
```
- Platformio tries to write to ${HOME}/.platformio, but $HOME is undefined for the specified user, so it tries to write to /.platformio
- PLATFORMIO_GLOBALLIB_DIR is hardcoded to /piolibs in Dockerfile, which should not be a problem, but libraries on this directory are owned by root user, with 700 permissions, unreadable for defined user. https://github.com/esphome/esphome/blob/93b628d9a856461e570b9e195f8dbd4c04346cde/docker/Dockerfile#L42

The drawback is that platformio will install required dependencies on compile time.

ptr727, 2024-06-15:
FYI: https://github.com/ptr727/ESPHome-NonRoot

janstadt, 2025-04-14:
Any chance this gets updated to support non root users for the esphome container? No offence to @ptr727 but i'd love to get this functionality from the mainline repo.

ptr727, 2025-04-15:
> Any chance this gets updated to support non root users for the esphome container? No offence to [@ptr727](https://github.com/ptr727) but i'd love to get this functionality from the mainline repo.

Lol, none taken, I completely agree non-root should be supported out of the box.