esphome/issues #2754: ESPHome doesn't come back from App.reboot()

fabiopedrosa, 2021-11-25:
### The problem

I'm doing this in ESPHome (on HA):

```yaml
esphome:
  ...
  includes:
    - custom_restart.h

custom_component:
- lambda: |-
    auto my_custom = new MyCustomRestartComponent();
    return {my_custom};

wifi:
  ssid: ...
  manual_ip:
    static_ip: 192.168.1.18
    gateway: 192.168.1.1
    subnet: 255.255.255.0
    dns1: 8.8.8.8
    dns2: 192.168.1.1
 
api:

ota:

```

and 

```cpp
#include "esphome.h"

class MyCustomRestartComponent : public Component {
	bool fired = false;

 public:
  void setup() override {
    ESP_LOGD("custom", "The GPIO pin 5 is HIGH!");
  }
  void loop() override {
    unsigned long currentSecond = millis() / 1000;

    if (currentSecond % (300) == 0) { // restart every 5min
    	if (!this->fired) {
            delay(500);
            App.reboot();
    	}
    	this->fired = true;
    } else {
    	this->fired = false;
    }
  }
};

```

but the device is not reachable after the restart.

Any idea why that happens?

### Which version of ESPHome has the issue?

v2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### What platform are you using?

ESP32

closed 2021-11-25 by fabiopedrosa

mmakaay, 2021-11-25:
The problem was solved?

fabiopedrosa, 2022-01-05:
@mmakaay yes, was my lousy code

> `if (currentSecond % (300) == 0) { // restart every 5min`

if number is 0, its true, so it keeps restarting on `timestamp=0` 🤦

mmakaay, 2022-01-05:
Ah, nice one. Thanks for reporting back.