esphome/issues #2756: Display max7219digit - Scroll problem

ninjax95, 2021-11-25:
### The problem

Following the upgrade of the esphome module for home assistant, my program does not work, whereas before I had no problems. Here is the error message when I want to upload the program to the chip.

Thank for you help

### Which version of ESPHome has the issue?

2021.11.1

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.5

### What platform are you using?

ESP8266

### Example YAML snippet

```yaml
display:
  - platform: max7219digit
    cs_pin: D8
    num_chips: 8
    intensity: 3
    update_interval: 100ms
    lambda: |-
          it.invert_on_off(false);
          it.printf(0, 0, id(digit_font), TextAlign::TOP_LEFT, "%s", id(hatxt).state.c_str());
          it.intensity(atoi(id(habri).state.c_str()));
          it.scroll(true, 0, 100, 2000, 2000);
```

### Anything in the logs that might be useful for us?

```txt
Error Message : 

Compiling /data/esp2_display/.pioenvs/esp2_display/src/main.cpp.o
/config/esphome/esp2-display.yaml: In lambda function:
/config/esphome/esp2-display.yaml:43:41: error: invalid conversion from 'int' to 'esphome::max7219digit::ScrollMode' [-fpermissive]
           it.scroll(true, 0, 100, 2000, 2000);
                                         ^
In file included from src/esphome.h:29:0,
                 from src/main.cpp:3:
src/esphome/components/max7219digit/max7219digit.h:73:8: error:   initializing argument 2 of 'void esphome::max7219digit::MAX7219Component::scroll(bool, esphome::max7219digit::ScrollMode, uint16_t, uint16_t, uint16_t)' [-fpermissive]
   void scroll(bool on_off, ScrollMode mode, uint16_t speed, uint16_t delay, uint16_t dwell);
        ^
*** [/data/esp2_display/.pioenvs/esp2_display/src/main.cpp.o] Error 1
========================== [FAILED] Took 3.04 seconds ==========================
```

oxan, 2021-11-25:
Your code should use `ScrollMode::CONTINUOUS` instead of 0:
```
          it.scroll(true, ScrollMode::CONTINUOUS, 100, 2000, 2000);
```

ninjax95, 2021-11-25:
> it.scroll(true, ScrollMode::CONTINUOUS, 100, 2000, 2000);

Thank for your reply, but don't work, I have this error message : 
```
Compiling /data/esp2_display/.pioenvs/esp2_display/src/main.cpp.o
/config/esphome/esp2-display.yaml: In lambda function:
/config/esphome/esp2-display.yaml:43:23: error: 'ScrollMode' has not been declared
           it.scroll(true, ScrollMode::CONTINUOUS, 100, 2000, 2000);
                       ^
*** [/data/esp2_display/.pioenvs/esp2_display/src/main.cpp.o] Error 1
========================== [FAILED] Took 2.94 seconds ==========================
```

oxan, 2021-11-25:
Oh, it should be `max7219digit::ScrollMode::CONTINUOUS`. Sorry about that.

ninjax95, 2021-11-25:
> Oh, it should be `max7219digit::ScrollMode::CONTINUOUS`. Sorry about that.

You don't have to be sorry, it's already nice to answer me, I just corrected and everything works, a huge thank you. I would have searched for a long time without your help

closed 2021-11-25 by oxan