esphome/issues #2757: Anova Component Functional/Parsing Issue

brad07x, 2021-11-27:
### The problem

### Summary
I'm experiencing issues while attempting first-time setup of the ESPHome Anova component with my Anova Precision Cooker stick (older model 900W bt/wifi).

Flash, config, and connection (according to logs) all went well, but it appears that there is a parsing or data collection error with my stick's reported temperature status data. 

No matter the actual current/target values on the stick, the Anova Climate component of ESPHome displays -17.78°C in the logs for both the current and target temperature.

### Example Logs

```
[19:01:54][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[19:01:54][D][climate:384]:   Mode: OFF
[19:01:54][D][climate:404]:   Current Temperature: -17.78°C
[19:01:54][D][climate:410]:   Target Temperature: -17.78°C
```

### Hypothesis

What I find particularly odd about this is that -17.78°C is -0.004°F when converted. Not sure about the actual root cause, but the proximity to '0' prior to the conversion to °C for the logs is suspect. This is also backed up by the state values from HA:

```
entity: climate.5424_anova_sous_vide_cooker
state: off
hvac_modes:
  - 'off'
  - heat
min_temp: 77
max_temp: 212
target_temp_step: 0.1
current_temperature: -0.0
temperature: -0.0
friendly_name: 5424 Anova Sous Vide Cooker
supported_features: 1
```

### Observations when interacting from HA/Lovelace:
- I can successfully toggle the stick on and off with the 'power' control.
- The 'heat mode' control toggles the stick on
- Setting the temperature using the climate setpoint slider works properly, updating to the correct temp on the stick. 

### Observations when interacting from the stick:
- State information (heat mode) stays in sync and updates in HA when starting/stopping the stick using its controls.
- Changing the temperature setpoint from the stick has no effect on the reported/parsed target temperature value (remains -17.78C)
- Changes to the actual temperature measured by the stick have no effect on the reported/parsed current temperature value (remains -17.78C)

### Conclusion
Overall, I think the core issue is related to either irregular/unexpected values coming back from the stick, or how the anova component is parsing them. The proximity to '0' before the conversion to -17.78C in the logs is suspect.

I'm happy collect debug logs if someone can point me in the best direction to enable them. I already tried setting the ESPHome logger to DEBUG level and have debug: configured, but this didn't seem to increase the verbosity in a meaningful way.

I may also attempt to set up a BLE sniffer to capture the Anova communication from my stick at a lower level to rule out the reported values from my stick.

Thanks in advance for your time/assistance!

### Which version of ESPHome has the issue?

2021.11.1 & 2021.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

[... 125 lines cut ...]
[00:00:52][I][ble_client:341]:  characteristic 0x2A27, handle 0x1a, properties 0x2
[00:00:52][I][ble_client:341]:  characteristic 0x2A28, handle 0x1c, properties 0x2
[00:00:52][I][ble_client:341]:  characteristic 0x2A29, handle 0x1e, properties 0x2
[00:00:52][I][ble_client:341]:  characteristic 0x2A2A, handle 0x20, properties 0x2
[00:00:52][I][ble_client:341]:  characteristic 0x2A50, handle 0x22, properties 0x2
[00:00:52][I][ble_client:161]: Service UUID: 0xFFE0
[00:00:52][D][esp32_ble_tracker:214]: Starting scan...
[00:00:52][D][anova:095]: Anova units is fahrenheit
[00:00:52][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[00:00:52][D][climate:384]:   Mode: OFF
[00:00:52][D][climate:404]:   Current Temperature: nan°C
[00:00:52][D][climate:410]:   Target Temperature: nan°C
[00:00:53][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[00:00:53][D][climate:384]:   Mode: OFF
[00:00:53][D][climate:404]:   Current Temperature: nan°C
[00:00:53][D][climate:410]:   Target Temperature: -17.78°C
[00:00:53][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[00:00:53][D][climate:384]:   Mode: OFF
[00:00:53][D][climate:404]:   Current Temperature: -17.78°C
[00:00:53][D][climate:410]:   Target Temperature: -17.78°C
[00:00:58][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[00:00:58][D][climate:384]:   Mode: OFF
[00:00:58][D][climate:404]:   Current Temperature: -17.78°C
[00:00:58][D][climate:410]:   Target Temperature: -17.78°C
```

oxan, 2021-11-27:
This is probably caused by esphome/esphome#2659 and related to the number parsing.

Could you add this to your YAML, and report back with the logs?
```yaml
external_components:
  - source: github://oxan/esphome@helpers/anova-temperature
    components: [ anova ]
```

brad07x, 2021-11-27:
Looks like I now get an extra `, out=0.00` line in the logs.

```
[10:46:39][D][ble_client:047]: Found device at MAC address [01:02:03:04:61:53]
[10:46:39][I][ble_client:085]: Attempting BLE connection to 01:02:03:04:61:53
[10:46:40][I][ble_client:161]: Service UUID: 0x1800
[10:46:40][I][ble_client:162]:   start_handle: 0x1  end_handle: 0xb
[10:46:40][I][ble_client:341]:  characteristic 0x2A00, handle 0x3, properties 0xe
[10:46:40][I][ble_client:341]:  characteristic 0x2A01, handle 0x5, properties 0x2
[10:46:40][I][ble_client:341]:  characteristic 0x2A02, handle 0x7, properties 0xa
[10:46:40][I][ble_client:341]:  characteristic 0x2A03, handle 0x9, properties 0x8
[10:46:40][I][ble_client:341]:  characteristic 0x2A04, handle 0xb, properties 0x2
[10:46:40][I][ble_client:161]: Service UUID: 0x1801
[10:46:40][I][ble_client:162]:   start_handle: 0xc  end_handle: 0xf
[10:46:40][I][ble_client:341]:  characteristic 0x2A05, handle 0xe, properties 0x20
[10:46:40][I][ble_client:161]: Service UUID: 0x180A
[10:46:40][I][ble_client:162]:   start_handle: 0x10  end_handle: 0x22
[10:46:40][I][ble_client:341]:  characteristic 0x2A26, handle 0x18, properties 0x2
[10:46:40][I][ble_client:341]:  characteristic 0x2A27, handle 0x1a, properties 0x2
[10:46:40][I][ble_client:341]:  characteristic 0x2A28, handle 0x1c, properties 0x2
[10:46:40][I][ble_client:341]:  characteristic 0x2A29, handle 0x1e, properties 0x2
[10:46:40][I][ble_client:341]:  characteristic 0x2A2A, handle 0x20, properties 0x2
[10:46:40][I][ble_client:341]:  characteristic 0x2A50, handle 0x22, properties 0x2
[10:46:40][I][ble_client:161]: Service UUID: 0xFFE0
[10:46:40][I][ble_client:162]:   start_handle: 0x23  end_handle: 0xffff
[10:46:40][I][ble_client:341]:  characteristic 0xFFE1, handle 0x25, properties 0x16
[10:46:40][D][esp32_ble_tracker:214]: Starting scan...
[10:46:41][D][anova:095]: Anova units is fahrenheit
[10:46:41][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[10:46:41][D][climate:384]:   Mode: OFF
[10:46:41][D][climate:404]:   Current Temperature: nan°C
[10:46:41][D][climate:410]:   Target Temperature: 0.00°C
, out=0.00
[10:46:42][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[10:46:42][D][climate:384]:   Mode: OFF
[10:46:42][D][climate:404]:   Current Temperature: nan°C
[10:46:42][D][climate:410]:   Target Temperature: -17.78°C
, out=0.00
[10:46:42][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[10:46:42][D][climate:384]:   Mode: OFF
[10:46:42][D][climate:404]:   Current Temperature: -17.78°C
[10:46:42][D][climate:410]:   Target Temperature: -17.78°C
[10:46:47][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[10:46:47][D][climate:384]:   Mode: OFF
[10:46:47][D][climate:404]:   Current Temperature: -17.78°C
[10:46:47][D][climate:410]:   Target Temperature: -17.78°C
, out=0.00
[10:46:47][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[10:46:47][D][climate:384]:   Mode: OFF
[10:46:47][D][climate:404]:   Current Temperature: -17.78°C
[10:46:47][D][climate:410]:   Target Temperature: -17.78°C
, out=0.00
[10:46:48][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[10:46:48][D][climate:384]:   Mode: OFF
[10:46:48][D][climate:404]:   Current Temperature: -17.78°C
[10:46:48][D][climate:410]:   Target Temperature: -17.78°C
[10:46:57][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[10:46:57][D][climate:384]:   Mode: OFF
[10:46:57][D][climate:404]:   Current Temperature: -17.78°C
[10:46:57][D][climate:410]:   Target Temperature: -17.78°C
, out=0.00
[10:46:57][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[10:46:57][D][climate:384]:   Mode: OFF
[10:46:57][D][climate:404]:   Current Temperature: -17.78°C
[10:46:57][D][climate:410]:   Target Temperature: -17.78°C
```

oxan, 2021-11-27:
Ah, my bad, not exactly what I wanted to see... can you try again?

You probably need to set `refresh` to zero to get the new code:
```yaml
external_components:
  - source: github://oxan/esphome@helpers/anova-temperature
    components: [ anova ]
    refresh: 0s
```

brad07x, 2021-11-27:
That looks like it made a difference - have a lot more now:
```
[11:25:37][D][anova:095]: Anova units is fahrenheit
[11:25:37][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[11:25:37][D][climate:384]:   Mode: OFF
[11:25:37][D][climate:404]:   Current Temperature: nan°C
[11:25:37][D][climate:410]:   Target Temperature: 0.00°C
[11:25:37][W][anova:108]: sizeof(buf)=32, buf=37 37 2E 31 0D 00 00 00, out=0.00
[11:25:37][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[11:25:37][D][climate:384]:   Mode: OFF
[11:25:37][D][climate:404]:   Current Temperature: nan°C
[11:25:37][D][climate:410]:   Target Temperature: -17.78°C
[11:25:38][W][anova:124]: sizeof(buf)=32, buf=37 32 2E 37 0D 00 00 00, out=0.00
[11:25:38][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[11:25:38][D][climate:384]:   Mode: OFF
[11:25:38][D][climate:404]:   Current Temperature: -17.78°C
[11:25:38][D][climate:410]:   Target Temperature: -17.78°C
[11:25:40][D][esp32_ble_tracker:719]: Found device 28:6B:09:2C:E9:DF RSSI=-96
[11:25:40][D][esp32_ble_tracker:740]:   Address Type: RANDOM
[11:25:42][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[11:25:42][D][climate:384]:   Mode: OFF
[11:25:42][D][climate:404]:   Current Temperature: -17.78°C
[11:25:42][D][climate:410]:   Target Temperature: -17.78°C
[11:25:43][W][anova:108]: sizeof(buf)=32, buf=37 37 2E 31 0D 00 00 00, out=0.00
[11:25:43][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[11:25:43][D][climate:384]:   Mode: OFF
[11:25:43][D][climate:404]:   Current Temperature: -17.78°C
[11:25:43][D][climate:410]:   Target Temperature: -17.78°C
[11:25:43][W][anova:124]: sizeof(buf)=32, buf=37 32 2E 37 0D 00 00 00, out=0.00
[11:25:43][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[11:25:43][D][climate:384]:   Mode: OFF
[11:25:43][D][climate:404]:   Current Temperature: -17.78°C
[11:25:43][D][climate:410]:   Target Temperature: -17.78°C
```

oxan, 2021-11-27:
Ah, the message from the Anova contains a carriage return, which the parsing code doesn't accept. That return also explains why the log messages were broken in my previous attempt...

Let me cook up a fix.

brad07x, 2021-11-27:
@oxan Thanks!!

amoebapl, 2021-11-27:
Hello.
I have an identical problem after the last update.
When can we expect a fix?

oxan, 2021-11-27:
> @oxan Thanks!!

Well, I was also the one who broke it :wink:

Can you try esphome/esphome#2816?
```yaml
external_components:
  - source: github://pr#2816
    components: [ anova ]
```

amoebapl, 2021-11-27:
In fact, I have this set:
unit_ of measure: c

After applying the above patch, everything is fine :)

[18:05:45][D][climate:381]: 'Anova Sous Vide' - Sending state:
[18:05:45][D][climate:384]:   Mode: HEAT
[18:05:45][D][climate:404]:   Current Temperature: 72.00°C
[18:05:45][D][climate:410]:   Target Temperature: 72.00°C

Thanks a lot.

brad07x, 2021-11-27:
@oxan Applying the patch you linked did the trick! 

Temp values parsing as expected now. If @amoebapl has `unit_of_measurement: c` set and working, I have `unit_of_measurement: f` covered/working as well.

```
[13:06:47][D][climate:381]: '5424 Anova Sous Vide Cooker' - Sending state:
[13:06:47][D][climate:384]:   Mode: HEAT
[13:06:47][D][climate:404]:   Current Temperature: 26.17°C
[13:06:47][D][climate:410]:   Target Temperature: 49.17°C
```

All the functional issues I described in the original issue report seem to be resolved as a result of the current/target temp values being sane now. 

Now I just have to iron out getting the third digit of my target temp setpoint to display in the lovelace card in HA. I'd guess that's a side effect of using an inferior unit of measure 😉  - I can imagine the HA devs would not have anticipated the climate/thermostat card facilitating normal household HVAC to have a heating setpoint of >99°F as that seems uncommon/downright hazardous.

closed 2021-11-28 by jesserockz