esphome/issues #2759: EZO component for EC sensor returns 0 due to parsing issue

LydiaSevelt, 2021-11-27:
### The problem

The EZO component returns zero due to a parsing issue.

By default the EC sensor returns four comma separated values, EC,TDS,Salinity,Specific Gravity
The new number parsing code is not able to handle this properly.

Current workaround is to insert a null in place of the first comma so that only the first value is read by the number parser.

In ezo.cpp line 77
 77   for (int i = 1; i < 20; i++) if (buf[i] == ',') { buf[i] = 0; break; }
 78   float val = parse_number<float>((char *) &buf[1], sizeof(buf) - 2).value_or(0);
 79   this->publish_state(val);

This ignores the other values which are returned by default.

### Which version of ESPHome has the issue?

2021.11.2

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

core-2021.11.5 and supervisor-2021.10.8

### What platform are you using?

ESP8266

### Board

nodemcu

### Component causing the issue

ezo

### Example YAML snippet

```yaml
i2c:
  - id: bus_a
    sda: D2
    scl: D1
    scan: True

sensor:
  - platform: ezo
    id: ec_1
    address: 100
    accuracy_decimals: 0
    unit_of_measurement: "cm"
    update_interval: 10s
    name: "EC 1"
```

### Anything in the logs that might be useful for us?

```txt
[13:14:48][VV][api.service:327]: on_ping_request: PingRequest {}
[13:14:48][VV][api.service:043]: send_ping_response: PingResponse {}
[13:14:48][VV][scheduler:152]: Running interval 'update' with interval=10000 last_execution=24563 (now=34568)
[13:14:48][VV][i2c.arduino:121]: 0x64 TX 52
[13:14:49][VV][i2c.arduino:097]: 0x64 RX 01323236312C313232312C312E31362C312E3030
[13:14:49][V][sensor:062]: 'EC 1': Received new state 0.000000
[13:14:49][D][sensor:113]: 'EC 1': Sending state 0.00000 cm with 0 decimals of accuracy
[13:14:49][VV][api.service:140]: send_sensor_state_response: SensorStateResponse {
  key: 4085476073
  state: 0
  missing_state: NO
}
```

oxan, 2021-11-27:
I created #2814 with the fix, could you test if that works?

```yaml
external_components:
  - source: github://pr#2814
    components: [ ezo ]
```

Long-term we could implement something to allow all sensor values to be used, but that'll require a more involved refactor.

LydiaSevelt, 2021-11-27:
I can confirm this code works with an EZO EC sensor to return the first value (the EC)

oxan, 2021-11-27:
Nice, thanks for the excellent bug report!

closed 2021-11-28 by jesserockz