esphome/issues #2760: DNS compression type 67 is unknown at 12

xtimmy86x, 2021-11-27:
### The problem

After the last update, i see in the LOG of the addon and in the LOG of any ESP configured, this line repeated endlessly
{'DNS compression type 67 is unknown at 12': (<class 'zeroconf._exceptions.IncomingDecodeError'>,
                                              IncomingDecodeError('DNS compression type 67 is unknown at 12'),
                                              <traceback object at 0xb5ed8508>)}

### Which version of ESPHome has the issue?

2021.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

nodemcu

### Anything in the logs that might be useful for us?

```txt
this error repeated endlessly
{'DNS compression type 67 is unknown at 12': (<class 'zeroconf._exceptions.IncomingDecodeError'>,
                                              IncomingDecodeError('DNS compression type 67 is unknown at 12'),
                                              <traceback object at 0xb5ed8508>)}
```

oxan, 2021-12-06:
I think this is caused by a zeroconf update in 2021.11? @jesserockz any idea?

ovanpeve, 2021-12-06:
Same kind of endless log with 8266/NodeMCU on 2021.11.5:
{'Corrupt packet received while decoding name': (<class 'zeroconf._exceptions.IncomingDecodeError'>,
                                                 IncomingDecodeError('Corrupt packet received while decoding name'),
                                                 <traceback object at 0x7fa87dba00>)}

rlust, 2021-12-12:
Any update here?
I just updated to 2021.12.1 and issue still happens.

sorted01, 2022-01-05:
For me it started once I've activated deep sleep in the yaml. Any ideas?

Ziggy815, 2022-02-14:
Same error still occurs on 2022.1.3 !

LuckyTriple7, 2022-03-16:
just updated to 2022.03.0
It's the first time the error occurs to me ...

rlust, 2022-03-17:
Just came back with the current release of ESPHome!

53845714nF, 2023-02-02:
I have the problem as well.

bdraco, 2023-11-11:
The incoming packet parser in zeroconf was re-written since this issue was created.  If its still happening, please open a new issue with logging for `zeroconf` set to debug so we can get a copy of the packet that causes the issue.

closed 2023-11-11 by bdraco