esphome/issues #2761: ESP32 camera webserver : Unable to access device if camera is streaming

Guyohms, 2021-11-28:
### The problem

I configured `esp32_camera_web_server` and added the camera to Blue Iris.
If the camera is enabled in Blue Iris, and I try to access the logs or reflash the firmware, the device is not reachable.

I also found the same behavior if I only access the video stream from a web browser.

### Which version of ESPHome has the issue?

2021.11.3

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.3

### What platform are you using?

ESP32

### Board

ESP32-CAM

### Component causing the issue

esp32_camera_web_server

### Example YAML snippet

```yaml
esp32_camera:
  name: Caméra du poulailler
  internal: true
  # Hardware setup
  external_clock:
    pin: GPIO0
    frequency: 20MHz
  i2c_pins:
    sda: GPIO26
    scl: GPIO27
  data_pins: [GPIO5, GPIO18, GPIO19, GPIO21, GPIO36, GPIO39, GPIO34, GPIO35]
  vsync_pin: GPIO25
  href_pin: GPIO23
  pixel_clock_pin: GPIO22
  power_down_pin: GPIO32

  # Config
  max_framerate: 10 fps
  idle_framerate: 0.1 fps
  resolution: 640x480
  jpeg_quality: 10
  contrast: 0
  brightness: 2
  saturation: 0
  vertical_flip: true
  horizontal_mirror: true

esp32_camera_web_server:
  - port: 8080
    mode: stream
  - port: 8081
    mode: snapshot
```

### Anything in the logs that might be useful for us?

```txt
INFO poulailler-camera.local: Error while reading incoming messages: Error while reading data: 0 bytes read on a total of 1 expected bytes
WARNING Can't connect to ESPHome API for poulailler-camera.local: Error while reading data: 0 bytes read on a total of 1 expected bytes
INFO Trying to reconnect to poulailler-camera.local in the background
INFO poulailler-camera.local: Error while reading incoming messages: Error while reading data: 0 bytes read on a total of 1 expected bytes
```

rradar, 2021-11-28:
This PR should improve the situation: https://github.com/esphome/esphome/pull/2813

ayufan, 2021-12-05:
@rradar A little. I'm waiting for this one to be merged to send a follow-up that does fully resolve it.

closed 2022-04-21 by github-actions[bot]