esphome/issues #2762: Climate mode_state_topic is not respected

barcar, 2021-11-28:
### The problem

I’ve been playing with the recently added MQTT custom topics for Climate. Everything seems to work fine except for the Mode topic.

My config is:

```mode_state_topic: ${topic_prefix}/${room}/mode/state ```

But the posts are showing up in the default topic. So, I was looking at the code and it seems like the line:

```cg.add(mqtt_.set_custom_state_topic(config[CONF_MODE_STATE_TOPIC])) ```

is inconsistent with the other changes and should maybe be:

``` cg.add(mqtt_.set_custom_mode_state_topic(config[CONF_MODE_STATE_TOPIC])) ```

Here’s a direct link https://github.com/esphome/esphome/blob/671d68bc2ceecb74756894cd634d1d3cf2653d6e/esphome/components/climate/__init__.py#L216

### Which version of ESPHome has the issue?

2021.11.2

### What type of installation are you using?

pip

### What platform are you using?

ESP8266

### Board

wemos d1 mini

### Component causing the issue

climate

### Example YAML snippet

```yaml
climate:
  - platform: mitsubishi_heatpump
    visual:
      min_temperature: 18 °C
      max_temperature: 25 °C
      temperature_step: 0.1 °C
    name: "${friendly_name}"
    id: "${id}"
    action_state_topic: ${topic_prefix}/${room}/action/state
    away_state_topic: ${topic_prefix}/${room}/away/state
    away_command_topic: ${topic_prefix}/${room}/away/command
    current_temperature_state_topic: ${topic_prefix}/${room}/current_temperature/state
    fan_mode_state_topic: ${topic_prefix}/${room}/fan_mode/state
    fan_mode_command_topic: ${topic_prefix}/${room}/fan_mode/command
    mode_state_topic: ${topic_prefix}/${room}/mode/state
    mode_command_topic: ${topic_prefix}/${room}/mode/command
    swing_mode_state_topic: ${topic_prefix}/${room}/swing_mode/state
    swing_mode_command_topic: ${topic_prefix}/${room}/swing_mode/command
    target_temperature_state_topic: ${topic_prefix}/${room}/target_temperature/state
    target_temperature_command_topic: ${topic_prefix}/${room}/target_temperature/command
    target_temperature_high_state_topic: ${topic_prefix}/${room}/target_temperature_high/state
    target_temperature_high_command_topic: ${topic_prefix}/${room}/target_temperature_high/command
    target_temperature_low_state_topic: ${topic_prefix}/${room}/target_temperature_low/state
    target_temperature_low_command_topic: ${topic_prefix}/${room}/target_temperature_low/command
```

### Additional information

The Climate MQTT mode messages are going to topic:

climate/esphome/bedroom/climate/bedroom_aircon/mode/state

Rather than the expected:

climate/esphome/bedroom/mode/state

All other topic settings are being respected but not this one.

oxan, 2021-11-28:
@definitio Could you take a look at this?

definitio, 2021-11-29:
This is a typo. I'll fix it soon.

closed 2021-11-29 by oxan