esphome/issues #2765: Bot closes PRs even when there is activity

edgimar, 2021-11-30:
### The problem

The PR [here](https://github.com/esphome/esphome-docs/pull/898) was recently closed "due to inactivity", although it was updated shortly after the initial bot warning.  I wonder how many PRs get closed in this way -- I also wonder why there isn't more of an effort to merge PRs, since they generally will enhance the project.

Personally I spent a fair amount of time determining what was needed for the above-mentioned PR.  Just having it closed with no comments whatsoever is rather demotivating, and makes me not want to contribute further to this project.

### Which version of ESPHome has the issue?

N/A

### What type of installation are you using?

pip

### What platform are you using?

ESP32

oxan, 2021-11-30:
> The PR [here](https://github.com/esphome/esphome-docs/pull/898) was recently closed "due to inactivity", although it was updated shortly after the initial bot warning.

That's weird, we're using the standard GitHub stale action. I think it only sees an edit/comment as activity, but not sure... I'll take a look but probably can't do anything about it.

> I also wonder why there isn't more of an effort to merge PRs, since they generally will enhance the project.

Lack of resources. Reviewing all new PRs is at this point almost a full-time job, let alone working through the backlog. The cookbook is unfortunately one of the things that then gets deprioritized.

dougle03, 2022-01-22:
> The cookbook is unfortunately one of the things that then gets deprioritized.

Perhaps the `cookbook` needs to be moved to a more user friendly, and GUI driven environment; as it stands, it's almost impenetrable for average users with basic `yaml` skills; and even less so if submissions are never going to be added.
Maybe there is an argument to consider utilising the wider user base of ESPHome to moderate cookbook entries and updates albeit with a disclaimer over the quality of the content.