esphome/issues #2766: Can not compile board espduomo32

wildblue, 2021-11-30:
### The problem

In file included from /data/cache/platformio/packages/framework-arduinoespressif32/libraries/ESPmDNS/src/ESPmDNS.h:44:0,
from src/esphome/components/mdns/mdns_esp32_arduino.cpp:5:
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:185:26: fatal error: pins_arduino.h: No such file or directory
**********************************************************************
* Looking for pins_arduino.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:pins_arduino.h"
* Web  > https://platformio.org/lib/search?query=header:pins_arduino.h
***********************************************************************
compilation terminated.

### Which version of ESPHome has the issue?

2021-11-4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021-11-05

### What platform are you using?

ESP32

### Board

Espduino32

### Component causing the issue

Dont know. 

### Additional information

Please fix this...

oxan, 2021-11-30:
What _exact_ board setting are you using? There's no `espduino32` board in ESPHome nor PlatformIO.

rradar, 2021-11-30:
No yaml?

closed 2022-04-14 by github-actions[bot]

putz612, 2022-07-24:
Picking this one up because I got the same board.  It is an ESP32 but looks like an Arduino UNO. 

https://docs.platformio.org/en/latest/boards/espressif32/esp32doit-espduino.html
https://www.cnx-software.com/2017/09/04/espduino-32-wemos-d1-r32-esp32-boards-support-some-arduino-uno-shields/
```
esphome:
  name: ${node_name}
  platform: ESP32
  board: esp32doit-espduino
```