esphome/issues #2767: light component does not report brightness feature

jkolczasty, 2021-11-30:
### The problem

light component does not report brightness feature to home assistant:
color_mode: unknown

but the same configuration used with esphome 1.17 works fine and reports:
color_mode: brightness

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Docker

### Which version of Home Assistant has the issue?

2021.8.7

### What platform are you using?

ESP32

### Board

ESP32-WROOM

### Component causing the issue

light monochromatic with ledc(pwm) output

### Example YAML snippet

```yaml
light:
  - platform: monochromatic
    output: "nightlight_pwm"
    name: "nightlight"
    id: "nightlight"

output:
  - platform: ledc
    pin: GPIO12
    id: "nightlight_pwm"
```

### Additional information

Very same configuration works fine with esphome 1.17

oxan, 2021-11-30:
Are you using MQTT?

jkolczasty, 2021-11-30:
No, native HA api.

mmakaay, 2021-11-30:
I have configured an esp8266 with this config, and that worked. In HA I see:

[image]

I don't have an esp32 lying around my office desk, so I couldn't test that.

oxan, 2021-11-30:
> No, native HA api. I'm also pretty sure it's not only 2021.11.4 problem, but in general 2021. as I tried to downgrade to e.g. 2021.10.3, but same effect, no brightness. I remmeber there was similar problem in 2021.8.1 and (maybe) fixed in 2021.8.2? I till test it with 2021.8.2.

Can you try it with a recent HA (i.e. 2021.11.x)? There's been a few rounds of fixes in the HA integration for color mode issues.

jkolczasty, 2021-11-30:
@mmakaay maybe it's related to ledc then, as esp8266 uses pwm output and esp32 ledc output.

jkolczasty, 2021-11-30:
@oxan well, I read many issues people had after upgrading HA to 2021.11.* so I decided to stay with 2021.8, but ok I will try...

oxan, 2021-11-30:
Thinking a bit more about it, I'm pretty sure this caused by esphome/esphome#2204 which indeed breaks with HA versions pre-2021.9. That was the result of an unfortunate mistake, and we've since switched to an API that doesn't break compatibility, but generally using a newer ESPHome than HA isn't well supported.

jkolczasty, 2021-11-30:
@oxan you're right, it works fine with updated HA version:
ESPHome: 2021.11.4
HomeAssistant: core-2021.11.5

mmakaay, 2021-11-30:
Great, I'll close this issue then.
Enjoy the brightness! :-)

closed 2021-11-30 by mmakaay