esphome/issues #2768: sensor filters calibrate_linear do not behave as expected

swifty99, 2021-11-30:
### The problem

sensors allow to calibrate to adjust from measured to true values.
See [calibrate linear](https://esphome.io/components/sensor/index.html#calibrate-linear)

I have tested several setups and never achieved what I would expect.
I would expect for this example:

```
    filters:
      - calibrate_linear:
          # Map 0.0 (from sensor) to 0.0 (true value)
          - 5.0 -> 0.0
          - 680 -> 510.0
          - 950 -> 5000.0
```

if the sensor measures a value above 950 the output would go higher than 5000.

However I see this:

> [16:35:29][D][sensor:113]: 'uhrgruen gemessene adc': Sending state 999.99622 mV with 2 decimals of accuracy
> [16:35:30][D][sensor:113]: 'uhrgruen gemessene Helligkeit': Sending state 3847.20557 lm with 1 decimals of accuracy

it goes even more crazy with a config like this:
```
    filters:
      - calibrate_linear:
          # Map 0.0 (from sensor) to 0.0 (true value)
          - 0.0 -> 0.0
          - 0.08 -> 9.0
          - 0.321 -> 143.0
          - 0.68 -> 510.0
          - 0.92 -> 5000.0
```
with this config the output gets negative!
Maybe the filter works not the way I understand, however documentation implies, another outcome than this:

> [17:09:19][D][sensor:113]: 'uhrgruen gemessene adc': Sending state 0.02835 mV with 2 decimals of accuracy
> [17:09:20][D][sensor:113]: 'uhrgruen gemessene Helligkeit': Sending state -510.23212 lm with 1 decimals of accuracy

for reference the config see below. I used two sensors to measure the conversion

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11

### What platform are you using?

ESP8266

### Board

Wemos D1 mini

### Component causing the issue

sensor core

### Example YAML snippet

```yaml
- platform: adc
    id: "measured_lumens"
    name: "${plug_name} gemessene adc"
    pin: A0
    update_interval: 0.1s   
    unit_of_measurement: "mV" 
    filters:
      - exponential_moving_average:
          send_every: 10
      - multiply: 1000
          

  - platform: template
    name: "${plug_name} gemessene Helligkeit"
    id: "measured_lumens_ha"
    unit_of_measurement: "lm"
    lambda: |-
      return id(measured_lumens).state ;    
    filters:
      - calibrate_linear:
          # Map 0.0 (from sensor) to 0.0 (true value)
          - 5.0 -> 0.0
          - 680 -> 510.0
          - 950 -> 5000.0

    update_interval: 1s
```

swifty99, 2021-12-01:
Hi ,

I have been looking at the code, which is let's say, not too complicated:
`optional<float> CalibrateLinearFilter::new_value(float value) { return value * this->slope_ + this->bias_; }`

I read this in a way, that calibrate linear accepts two points, and two points only to calculate bias and slope.

The first fix should be to mention this in the documentation. If my understanding is right, I would then open a feature request to allow multiple points with liniear interpolations in between. Which would be exactly what I and other people need,  tried (and failed).

cheers

oxan, 2021-12-01:
The `calibrate_linear` function creates one linear (least-square) fit over the whole data range. There's no reason to constrain this to allow just two points, it's perfectly fine to use multiple datapoints for this. Your issue is that your datapoints aren't actually linear, so the linear fit doesn't make any sense. A polynomial fit (`calibrate_polynomial`) will probably work better.

swifty99, 2021-12-01:
Thanx for explanation.
Maybe the documentaion should reflect this.

A polynomal fit is also difficult für nonlinear behaviour and may lead to unexpted results.
I wished for a multistage fit. If this is too special for most I will write my own code, no problem.

oxan, 2021-12-01:
> Thanx for explanation.
> Maybe the documentaion should reflect this.

Feel free to submit a PR with your suggested improvements :)

> I wished for a multistage fit. If this is too special for most I will write my own code, no problem.

A piecewise linear fit would probably be useful to more people, but it should be separate from the current linear calibration filter (e.g. a new `calibrate_piecewise` filter).

swifty99, 2021-12-05:
alright. Will do.

I will close this and add a feature request to implement a spline lookup like this:
https://github.com/ttk592/spline

I will try to do a reference implementation, however my object oriented skills are still to be improved.

closed 2021-12-05 by swifty99