esphome/issues #2769: DNS compression type 95 is unknown at 195' Error during log output.

rlust, 2021-11-30:
### The problem

I am getting the following error in the logs of all of my ESPHome devices since recently updating to the most current release.

### Which version of ESPHome has the issue?

2021.11.4

### What type of installation are you using?

Home Assistant Add-on

### Which version of Home Assistant has the issue?

2021.11.5

### What platform are you using?

ESP8266

### Board

ESP8266 & ESP32

### Component causing the issue

logger

### Example YAML snippet

```yaml
This is happening in many of my devices of ESP32 & ESP8266
```

### Anything in the logs that might be useful for us?

```txt
ESPHOME Log
{'DNS compression type 95 is unknown at 195': (<class 'zeroconf._exceptions.IncomingDecodeError'>,
                                               IncomingDecodeError('DNS compression type 95 is unknown at 195'),
                                               <traceback object at 0x7f6b20954f00>)}
{'DNS compression type 95 is unknown at 195': (<class 'zeroconf._exceptions.IncomingDecodeError'>,
                                               IncomingDecodeError('DNS compression type 95 is unknown at 195'),
                                               <traceback object at 0x7f6b20954f00>)}
```

### Additional information

Most seem to be functioning, but errors started within the last month after updating.
I have 2 Martin Jerry Dimmer switches that are now showing unavailable since issues started.
I have a complete UNIFI system with APs and Dream Machine Pro and Unify Switches.

If I open WEB IF I do not see errors in output.

Minglarn, 2021-12-03:
I've got something similar... Reprogramed one of my ESPs and the logs now show:
```
{'Corrupt packet received while decoding name': (<class 'zeroconf._exceptions.IncomingDecodeError'>,
                                                 IncomingDecodeError('Corrupt packet received while decoding name'),
                                                 <traceback object at 0x7f7630eef0c0>)}
```

oxan, 2021-12-06:
Duplicate of #2760.

rlust, 2022-03-17:
This problem just came back with the current release of ESPHome 2022.3.